openscad/openscad #5435: which version to use?

henning, 2024-11-18:
I wonder which version is recommended to use...

I see the latest official release is from 2021 but the code seems to be in constant development, so I guess the master branch will probably be much further developed, old bugs fixed etc.

Is the master branch usually stable enough for most things or should i better start with the release to get confident with the application first?

I dont mind building if from source and am fine  if the development version has one or another roughness  - that might be outweighted by better features and old bugs fixed.  

But if it's very hard to build and expected to crash every 5min i'd rather go for the older release

jordanbrown0, 2024-11-18:
Many people are happily using the development snapshots, and their rendering is far faster.

For questions like this the mailing list or IRC are probably better.

closed 2024-11-18 by jordanbrown0

t-paul, 2024-11-18:
Snapshot builds at https://openscad.org/downloads.html#snapshots