openscad/openscad #5436: search function interprets named arguments incorrectly

MethylBromide, 2024-11-19:
**Describe the bug**
When using **search** with **index_col_num** and **num_returns_per_match** specified by name, the argument names are ignored and the arguments are interpreted based on their position instead.

**To Reproduce**
Run the code shown below.
Check console output. I'm seeing this:
```
ECHO: [2, 8]
ECHO: []
ECHO: [2]
ECHO: []
ECHO: [2, 8]
ECHO: [0]
```

**Expected behavior**
```
ECHO: [2, 8]
ECHO: [2, 8]
ECHO: [2]
ECHO: [2]
ECHO: []
ECHO: []
```
The point of specifying the names of input arguments is so that they can be listed in any order. Therefore lines 1 and 2 of the output, which use the same argument values with just the positions of the last two named arguments switched, should produce the same output.
Line 4 of the output should be the same as line 3, since we're just allowing **num_returns_per_match** to have its default value as opposed to specifying a value which happens to be the same as the default.
Lines 5 and 6 show what's happening to cause this. Since we're searching for a number at the index position that contains letters, and then vice versa, there should be zero matches. Line 5 also returns multiple results even though **num_returns_per_match** was specified to be 1 -- because the argument **index_col_num**=0 is being interpreted as the **num_returns_per_match** value instead.

**Code reproducing the issue**
```
data= [ ["a",1],["b",2],["c",3],["d",4],["a",5],["b",6],["c",7],["d",8],["e",3] ];
echo(search(3, data, num_returns_per_match=0, index_col_num=1));
echo(search(3, data, index_col_num=1, num_returns_per_match=0));
echo(search(3, data, num_returns_per_match=1, index_col_num=1));
echo(search(3, data, index_col_num=1));
echo(search(3, data, index_col_num=0, num_returns_per_match=1));
echo(search("a", data, index_col_num=1, num_returns_per_match=0));
```

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2024.11.14 (git bc0d078e0) from OpenSCAD.org site.

jordanbrown0, 2024-11-19:
Inspection confirms that the arguments to search() are interpreted positionally, not by name.

kintel, 2024-12-06:
I don't think this is something we're officially supporting, but it could be done I think.
Changing this to be an Enhancement request.

Note: The same probably goes for other functions as well. To solve this we should probably evaluate all functions, decide which should have named argument, implement and write tests for those, then update docs, cheat sheet etc.