openscad/openscad #5438: App crashes with SIGSEGV when using `minkowski(…)` with Manifold, different behaviour depending on how OpenSCAD was invoked

lgarron, 2024-11-19:
**Describe the bug**
OpenSCAD app crashes while previewing

**To Reproduce**

```
brew install openscad@snapshot # Currently installs version 2024.11.10 on macOS

cd /tmp && git clone https:///github.com/lgarron/cube-box/ && cd cube-box
git switch --detach d123b39c4e88c763ebd9c33070f3e98af3f05af4
npm install # or: bun install

# yay 😄
openscad --enable lazy-union --backend Manifold -o ./cube-box-v0.1.3.scad.3mf ./cube-box-v0.1.3.scad

# crashy crashy bad times when the app is set to Manifold
openscad ./cube-box-v0.1.3.scad
```

There are a whole bunch of warnings that are always output when running `openscad` this way:

```
Could not initialize localization (application path is '/opt/homebrew/bin').
qt.qpa.fonts: Populating font family aliases took 146 ms. Replace uses of missing font family "DejaVu Sans" with one that exists to avoid this cost.
Error reading examples.json: examples.json: cannot open file
WARNING: GLView: unknown colorscheme Tomorrow Night
ERROR: Cannot open shader source file: 'MouseSelector.vert'
ERROR: Cannot open shader source file: 'MouseSelector.frag'
/Users/distiller/project/src/gui/MouseSelector.cc: OpenGL Program Linker Error:
ERROR: Compiled vertex shader was corrupt.
ERROR: Compiled fragment shader was corrupt.

/Users/distiller/project/src/gui/MouseSelector.cc: OpenGL symbol retrieval went wrong, id is -1

FALLBACK (log once): Fallback to SW vertex for line stipple
FALLBACK (log once): Fallback to SW vertex processing, m_disable_code: 2000
FALLBACK (log once): Fallback to SW vertex processing in drawCore, m_disable_code: 2000
2024-11-19 04:39:32.822 openscad[72440:2746090] +[IMKClient subclass]: chose IMKClient_Modern
2024-11-19 04:39:32.822 openscad[72440:2746090] +[IMKInputSession subclass]: chose IMKInputSession_Moder
```

When it crashes, only the following additional line is shown:

```
fish: Job 1, 'openscad ./cube-box-v0.1.3.scad' terminated by signal SIGSEGV (Address boundary error)
```

**Expected behavior**
No crash

**Code reproducing the issue**
See above

**Screenshots**
See above

**Environment and Version info (please complete the following information):**
 - OS: [e.g. Windows 10, macOS 12 Monterey, Ubuntu 22.04]
 - System: [e.g. Intel PC 64-bit, Raspberry PI 4]
 - OpenSCAD Version [e.g. 2021.01, 2023.04.16] and where it was installed from

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

**Additional context**
Add any other context about the problem here.

pca006132, 2024-11-19:
Can you try if it fails with the latest snapshot? We updated manifold version recently.

https://openscad.org/downloads.html#snapshots

kintel, 2024-11-19:
You cannot run OpenSCAD in cmd-line mode directly from a symlink in a bin/ folder. This looks like a bug in homebrew's installer.

Try /Applications/OpenSCAD/Contents/MacOS/OpenSCAD

lgarron, 2024-11-20:
> You cannot run OpenSCAD in cmd-line mode directly from a symlink in a bin/ folder. This looks like a bug in homebrew's installer.

Homebrew is one of the recommended install methods at the top of https://openscad.org/downloads.html , so that sounds worth fixing. Are the issues with this spelled out somewhere? Would it be sufficient to implement a wrapper script that invokes `/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD` and passes on all parameters?

> Can you try if it fails with the latest snapshot? We updated manifold version recently.

Same crash, but `/Applications/OpenSCAD.app/Contents/MacOS/OpenSCAD ./cube-box-v0.1.3.scad` works fine.

Sounds like that particular crash for the repro was a red herring. That said, the `openscad@snapshot` app crashes for me like a dozen times a day during normal use, and I'm gonna try to run it from the non-symlinked path to see if I can get a repro for my actual issue.

t-paul, 2024-11-20:
> Homebrew is one of the recommended install methods at the top of https://openscad.org/downloads.html , so that sounds worth fixing. Are the issues with this spelled out somewhere?

Good questions, but wrong place. Homebrew builds/packages a not maintained here, I think https://github.com/Homebrew/homebrew-cask/issues might be the right place for Homebrew specific issues.

It might be a side effect of https://github.com/openscad/openscad/pull/5395, but still needs help/investigation on Homebrew side.

kintel, 2024-11-21:
As a start, I'll investigate the symlink issue outside of the scope of Homebrew, to make sure we have a good story for that.

kintel, 2024-11-23:
I belive my fix above will work; at least it works for manually created symlinks.

closed 2024-11-23 by kintel

lgarron, 2024-11-23:
Glad to hear it!

For what it's worth, crashes are still happening to me but without the red herring repro they're too rare and I have yet to get one with any useful logs.

PR #5448 (2024-11-23): Get canonical location before asking for parent path 
…to correctly resolve a possible symlink of the openscad executable itself

Also removes boost::filesytem from OpenSCAD's dependencies.

Fixes #5438