openscad/openscad #5439: floor not working properly

julvr123, 2024-11-19:
**Describe the bug**
floor(0.6/0.2) returns 2 instead of 3:

  echo("floor(3):",floor(3));
  echo("0.6/0.2:",0.6/0.2);
  echo("floor(0.6/0.2):",floor(0.6/0.2));
produces:
  ECHO: "floor(3):", 3
  ECHO: "0.6/0.2:", 3
  ECHO: "floor(0.6/0.2):", 2

**To Reproduce**
The following code:
    echo("floor(0.6/0.2):",floor(0.6/0.2));

**Expected behavior**
I expect this to return 3

**Code reproducing the issue**
  echo("floor(0.6/0.2):",floor(0.6/0.2));

**Screenshots**
  N/A

**Environment and Version info (please complete the following information):**
 - OS: Windows 10 Home
 - System: Intel(R) Core(TM) i7-7700 CPU @ 3.60GHz   3.60 GHz
 - OpenSCAD  2021.01

nophead, 2024-11-19:
It is because 0.2 and 0.6 are not representable exactly in binary floating point. So the result of the division with be slightly less than 3. Try `echo(0.6/0.2 - 3); ` it gives `ECHO: -4.44089e-16`.

rcolyer, 2024-11-19:
This is an understandable expectation violation if one is not used to how finite representation of floating point values works, but the exact same value of 2 will arise in Python, C, C++, and so on for this calculation on the most common processors.  For example:  https://ideone.com/E7Mr4f

So this is ultimately not a bug.  If the result is shocking, it is a worthwhile expenditure of time to dive into reading about the fundamental consequences of using finite representations, and the way in which this makes situations like this fundamentally unavoidable.  Any step to fix one set of them (e.g. the intuitively "obvious" ones like this when using a decimal representation) will introduce or leave another similar set.  The only real solution is to rethink what floating point values fundamentally mean, as values specified only to a certain finite precision level, and do calculations within this definition.  (Or use integer math.)

julvr123, 2024-11-24:
I appreciate your support.   I have created a function _floor(a) = floor(a+0.000001), which will gets around the issue.