openscad/openscad #5440: Opening an empty file when double-clicking on a `.scad` file in the Finder under macOS 15.1

dhyananbho, 2024-11-19:
**Describe the bug**
I have encountered an issue with OpenSCAD version 2024.11.10 on macOS 15.1 where the application always opens an empty file when double-clicking on a `.scad` file in the Finder.

**To Reproduce**
Double-click on a `.scad` file in the Finder.

**Expected behavior**
OpenSCAD should open with one file – the double-clicked file only.

**Actual Behavior**:
OpenSCAD opens with two files – an empty file and the double-clicked file.

**Environment and Version info (please complete the following information):**
 - macOS Sequoia 15.1
 - MacBook Pro 16" (2021) with Apple M1 Max, 64GB
 - OpenSCAD Version 2024.11.10 (git 6f5932feb)

dhyananbho, 2024-11-21:
That is not the point. OpenSCAD opens the double-clicked file correctly, but it also opens an empty file, which is not the expected behaviour.

Maybe the problem has to do with the fact, that the default file behaviour on macOS is quite different from Windows.
* If you close the last window of an application, under Windows the application is also closed, under macOS the application remains open and only the last window or file is closed (the menu bar of this application is still visible, because under macOS the menu bar is only displayed once for the entire screen, unlike Windows, where the menu bar is displayed in every window. This means that on Windows it is not possible to have an app open without a window, as you cannot interact with the app due to the missing menu bar).
* If you double-click on an app, under macOS a pop-up window appears asking which file should be opened; under Windows, an app opens directly with an empty file, as does OpenSCAD.
-> When double-clicking on a ‘.scad’ file, OpenScad seems to do the standard Windows behaviour (opening the app with an empty file) and then additionally open the ‘.scad’ file, but the app should not open the empty file at all.

kintel, 2024-11-21:
Double-click to open used to work, so it must have broken, just trying to assess if it's broken in the same way everywhere.

kintel, 2024-11-28:
Closing as duplicate of #3307

closed 2024-11-28 by kintel