openscad/openscad #5441: Unable to convert .stl to.png from command line

rullbandspelare, 2024-11-19:
**Describe the bug**
Unable to convert .stl to.png from command line.  
Actually I have two issues as described as A and B below.

**To Reproduce**
https://upload.wikimedia.org/wikipedia/commons/8/82/216_Kleopatra.stl
https://dexma.se/files/trooper1.stl

**THIS DOES NOT WORK**
A.
_C:\openscad>openscad.com -o 216_Kleopatra.png 216_Kleopatra.stl_
ERROR: Parser error: syntax error in file 216_Kleopatra.stl, line 1
Can't parse file '216_Kleopatra.stl'!

B.
_c:\openscad>openscad.com --view axes,scales -o trooper1.png trooper1.stl_
Generates a empty image. There is no object to be found. I have tried to "zoom out" to see where it is supposed to be in the planes.
Compiling design (CSG Products normalization)...
Normalized CSG tree has 1 elements
Geometries in cache: 0
Geometry cache size in bytes: 0
CGAL Polyhedrons in cache: 0
CGAL cache size in bytes: 0
Total rendering time: 0:00:00.130

**THIS WORKS**. And displays figure after render and viewall:
_C:\openscad>openscad.com 216_Kleopatra.stl
C:\openscad>openscad.com  trooper1.stl_

**Environment and Version info (please complete the following information):**
Windows10 22H2 Pro 64-bit x64-based
OpenSCAD Version: 2021.01
System information: Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833 4 CPUs 7.90 GB RAM
User Agent: OpenSCAD/2021.01 (Windows(TM) 10.0 SP 0.0 NTW 1 MSDN 724833)
Compiler: GCC "7.4.0" 64bit

t-paul, 2024-11-19:
Certainly not a bug. STL is not a valid SCAD input. The GUI has a convenience feature to generate an import statement when opening a file or dropping it into the GUI but that's not going to be in scope for the command line.

One option is using the standard input with a mini scad script:
```
echo 'import("216_Kleopatra.stl");' | openscad -o 216_Kleopatra.png -
```

closed 2024-11-19 by t-paul

rullbandspelare, 2024-11-19:
Thanks for the clarification!

I guess Your example is for Linux and not Windows. Not sure how that syntax would be on windows cmd. But saving a file, 216_Kleopatra.scad with the content  import("216_Kleopatra.stl");  is working great

c:\openscad\openscad.com -o 216_Kleopatra.png 216_Kleopatra.scad
Gives me exactly what i want.