openscad/openscad #5443: threads.scad no longer working in 2024 dev snapshots

mgrahamfa, 2024-11-20:
Using the following module works in OpenSCAD v 2021.01 but not in Development snapshots from 2024.
https://www.dkprojects.net/openscad-threads/threads.scad
**Describe the bug**
Using the following module works in OpenSCAD v 2021.01 but not in Development snapshots from 2024.
https://www.dkprojects.net/openscad-threads/threads.scad
There are no errors, but the threads are not rendered

**To Reproduce**
Using Debian 12 Linux
1. download https://www.dkprojects.net/openscad-threads/threads.scad

2. put the following in a file  
```
include <threads.scad>
translate([0,0,5])metric_thread (diameter=59.5, pitch=2.3, length=8, angle=15, leadin=1);
cylinder(h=5,d=65);

```
3.put the threads.scad  in the same directory as the file you create

3.load the file into OpenScad appimage

4. note the rendered object does not have threads in development versions from 2024 but does in released version 2021.01

**Expected behavior**
rendered part will have threads

**Code reproducing the issue**
include <threads.scad>
translate([0,0,5])metric_thread (diameter=59.5, pitch=2.3, length=8, angle=15, leadin=1);
cylinder(h=5,d=65);

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**Environment and Version info (please complete the following information):**
 - OS: Debian 12 derivative bookwormPup64 
 - System: Intel PC 64-bit
 - OpenSCAD Version [2021.01, OpenSCAD-2024.11.20.ai21252-x86_64.appimage] 
 - where it was installed from https://files.openscad.org/snapshots/OpenSCAD-2024.11.20.ai21252-x86_64.AppImage
 - 

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

VGA compatible controller		: Intel Corporation IvyBridge GT2 [HD Graphics 4000] (rev 09) (prog-if 00 [VGA controller])
OpenSCAD Version: 2024.11.20.ai21252 (git 3f0038daa)
System information: Linux 6.1.55 #1 SMP PREEMPT_DYNAMIC Sun Oct 15 02:42:41 UTC 2023 x86_64 Bookworm Pup64 10.0.3 8 CPUs 31.25 GB RAM
User Agent: OpenSCAD/2024.11.20.ai21252 (git 3f0038daa) (Linux x86_64; Bookworm Pup64 10.0.3)
Compiler: GCC "9.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Manifold version: 3.0.0
Qt version: 5.15.4
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20230410
libzip version: 1.5.1
fontconfig version: 2.13.1 (runtime: 2.14.1)
freetype version: 2.10.1 (runtime: 2.12.1)
harfbuzz version: 2.6.4 (runtime: 6.0.0)
cairo version: 1.16.0
lib3mf version: 1.8.1

**Additional context**
Add any other context about the problem here.

t-paul, 2024-11-20:
Works fine here with OpenSCAD Version: 2024.11.18.nightly (git d65040e):

[image]

Please post the output oft the console window after either fresh start of OpenSCAD or after `Design-‍>Clear Cache`

Maybe enable the Manifold Backend: https://fosstodon.org/@OpenSCAD/113256867413539398

mgrahamfa, 2024-11-21:
I tried enabling Manifold, but I've got version 2024.11.20.ai21252 (git 3f0038daa) and it didn't work.
I might try to setup to compile your version, but it may take a while as I've never done that before.
Thanks for your time.  
[image]

closed 2024-11-21 by mgrahamfa

pca006132, 2024-11-21:
@mgrahamfa did you enable lazy union? Your output seems to be different from the output @t-paul got above.

mgrahamfa, 2024-11-21:
yes, lazy union is checked.  I'll try turning it off

mgrahamfa, 2024-11-21:
Disabling it worked!  Thank you so much!

nophead, 2024-11-21:
So that presumably means there is a missing union in threads.scad. But why does that make the thread disappear? Shouldn't it just be a separate object? I have never enabled lazy union.

pca006132, 2024-11-21:
Yeah, this is very weird, probably a bug in lazy union?

KungPhoo, 2025-06-23:
I have the problem, even with lazy unions turned off. Here's my code. Press F5 to see what I wand and F6 or export to see the bug.

[image]
vs.

[image]

```
// use <threadlib/threadlib.scad>
//use <r_shapes.scad>

use <thread_profile.scad>

// bottom
h1 = 12;
d1o = 36;
d1i = 33.3;
t1=1.0;

// top
dhandle=70;

t2=4.2-2.6; // width of thread

tmid=3;

difference(){
    cylinder(h=h1+tmid*2+0.1, d=dhandle); // r_cylinder(.., d1... fr=tmid);
    translate([0,0,-2]) union(){user_thread(mode="nut",  pitch=4, d_minor=d1o-2*t1, d_major=d1o, turns=2, crest_height=1.5);}
    
    // half away for debugging
    translate([0,-50,-1]) cube(100);
}

// ^X^
//  .         |_
//  |            --__ 
//  |                |
//  .                |
//  |            __--+    _
//  |         +--         _ (dz_incl)    -/-
//  .         |           flat_height     |
//  |         |_          _               |
//  |            --__                   pitch
//  .                |   crest_height     |
//  |                |                    |
//  |            __--+    _               |
//  .         +--                        -/-
//  |         |
// 
//
//            |      |
//   d_minor  +      |
//     d_major       +
// d_major:      outer diameter of the thread (size of bore hole)
// d_minor:      inner diameter
// pitch:        vertical distance between crests
// turns:        number of rotations
// crest_height: vertical height of one thead line
// flat_height:  vertical height of the flat space between the thread lines (undef: same as crest_height)
// higbee_arc:   angle to smoothen out the thread at the top and bottom
// mode:         "bolt", "nut"
// nutspace:     additional space for 3D printing, when creating a nut.
// fn:           face number $fn
module user_thread(d_minor, d_major, pitch=4, turns=3, crest_height=0, flat_height, higbee_arc=45, fn=120, mode="bolt", nutspace=0.1){
    $fn=fn;
    add=(mode=="nut" ? nutspace : 0.0);
    t=(d_major-d_minor)/2;
    
    ht = crest_height;
    hf = flat_height == undef ? ht : flat_height;
    

    // 2D profile. [delta_radius,delta_z]
    // [0,0] is at radius d_inner/2 - see below
    dz_incl = max(0, (pitch - ht - hf)/2); // height of the inclined profile part
    prof = [
        [0, 0],
        [t, dz_incl-add],
        [t, dz_incl+ht+add],
        [0, pitch-hf+add]
    ];

    if(mode=="bolt"){
        union(){
            straight_thread(section_profile=prof, r=d_minor/2, pitch=pitch, turns=turns, higbee_arc=higbee_arc, fn=fn);
            cylinder(h=pitch*(1+turns), d=d_minor);
        }
    }
    if(mode=="nut"){
        union(){
            straight_thread(section_profile=prof, r=d_minor/2+add, pitch=pitch, turns=turns, higbee_arc=higbee_arc, fn=fn);
            translate([0,0,-0.05]) cylinder(h=pitch*(1+turns)+0.1, d=d_minor+add*2+0.16);
        }
    }
}
```