openscad/openscad #5444: Quiting the app does not warn about unsaved files under macOS 15.1

dhyananbho, 2024-11-20:
**Describe the bug**
There is no popup dialogue for unsaved files when closing the OpenSCAD app.

**To Reproduce**
1. Double-click on a `.scad` file or open it
2. Change something in the `.scad` file
3. Quit the OpenSCAD app with cmd-Q or the menu quit

**Expected behavior**
OpenSCAD should open a pop up dialog for unsaved changes.

**Actual behavior**:
OpenSCAD simply quits and leaves the `.scad` file unchanged.

**Environment and Version info (please complete the following information):**
 - macOS Sequoia 15.1
 - MacBook Pro 16" (2021) with Apple M1 Max, 64GB
 - OpenSCAD Version 2024.11.10 (git 6f5932feb) and 2024.11.20 (git 3f0038daa)

kintel, 2024-11-21:
This definitely used to work.
Note: Closing the editor tab or the main window does offer such a popup, so the feature is still working, just not for Cmd-Q.

dhyananbho, 2024-11-21:
That is correct. It is only not working for cmd-Q or menu "Quit OpenSCAD".

kintel, 2024-11-30:
I can reproduce this with older snapshots, but it works correctly starting with 2024.11.28.
Could you double-check that on your system as well?

dhyananbho, 2024-11-30:
NOP.
OpenSCAD still does not warn for unsaved files when pressing Cmd-Q in Version 2024.11.29 (git 69632b861) on macOS 15.1.1

dhyananbho, 2024-11-30:
I even deleted the preference file and restarted the Mac.
–> OpenSCAD still does not warn for unsaved files when pressing Cmd-Q

kintel, 2024-11-30:
Right, macOS 15.. I can reproduce on macOS 15, but it works on macOS 14.

kintel, 2024-11-30:
I'll try to upgrade the snapshots to Qt6. Let's try again after that. Wait for #5455 to merge

kintel, 2024-12-01:
Try this: https://output.circle-artifacts.com/output/job/e61fa4a4-7cf7-41c7-9662-ffc4beda95f0/artifacts/0/tmp/out/OpenSCAD-2024.12.01.dmg

It will become published on openscad.org once the nightly build script runs

dhyananbho, 2024-12-01:
I tested version 2024-12-01 git 1ad142a44 with mixed results:

#5444 Quitting the app does not warn about unsaved files under macOS 15.1
–> Now there is warning when pressing Cmd-Q or clicking menu „Quit OpenSCAD“
–> Surprisingly:
	1. when pressing Cmd-W to close the window –> this warning appears in the middle off the external screen where the SCAD-Window is
	2. when pressing Cmd-Q to quit OpenSCAD –> this warning appears in the middle off the internal screen where the menubar of OpenSCAD is.
	This is not a problem, but not consistent either.
–> solved

#5445 The size and position of the window on the screen is not saved for the next time the app opens under macOS 15.1
–> When pressing Cmd-Q the window position and window layout is saved as expected, no matter if there are unsaved file and a warning appears or not, no matter what option is selected in the warning.
–> solved

#5451 The OpenSCAD editors ‘Find and Replace’ sometimes distinguishes between upper and lower case, sometimes not
–> This bug is still there as before.
–> not yet solved

#5463 Copy/paste does not work in file dialogue windows under macOS 15.1.1
–> This bug is still there as before.
–> not yet solved

#5466 Export of STL-Files does not work anymore on macOS 15.1.1 since version 2024-12-01
–> This is a new and very serious bug.
–> I switched back to version 2024-11-29

kintel, 2024-12-01:
Fixed by upgrading to Qt6.

If the inconsistent behavor prevails and is annoying, let's open a separate issue for it.

closed 2024-12-01 by kintel