openscad/openscad #5445: The size and position of the window on the screen is not saved for the next time the app opens under macOS 15.1

dhyananbho, 2024-11-20:
**Describe the bug**
When opening a ‚scad’ file or the OpenSCAD app with an empty file, the window is always of the same small size and positioned in the top center of the screen. The app does not remember the window size and position from the last time it was opened.

**To Reproduce**
1. Double-click on a `.scad` file or open it
2. Change the window size and the position of the window on the screen
3. Change something in the `.scad` file
4. Save the the `.scad` file and exit the OpenSCAD app
5. Double-click on the `.scad` file or open it again

**Expected Behavior**
OpenSCAD should open the `.scad` file at the last (the changed) position on the screen with the last (the changed) size of the window.
This behavior is the default behavior of all apps on macOS (exeptions are extremly rare).

**Actual Behavior**:
OpenSCAD simply opens the default small window in the top center of the screen.

**Environment and Version info (please complete the following information):**
 - macOS Sequoia 15.1
 - MacBook Pro 16" (2021) with Apple M1 Max, 64GB
 - with external DELL U4021QW (40 Inch screen)
 - OpenSCAD Version 2024.11.10 (git 6f5932feb) and 2024.11.20 (git 3f0038daa)

kintel, 2024-11-21:
I'll run some tests using Qt5 vs. Qt6 and see if I can reproduce this issue and whether it depends on Qt version

dhyananbho, 2024-11-21:
Maybe it has to do with using two screens. I use the internal MacBook Pro 16‘ screen as main screen with the menu bar and an external 40’ screen above it, on which I mainly work with documents.

kintel, 2024-11-28:
@dhyananbho Can you reproduce the issue using a single screen?

kintel, 2024-11-29:
It works fine here (using a single screen) with the last few dev snapshots.

dhyananbho, 2024-11-29:
I have tried various combinations of just opening the application (which opens a blank file tab) or double clicking an existing ‘.scad’ file with the internal screen only or with two screens; plugging or unplugging the external screen when the computer was shut down or when the computer was running, when SCAD was running or not running.

There was only one single occasion while using the internal screen only, where I was able to change the window size to full screen size. Since then, OpenSCAD always opens the window on the internal screen with exactly this window size, window position and window layout, no matter how I open OpenSCAD, no matter in which window size I closed OpenSCAD before, no matter I use one or two screens.

-> On my computer it does not work as expected on a single screen either.
-> It may have something to do with the permissions of some preference files (just an idea)
-> Possibly it is related to bug #5440 which was closed due to a duplicate of #3307 (I think they are actually two different bugs but related in some way)

kintel, 2024-11-29:
Following the idea that something may be off with the preference file, could you try the following?
1. With OpenSCAD closed, backup and remove `~/Library/Preferences/org.openscad.OpenSCAD.plist`
2. Run OpenSCAD "normally": Double-click on the app icon. Use OpenSCAD-2024.11.29
3. Resize/move the main window, then close OpenSCAD
4. Verify that ~/Library/Preferences/org.openscad.OpenSCAD.plist` has been saved
5. Restart OpenSCAD, and check if Window position stuck.

Some internal explanation:

When the main window closes, we will store the window geometry in the preferences PList file under the `window.geometry` key, and when opening a new window, we will use the data stored there to initialize window geometry. We use standard Qt mechanisms to do this, so there should be nothing magic going on.

dhyananbho, 2024-11-30:
I tried the steps you described and observed the creation and modification date of the file ~/Library/Preferences/org.openscad.OpenSCAD.plist
* This file is created when OpenSCAD is started for the first time after clicking ‘New’ in the welcome dialogue. –> seems OK.
* There are (at least) two ways to close OpenSCAD:
1. by clicking on the red cross bullet in the upper left corner of the window -> This will save the pref file and the window position is saved correctly. –> seems OK.
2. by entering CMD-Q -> This does not save the pref file and the changed window position is lost -> Not OK.

Conclusion: The cause of the problem is the same as in the previous bug #5444

kintel, 2024-11-30:
Thanks for the test - I'll focus on the Cmd-Q issue for now

kintel, 2024-11-30:
OK, I can reproduce this on macOS 15 now

kintel, 2024-12-01:
Try this: https://output.circle-artifacts.com/output/job/e61fa4a4-7cf7-41c7-9662-ffc4beda95f0/artifacts/0/tmp/out/OpenSCAD-2024.12.01.dmg

It will become published on openscad.org once the nightly build script runs

dhyananbho, 2024-12-01:
I tested version 2024-12-01 git 1ad142a44 with mixed results:

#5444 Quitting the app does not warn about unsaved files under macOS 15.1
–> Now there is warning when pressing Cmd-Q or clicking menu „Quit OpenSCAD“
–> Surprisingly:
	1. when pressing Cmd-W to close the window –> this warning appears in the middle off the external screen where the SCAD-Window is
	2. when pressing Cmd-Q to quit OpenSCAD –> this warning appears in the middle off the internal screen where the menubar of OpenSCAD is.
	This is not a problem, but not consistent either.
–> solved

#5445 The size and position of the window on the screen is not saved for the next time the app opens under macOS 15.1
–> When pressing Cmd-Q the window position and window layout is saved as expected, no matter if there are unsaved file and a warning appears or not, no matter what option is selected in the warning.
–> solved

#5451 The OpenSCAD editors ‘Find and Replace’ sometimes distinguishes between upper and lower case, sometimes not
–> This bug is still there as before.
–> not yet solved

#5463 Copy/paste does not work in file dialogue windows under macOS 15.1.1
–> This bug is still there as before.
–> not yet solved

#5466 Export of STL-Files does not work anymore on macOS 15.1.1 since version 2024-12-01
–> This is a new and very serious bug.
–> I switched back to version 2024-11-29

kintel, 2024-12-01:
Fixed by upgrading to Qt6

closed 2024-12-01 by kintel