openscad/openscad #5447: nondeterministic export_stl.cc assertion failure at line 172

dankamongmen, 2024-11-22:
Several days ago, I started seeing failures on the CI for a project of mine which uses OpenSCAD. The command line is `openscad --hardwarnings --backend Manifold --summary all -o out/scad/croom.stl scad/croom.scad`, and an assertion is being triggered:

```c
time openscad --hardwarnings --backend Manifold --summary all -o out/scad/croom.stl scad/croom.scad
openscad: /home/dank/src/openscad/src/io/export_stl.cc:172: uint64_t {anonymous}::append_stl(std::shared_ptr<const PolySet>, std::ostream&, bool): Assertion `s0 != s1 && s0 != s2 && s1 != s2' failed.
Command terminated by signal 6
```

This does not happen all the time. I can reproduce it on my Linux workstation and laptop; a second invocation is usually sufficient to get things working. I'm going to dig in with valgrind and see what can be seen (I'll also test without Manifold), but wanted to get this report up.

**To Reproduce**
The file listed above is https://github.com/dankamongmen/dankdryer/blob/master/scad/croom.scad.
If you want to look into this, you can check out https://github.com/dankamongmen/dankdryer and run `make out/scad/croom.stl`. As mentioned, it doesn't always happen.

**Expected behavior**
I expect to run to completion without triggering the `assert()`.

**Code reproducing the issue**
Working on this.

**Environment and Version info (please complete the following information):**
 - OS: Debian Linux Unstable
 - System: amd64
 - OpenSCAD Version: 2024.10.17-dirty

**Library & Graphics card information**
representative but not nominative:

```c
[grimes](0) $ openscad --info | grep -v fonts
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.10.17
System information: Linux 6.11.3-arch1-1 #1 SMP PREEMPT_DYNAMIC Thu, 10 Oct 2024 20:11:06 +0000 x86_64 Arch Linux 8 CPUs 15.49 GB RAM
User Agent: OpenSCAD/2024.10.17 (Linux x86_64; Arch Linux)
Compiler: GCC "14.2.1 20240910" 64bit
MinGW build: No
Debug build: No
Boost version: 1_86
Eigen version: 3.4.0
CGAL version, kernels: 6.0, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Qt version: 5.15.15
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.82.1
lodepng version: 20230410
libzip version: 1.10.1
fontconfig version: 2.15.0
freetype version: 2.13.3
harfbuzz version: 10.0.1 (runtime: 10.1.0)
cairo version: 1.18.2
lib3mf version: 2.3.0
Application Path: /usr/local/bin
Documents Path: /home/dank/.local/share
User Documents Path: /home/dank
Resource Path: /usr/local/share/openscad
User Library Path: /home/dank/.local/share/OpenSCAD/libraries
User Config Path: /home/dank/.config/OpenSCAD
Backup Path: /home/dank/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/dank/.local/share/OpenSCAD/libraries
  /usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /System/Library/Fonts
  /Library/Fonts
  /home/dank/Library/Fonts

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLAD version: 2.0.4
OpenGL Version: 4.6 (Compatibility Profile) Mesa 24.2.7-arch1.1
GL Renderer: Mesa Intel(R) UHD Graphics 620 (KBL GT2)
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

[grimes](0) $ 
```

**Additional context**
Like I said, I feel capable of resolving this, and will look into it today. If this is a duplicate or otherwise related to some other bug, though, please link them up. Thanks!

dankamongmen, 2024-11-22:
alright we've got a hot hit from valgrind right off the bat

```c
==160944== Mismatched free() / delete / delete []
==160944==    at 0x48478EF: free (vg_replace_malloc.c:989)
==160944==    by 0x1D931E: CGAL::Failure_exception::Failure_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /usr/local/bin/openscad)
==160944==    by 0x1D9AC8: CGAL::Assertion_exception::Assertion_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /usr/local/bin/openscad)
==160944==    by 0x1DA02D: CGAL::assertion_fail(char const*, char const*, int, char const*) (in /usr/local/bin/openscad)
==160944==    by 0x2003BE: CGAL::Interval_nt<false>::Test_runtime_rounding_modes::Test_runtime_rounding_modes() (in /usr/local/bin/openscad)
==160944==    by 0x1D8249: __static_initialization_and_destruction_0() (in /usr/local/bin/openscad)
==160944==    by 0x1D8333: _GLOBAL__sub_I_ManifoldGeometry.cc (in /usr/local/bin/openscad)
==160944==    by 0x72FAF43: call_init (libc-start.c:145)
==160944==    by 0x72FAF43: __libc_start_main@@GLIBC_2.34 (libc-start.c:347)
==160944==    by 0x1B3334: (below main) (in /usr/local/bin/openscad)
==160944==  Address 0xb5efe50 is 0 bytes inside a block of size 497 alloc'd
==160944==    at 0x4846613: operator new[](unsigned long) (vg_replace_malloc.c:729)
==160944==    by 0x511782D: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_mutate(unsigned long, unsigned long, char const*, unsigned long) (in /usr/lib/lib3mf.so.2.3.0.0)
==160944==    by 0x51320A0: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::_M_append(char const*, unsigned long) (in /usr/lib/lib3mf.so.2.3.0.0)
==160944==    by 0x1F0389: std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > std::operator+<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&&) (in /usr/local/bin/openscad)
==160944==    by 0x1D92F9: CGAL::Failure_exception::Failure_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /usr/local/bin/openscad)
==160944==    by 0x1D9AC8: CGAL::Assertion_exception::Assertion_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /usr/local/bin/openscad)
==160944==    by 0x1DA02D: CGAL::assertion_fail(char const*, char const*, int, char const*) (in /usr/local/bin/openscad)
==160944==    by 0x2003BE: CGAL::Interval_nt<false>::Test_runtime_rounding_modes::Test_runtime_rounding_modes() (in /usr/local/bin/openscad)
==160944==    by 0x1D8249: __static_initialization_and_destruction_0() (in /usr/local/bin/openscad)
==160944==    by 0x1D8333: _GLOBAL__sub_I_ManifoldGeometry.cc (in /usr/local/bin/openscad)
==160944==    by 0x72FAF43: call_init (libc-start.c:145)
==160944==    by 0x72FAF43: __libc_start_main@@GLIBC_2.34 (libc-start.c:347)
==160944==    by 0x1B3334: (below main) (in /usr/local/bin/openscad)
==160944== 
==160944== Mismatched free() / delete / delete []
==160944==    at 0x48478EF: free (vg_replace_malloc.c:989)
==160944==    by 0x1D93E3: CGAL::Failure_exception::Failure_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /usr/local/bin/openscad)
==160944==    by 0x1D9AC8: CGAL::Assertion_exception::Assertion_exception(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >) (in /usr/local/bin/openscad)
==160944==    by 0x1DA02D: CGAL::assertion_fail(char const*, char const*, int, char const*) (in /usr/local/bin/openscad)
[... cut ...]
==160944==    still reachable: 156,624 bytes in 574 blocks
==160944==                       of which reachable via heuristic:
==160944==                         stdstring          : 285 bytes in 1 blocks
==160944==         suppressed: 0 bytes in 0 blocks
==160944== Rerun with --leak-check=full to see details of leaked memory
==160944== 
==160944== For lists of detected and suppressed errors, rerun with: -s
==160944== ERROR SUMMARY: 10 errors from 10 contexts (suppressed: 0 from 0)
Aborted (core dumped)
```

dankamongmen, 2024-11-22:
note that the output above is a different exception, but if we've got memory safety problems, all bets afterwards are of course off. looking into this more deeply now.

dankamongmen, 2024-11-22:
adding `--debug all` and rebuilkding OpenSCAD with `-DCMAKE_BUILD_TYPE=debug -DMANIFOLD_DEBUG=ON`

dankamongmen, 2024-11-22:
using a debug build, i noticed the following output before a failure:
```
CGAL Cache insert: multmatrix([[1,0,0,327.683],[0,1,0,327.6 (0 bytes)
mimalloc: warning: thread 0x7f46fcff06c0: unable to allocate aligned OS memory directly, fall back to over-allocation (268435456 bytes, address: 0x7f46dee00000, alignment: 4194304, commit: 1)
mimalloc: warning: thread 0x7f46f1fc56c0: unable to allocate aligned OS memory directly, fall back to over-allocation (268435456 bytes, address: 0x7f46cee00000, alignment: 4194304, commit: 1)
CGAL Cache insert: multmatrix([[1,0,0,0],[0,1,0,0],[0,0,1,- (0 bytes)
openscad: /home/dank/src/dankamongmen/openscad/src/io/export_stl.cc:172: uint64_t {anonymous}::append_stl(std::shared_ptr<const PolySet>, std::ostream&, bool): Assertion `s0 != s1 && s0 != s2 && s1 != s2' failed.
Command terminated by signal 6
```

that mimalloc output might not be related, but i thought it worth mentioning.

dankamongmen, 2024-11-22:
if i compile with `MIMALLOC_USE=off`, i no longer get those valgrind warnings. it appears mimalloc doesn't get along well with valgrind: https://github.com/microsoft/mimalloc/issues/251. looking deeper.

kintel, 2024-11-22:
@pca006132 this sounds like expected behavior of Manifold V3. Do you agree?

kintel, 2024-11-23:
I can reproduce on macOS. I feel this is a double->float conversion issue, which was introduced almost a year ago. ..but it's possible that recent changes to Manifold has triggered creation of vertices close enough together to trigger this issue.

I'd love a minimal example of the issue for regression test purposes, but expect that to be tricky to pull together.

pca006132, 2024-11-23:
For manifold, it is expected to have degenerate triangles in the output. We will need an improved mesh decimator to remove degenerate triangles with a higher epsilon, which @elalish is working on now iirc.

pca006132, 2024-11-23:
But I'm wondering why degenerate triangles are not allowed in STL?

dankamongmen, 2024-11-23:
> I can reproduce on macOS. I feel this is a double->float conversion issue, which was introduced almost a year ago. ..but it's possible that recent changes to Manifold has triggered creation of [...]

i can try, but do remember that this only happens sometimes. does that not sound more like an uninitialized value or perhaps a race (manifold is multithreaded, right)? i might try running with TSAN/ASAN.

dankamongmen, 2024-11-23:
it's also weird that i only seem to hit this:

```
terminate called after throwing an instance of 'CGAL::Assertion_exception'
  what():  CGAL ERROR: assertion violation!
Expr: -CGAL_IA_MUL(-1.1, 10.1) != CGAL_IA_MUL(1.1, 10.1)
```

when running under valgrind, and i otherwise (sometimes) hit this:

```
openscad: /home/dank/src/openscad/src/io/export_stl.cc:172: uint64_t {anonymous}::append_stl(std::shared_ptr<const PolySet>, std::ostream&, bool): Assertion `s0 != s1 && s0 != s2 && s1 != s2' failed.
```

pca006132, 2024-11-23:
> i can try, but do remember that this only happens sometimes. does that not sound more like an uninitialized value or perhaps a race (manifold is multithreaded, right)? i might try running with [...]

This is unlikely to be a bug, we know that manifold can generate degenerated triangles and it is not deterministic. The non-determinism is kind of fundamental in what manifold does when run in parallel, so we don't know any way to make it deterministic.

kintel, 2024-11-23:
@dankamongmen  I'd say forget about `CGAL::Assertion_exception` in the  context of this issue, as that looks unrelated. ..but if you think it's an indication of a real issue, please open a separate ticket.

elalish, 2024-11-23:
Fundamentally, this is why we say never use STL when manifoldness is important. It's lossy, because it doesn't store the manifold topology. To try and get around this, OpenSCAD has this assertion to avoid duplicate verts, but that is not in general possible - see the classic two cubes joined along an edge. If you want this to be robust, you should remove that assertion and instead use a format that handles manifoldness, like 3MF or GLB (especially with `EXT_mesh_manifold`).

dankamongmen, 2024-11-24:
> @dankamongmen I'd say forget about `CGAL::Assertion_exception` in the context of this issue, as that looks unrelated. ..but if you think it's an indication of a real issue, please open a separate ticket.

agreed

dankamongmen, 2024-11-24:
> This is unlikely to be a bug, we know that manifold can generate degenerated triangles and it is not deterministic. The non-determinism is kind of fundamental in what manifold does when run in parallel, so we don't know any way to make it deterministic.

this sounds rather problematic to me, but i trust that y'all know your problem space and solutions better than i do.

pca006132, 2024-11-25:
> this sounds rather problematic to me, but i trust that y'all know your problem space and solutions better than i do.

We want things to be deterministic, but we also want to maximize the performance, and that requires changing the control flow depending on the execution time of individual operations. However, due to both floating point associativity issue, mesh simplification and some minor details in the algorithm, re-associating the operations will change the output. We can currently make things run ~5 times faster with parallelization (~8 threads. Doesn't scale further with more threads due to sequential operations being the bottleneck). Disabling those optimizations that introduce non-determinism will make it a lot slower, I don't have a concrete number, but probably only around ~1.5 faster than the sequential version. This is why we just ask users to disable parallelization when they want determinism...

There are ways to improve this slightly, but it probably doesn't worth the effort.

kintel, 2024-11-26:
@whitelynx Do you have a reasonably small (as small as possible or at least stand-alone) examples triggering this issue. I'm going to fix it, but I'd like to cover it with a test.

whitelynx, 2024-11-26:
I haven't come up with a suitably small example yet, but I'll work on that. The files generated by this project are pretty complex.

nophead, 2024-11-26:
You can't in general simply remove a degenerate triangle unless all its sides are zero length because it leaves a topological hole in the mesh of zero area.

whitelynx, 2024-11-26:
Sounds like the topology issue should be something that could then be handled by deduplicating vertices, right?

pca006132, 2024-11-26:
> Sounds like the topology issue should be something that could then be handled by deduplicating vertices, right?

There are cases where you should not deduplicate vertices based on their coordinates. In general, STL is just a bad format and people should really switch to something else...

whitelynx, 2024-11-26:
I was able to chop one of my models down to the smallest subset of second-level items that would still exhibit this issue, but it's still pretty big. (640 lines) I'll see if I can come up with an even more minimal example, but here's what I have so far: https://gist.github.com/whitelynx/fa74d8aeb2bd66615808049d6ec6b415

nophead, 2024-11-26:
There is nothing wrong with using STL to represent 3D printable objects, which is what it was intended for. Yes it can't represent two cubes sharing an edge but that can't exist physically and can't be 3D printed. It can represent any 3D printable object. But any 3D model represented in a higher resolution than a float can't be simply truncated to float as that can break the topology. OpenSCAD has always been able to make broken STLs because it went from CGAL coordinates to floats without any mesh repair afterwards,

For my own use STLs need to be deterministic so that I can tell if they have changed.

whitelynx, 2024-11-26:
> For my own use STLs need to be deterministic so that I can tell if they have changed.

I've started using `--enable predictible-output` when using the git version to help with that; it's been pretty solid so far, when the generation actually works. (you can also enable this feature in the GUI)

kintel, 2024-11-26:
I think the gist of what @pca006132 explains above is that starting with Manifold V3, output is not longer deterministic.
That's orthogonal to this issue though. 

@whitelynx Thanks for the slim-down. When I have some time I'll take a look. Feel free to post smaller versions in the meantime.

pca006132, 2024-11-26:
> I think the gist of what @pca006132 explains above is that starting with Manifold V3, output is not longer deterministic. That's orthogonal to this issue though.
> 
> @whitelynx Thanks for the slim-down. When I have some time I'll take a look. Feel free to post smaller versions in the meantime.

Not really starting from V3, but from the start we never guarantee deterministic output when parallelization is enabled. We tried adding a deterministic flag, but it is hard to make sure enabling it make things deterministic, and in general we feel that it is not something we want to optimize for now so we removed it entirely. And this is just about run-to-run determinism, we never guarantee the same output across versions or compilers, because decimator may change, and floating point behavior can change with different compilers.

TLDR; If you want run-to-run deterministic output, disable parallelization (taskset to 1 core should also work). If you want deterministic output across versions, use CGAL.

dankamongmen, 2024-11-27:
> TLDR; If you want run-to-run deterministic output, disable parallelization (taskset to 1 core should also work). If you want deterministic output across versions, use CGAL.

good to know, thanks! in that case i suppose this bug can be closed up as working as intended. i'll do so, but feel free to reopen it.

dankamongmen, 2024-11-27:
> Is there a way to handle the degenerate triangles more gracefully than with a `SIGABRT`? If a triangle is degenerate, can we just remove it from the output?

i'd personally prefer to rerun the procedure internally until it worked (ideally with feedback to the user, ideally with an obvious way to interrupt interactively, and it's probably best if some limit can be provided so that jobs don't run to an epsilon of eternity for cursed models). i.e., you've clearly been able to detect that this instance of the algorithm failed; rather than dumping, retry smoothly.

kintel, 2024-11-28:
Reopening: The assert failure still needs fixing.

closed 2024-12-06 by kintel