openscad/openscad #5451: The OpenSCAD editors ‘Find and Replace’ sometimes distinguishes between upper and lower case, sometimes not.

dhyananbho, 2024-11-25:
**Describe the error**
When using the ‘Find and Replace’ command in the OpenSCAD editor, the search itself is case sensitive (it displays the yellow highlighted results in the text and the number of occurrences at the right end of the search box), but it is not case sensitive when using any of the ‘Reverse’, ‘Next’, ‘Replace’ and ‘All’ buttons.

**To reproduce**
1. open OpenSCAD with an empty file
2. write a word with capital letters (e.g. ‘Error’)
3. write the same word in lower case only (e.g. ‘error’) in the next line
4. select ‘Find and Replace’
5. enter the word with capital letters (e.g. ‘Error’) in the search field
-> This only highlights the word with capital letters and displays the number of currencies (e.g. ‘1’) at the end of the search field.
-> search is case sensitive
1. click on the ‘Next’ button
-> the second word is highlighted in blue
-> Does not pay attention to upper and lower case
1. write a word in the replacement field (e.g. ‘mistake’)
2. click on the ‘All’ button
-> both words are replaced
-> Upper and lower case is not taken into account

**Expected behaviour**
At least the search should always work in the same way.
It would be better if there is a checkbox ‘Case sensitive’ where you can select the type of search.
It would be even better if there is a pop-up window with many more search criteria, such as in BBEdit (see screenshot)

**Environment and Version info (please complete the following information):**
 - macOS Sequoia 15.1.1
 - MacBook Pro 16" (2021) with Apple M1 Max, 64GB
 - OpenSCAD Version 2024.11.23 (git 50ca97a0b)

[image]

IamLRBA, 2024-11-29:
@kintel , this seems to be a good issue to start my contributions to this project.

kintel, 2024-11-29:
Sounds good.

IamLRBA, 2024-11-29:
Mind assigning me to it?

pakhare, 2025-12-16:
/assign-me