openscad/openscad #5453: Setting for icon color

hackwin, 2024-11-27:
**Is your feature request related to a problem? Please describe.**
The black icon for OpenSCAD is hard to spot on my Windows task bar compared to the yellow icon.

**Describe the solution you'd like**
Put a setting into the program to allow the icon to be yellow or black.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
[image]

kintel, 2024-11-27:
FYI: The blue icon means you're using a nightly build. Official releases will have the yellow icon.

kintel, 2024-11-28:
Btw., your screenshot shows a blue logo. We don't have any plans to change the logo, but if someone are artistically inclined and willing to design a set of logos suitable for release vs. nightly, I'm not opposed to modernizing them. It's a fair bit of work though, as logos are used in a number of places. Until then, I'm closing this as not planned.

closed 2024-11-28 by kintel