openscad/openscad #5463: Copy/paste does not work in file dialogue windows under macOS 15.1.1

dhyananbho, 2024-11-30:
**Describe the error**
When opening a file, saving a file or exporting an stl file, a file dialogue window opens. Copying or pasting in the file name box of this window does not work correctly. 

**To reproduce the error**
1. start OpenSCAD with an empty file
2. write several words in the editor
3. select the first word and copy it
4. place the cursor at the end of the editor
5. save the `.scad` file -> the file dialogue appears with the file name highlighted (without file extension)
6. select the Insert menu or press Cmd-V

**Expected behaviour**
Copying or pasting should affect the selected part of the file name.

**Is the behaviour**
Copying or pasting takes place in the editor window at the cursor position.

**Environment and version information**
 - macOS Sequoia 15.1.1
 - MacBook Pro 16’ (2021) with Apple M1 Max, 64GB
 - OpenSCAD version 2024.11.29 (git 69632b861)

kintel, 2024-11-30:
Copy/paste in the search field works for me using 2024.11.29, both on macOS 14 and macOS 15.1.
* Are you trying to paste a specific style? Have your tried copy from various sources to see if all paste attempts fail?
* Does paste fail both with Cmd-V, and Right-click->Paste?

dhyananbho, 2024-11-30:
Which search field do you mean?

I didn't mean the normal search field for find and replace. I was only talking about the file dialogue windows. When you want to open a ‘scad’ file in OpenSCAD, a file open dialogue window appears. There is a search box in the top right corner of this window. This is the search field where copy and paste does not work.

nophead, 2024-11-30:
It works fine for me on Win11. I can paste into it with Ctrl-V. I think it is a function of the OS rather than OpenSCAD.

kintel, 2024-11-30:
This works for me using the latest 2024.11.29 snapshot on both macOS 14 and macos 15.1:

[image]

kintel, 2024-11-30:
@nophead This is a macOS 15-specific issue

dhyananbho, 2024-11-30:
Now I see it:
It works with right-click paste (–> it inserts the clipboard into the search field).
It does not work with Cmd-V (–> it inserts the clipboard into the editor at the cursor position).

I almost never use these right-click-popups, but I use the Cmd-Shortcut a lot.

kintel, 2024-11-30:
@dhyananbho Does it not work, even if you first click in the field to change focus?

dhyananbho, 2024-11-30:
Exact, it does not work with Cmd-V even when I first click in the search field (of the file open dialogue window)
and the same in the file save dialogue window, even when I first click in the file name field and or mark file name without extension.

kintel, 2024-12-01:
Try this: https://output.circle-artifacts.com/output/job/e61fa4a4-7cf7-41c7-9662-ffc4beda95f0/artifacts/0/tmp/out/OpenSCAD-2024.12.01.dmg

It will become published on openscad.org once the nightly build script runs

dhyananbho, 2024-12-01:
I tested version 2024-12-01 git 1ad142a44 with mixed results:

#5444 Quitting the app does not warn about unsaved files under macOS 15.1
–> Now there is warning when pressing Cmd-Q or clicking menu „Quit OpenSCAD“
–> Surprisingly:
	1. when pressing Cmd-W to close the window –> this warning appears in the middle off the external screen where the SCAD-Window is
	2. when pressing Cmd-Q to quit OpenSCAD –> this warning appears in the middle off the internal screen where the menubar of OpenSCAD is.
	This is not a problem, but not consistent either.
–> solved

#5445 The size and position of the window on the screen is not saved for the next time the app opens under macOS 15.1
–> When pressing Cmd-Q the window position and window layout is saved as expected, no matter if there are unsaved file and a warning appears or not, no matter what option is selected in the warning.
–> solved

#5451 The OpenSCAD editors ‘Find and Replace’ sometimes distinguishes between upper and lower case, sometimes not
–> This bug is still there as before.
–> not yet solved

#5463 Copy/paste does not work in file dialogue windows under macOS 15.1.1
–> This bug is still there as before.
–> not yet solved

#5466 Export of STL-Files does not work anymore on macOS 15.1.1 since version 2024-12-01
–> This is a new and very serious bug.
–> I switched back to version 2024-11-29

kintel, 2024-12-01:
I still cannot reproduce in a VM. I will upgrade a machine to macOS 15.1 and re-test.

If anyone else on macOS 15 could test this and report back, that would be appreciated. Use snapshot 2024.12.01 or newer.

kintel, 2026-09-03:
Weird, I tested 2026.06.12 and 2026.08.30 on macOS 26.6, still works as expected for me.

Does anyone else have this issue, or is it isolated to one specific computer?

dhyananbho, 2026-09-03:
May be we are not testing the same. There for I added a screenshot of the STL-FIle-Export-Window. I marked the filename (without extension) and pressed cmd-V (or Menu Edit - paste) and nothing happens.
I use a MacBook Pro M1 max with macOS 26.6.2 and OpenSCAD version 2026.06.12 (git 0a66508c)

[image]

dhyananbho, 2026-09-03:
There is another bug in OpenSCAD on macOS related to file tags:
When you open an .scad file that has a Finder tag (e.g., yellow), make changes, and save it, the tag is lost. The tag should be preserved.
I suspect that OpenSCAD does not read the existing Finder tag when opening the file, and as a result, it is not displayed as the default value in the "Save" or "Save As" dialog.
This bug is not critical, but it disrupts my workflow: I use the yellow tag for all .scad files and sort my files by tag in Finder. This keeps all .scad files at the top and other files (e.g., .stl, .3mf) below. With this bug, every time I edit and save an .scad file, my file order is lost.

kintel, 2026-09-03:
Re. Cmd-V - I do exactly the same as you, and it works as expected. Puzzling..

Re. tags - when I save to an existing file with a tag, the tag is preserved. If I Save As to a new file, there is no tag. My understanding is that your request is remember the tags of the currently open file and offer those as defaults in the Save As dialog? Please open a new ticket as this is a distinct issue from the copy&paste issue.

dhyananbho, 2026-09-04:
Re. Cmd-V - I do exactly the same as you, and it works as expected. Puzzling..
Which computer do you use?

Re. tags - ... Please open a new ticket as this is a distinct issue from the copy&paste issue.
Yes, I'll open a new ticket.

kintel, 2026-09-04:
> Re. Cmd-V - I do exactly the same as you, and it works as expected. Puzzling.. Which computer do you use?
> 
macOS 26.6.2, M1 Pro

dhyananbho, 2026-09-04:
> macOS 26.6.2, M1 Pro

Yes, really puzzling ...

I'm going to try asking a friend who also has a Mac and occasionally uses OpenScad. But it'll take a few days (or weeks) before I see him.