openscad/openscad #5467: Make running `--animate` on multiple cores ?

hzeller, 2024-12-02:
I like using openscad in batch mode to create animations using the [`--animate` option](https://github.com/hzeller/hot-snipper/blob/master/Makefile#L61-L66)

Running this only create a single thread building all the frames, which can take a while. Machines typically have many cores these days so running that in parallel would be neat.

I see two possible solutions; the first might be a bit more involved, but the second should be fairly trivial:

  1. Provide an option `--animation_threads` of sorts, that tells openscad to run multiple renderings in parallel for different frames. Since each frame is entirely state-less, it can be parallelized like that and would be the 'neatest', but might be a bit involved as it could uncover some subtle multi-threading issues (e.g. are all backend-libraries thread-safe ?)
  2. Provide a 'range' option. Next to, say, `--animate=100` , the user can provide a range of frames to be produced, e.g. `--animate_frame_range=50-99` to tell openscad to only render the given range. Then the user can invoke multiple independent openscad instances with different ranges (`--animate_frame_range=0-49` and `--animate_frame_range=50-99`) in scripts or Makefiles. This should be fairly  easy to implement (Alternatively, consider some other sharding functionality, but range seems to be simplest).
  
With the second option in place, we could even simulate the first solution by spawning multiple processes from within openscad itself providing different range options (though implementing platform-independent spawning might be annoying as not every operating system has a neat `posix_spawn()` or even `fork()`).

Anything I am missing ? I might consider test-implementing (2).

kintel, 2024-12-02:
The second options is safe, but won't benefit from any caching, which could limit its usefulness, depending on what kind of designs are being animated.

The first option it tricky for various reasons, but yeah, since we essentially parse-eval-export on unique data structures, we "only" need to be thread compatible everywhere + thread-safe caches.

A third option would be to work on optimizing re-evaluations, as we currently parse the source code from scratch for every frame.

hzeller, 2024-12-02:
To see where time is spent, created a quick flame-graph (runing on a linux `perf` output, using [`pprof`](https://github.com/google/pprof) `-http=:9000` with the [`perf_data_converter`](https://github.com/google/perf_data_converter) installed), it seems the most cycles are used at `VertexArray::createVertex` and in particular using the shared pointers and all their associated costs (could this be put in an arena instead ?).
Is this part used in the re-evaluations that you think might be worth optimizing ?

Also `encodeLZ77` seems to also use significant time (generating PNGs ?)

[image]

Here a `top 30`

```
131409911060 12.34% 12.34% 131409911060 12.34%  __gnu_cxx::__atomic_add (inline)
124136450164 11.66% 24.00% 124136450164 11.66%  __gnu_cxx::__exchange_and_add (inline)
  89585735774  8.41% 32.42% 92639982075  8.70%  encodeLZ77
  56307512612  5.29% 37.71% 56307512612  5.29%  filter
  39370119541  3.70% 41.41% 70962046472  6.67%  filterScanline
  38483747849  3.61% 45.02% 181283013487 17.03%  std::_Sp_counted_base::_M_release (inline)
  33923698888  3.19% 48.21% 33923698888  3.19%  __memcpy_avx_unaligned_erms
  32528801510  3.06% 51.26% 33267486521  3.12%  VertexData::getLastVertex
  31591926931  2.97% 54.23% 31591926931  2.97%  paethPredictor (inline)
  29538871657  2.77% 57.00% 29538871657  2.77%  updateHashChain
  20658276119  1.94% 58.94% 24401125202  2.29%  addAttributeValues (inline)
  19609971900  1.84% 60.79% 19609971900  1.84%  getHash
  18482402025  1.74% 62.52% 31720613156  2.98%  AttributeData::addData (inline)
  18268196041  1.72% 64.24% 233574474300 21.94%  VertexArray::createVertex
  15052344624  1.41% 65.65% 143093046636 13.44%  __gnu_cxx::__exchange_and_add_dispatch (inline)
  13324718970  1.25% 66.90% 13326747247  1.25%  _mi_page_malloc
  11904754688  1.12% 68.02% 11904754688  1.12%  __memset_avx2_unaligned_erms
  11287767979  1.06% 69.08% 30099979375  2.83%  VBORenderer::create_triangle
   9346202275  0.88% 69.96% 13452058902  1.26%  std::vector::emplace_back (inline)
   8807775024  0.83% 70.79% 11639385864  1.09%  VBORenderer::create_vertex
   8379627117  0.79% 71.57% 8379627117  0.79%  boost::hash_detail::hash_mix_impl::fn
   8209577830  0.77% 72.35% 141395346623 13.28%  __gnu_cxx::__atomic_add_dispatch (inline)
   7949201276  0.75% 73.09% 51097683526  4.80%  VBORenderer::add_shader_attributes
   7489462009   0.7% 73.80% 7489462009   0.7%  std::__detail::_Mod_range_hashing::operator() (inline)
   6840164060  0.64% 74.44% 17186104717  1.61%  std::_Hashtable::_M_find_before_node
   6724223185  0.63% 75.07% 6727187969  0.63%  std::vector::operator[] (inline)
   6699064288  0.63% 75.70% 6864026861  0.64%  writeBitsReversed
   6540960585  0.61% 76.31% 12989021566  1.22%  std::vector::_M_default_append
   6514760983  0.61% 76.93% 6567610059  0.62%  [openscad]
   6463831349  0.61% 77.53% 103485093613  9.72%  std::__shared_count::~__shared_count (inline)
```

kintel, 2024-12-03:
Animated PNG export in preview mode are particularly expensive as we rely heavily on GL caching. Having to rebuild VBOs for each frame is going to kill performance, as your results indicate.

pca006132, 2024-12-03:
Added some prints:

```diff
diff --git a/src/glview/VertexArray.cc b/src/glview/VertexArray.cc
index 0363e0007..0aa418d8b 100644
--- a/src/glview/VertexArray.cc
+++ b/src/glview/VertexArray.cc
@@ -62,6 +62,9 @@ void VertexArray::createVertex(const std::array<Vector3d, 3>& points,
                     primitive_index, shape_size, outlines);
   }
   
+  printf("use_count: %ld\n", data()->positionData().use_count());
+  printf("use_count: %ld\n", data()->normalData().use_count());
+  printf("use_count: %ld\n", data()->colorData().use_count());
   addAttributeValues(*(data()->positionData()), points[active_point_index][0], points[active_point_index][1], points[active_point_index][2]);
   if (data()->hasNormalData()) {
     addAttributeValues(*(data()->normalData()), normals[active_point_index][0], normals[active_point_index][1], normals[active_point_index][2]);
diff --git a/src/glview/VertexArray.h b/src/glview/VertexArray.h
index b1dad6913..3b1bed18d 100644
--- a/src/glview/VertexArray.h
+++ b/src/glview/VertexArray.h
@@ -68,6 +68,7 @@ void addAttributeValues(IAttributeData&);
 // Template helper function to load multiple attribute values in one call
 template <typename T, typename ... Args>
 void addAttributeValues(IAttributeData& attrib, T value, Args... values) {
+  printf("%ld\n", attrib.size());
   attrib.addData(value);
   addAttributeValues(attrib, values ...);
 }
```

This is what I got:

```
use_count: 2
use_count: 2
use_count: 2
0
1
2
0
1
2
0
1
2
3
0
1
2
3
use_count: 2
use_count: 2
use_count: 2
0
1
2
0
1
2
0
1
2
3
0
1
2
3
...
```

It seems to me that most of the time we don't really need a vector (SmallVector can avoid allocation in most cases), and we probably don't need the shared pointers.

hzeller, 2024-12-03:
In #5472 is a suggestion for an implementation of (2) (using simple mod-sharding which is the most convenient for a user as they don't have to calculate ranges)

hzeller, 2024-12-07:
The issue is addressed with #5473 

The comments on this issue revealed other aspects that might be worthwhile looking into, in particular revisit the use of shared pointers and possibly simplified containers. These should be filed as separate issues.

@kintel @pca006132 if you have some concrete ideas what to look at, it might be good if you could formulate them in separate issues that they are not forgotten.

closed 2024-12-07 by hzeller