openscad/openscad #5471: [Suggestion] Store objects in variables

Monniasza, 2024-12-03:
**Is your feature request related to a problem? Please describe.**
I want to have ability to store, retrieve and modify model objects like they're any other data object. Particularly, the features of objects should be extractable using methods like [https://github.com/BelfrySCAD/BOSL2/wiki/vnf.scad](BOSL2 VNFs).

**Describe the solution you'd like**
Example of code that extracts features from a simple object:
```scad
   include <BOSL2/std.scad>
   objectInVar = cube(15);
   cubeVNF = vnf(objectInVar);
   offsetCubeVNF = up(50, p=cubeVNF);
   wonkyCubeVNF = vnf_bend(cubeVNF, axis="Y");
```
Example of code that extracts features from a complex object without a VNF features:
```scad
   include <BOSL2/std.scad>
   organizer = fancyGridfinityOrganizationSystem(...); //This has no VNF version
   processedOrganizer = fancyLayoutAndOptimizer(organizer); //This is a function that returns and takes objects
   fancyPaintedOrganizer = paint("rainbow")[processedOrganizer.children(0)]; //Place objects like module(){ [object] }
   fancyOrganizersRest = [ for (i = [1:processedOrganizer.$children-1]) processedOrganizer.children(i) ] //List comprehension with object-to-value
   [fancyPaintedOrganizer]; [fancyOrganizersRest];
```

**Describe alternatives you've considered**
[https://github.com/BelfrySCAD/BOSL2/wiki/vnf.scad](BOSL2 VNFs), but not all models are available as VNFs, like https://github.com/ostat/gridfinity_extended_openscad

**Additional context**
Code that now uses VNFs could now run faster, as it now would use native operations. Also, code that mixes VNFs and modules would be able to freely convert between the two.

gsohler, 2024-12-03:
There is very famous PR which exactly addresses that topic

https://github.com/openscad/openscad/pull/4478

which Did not yet make the merge

until then, if you accept to use another language for being able to store
3D  objects in variables, consider pythonscad.org   ;)

jordanbrown0, 2024-12-04:
Yes, PR #4478 has this feature.  VNF is not an exact match, because VNF is a polyhedron format - it has all of the vertices and edges directly in the value.  PR#4478's geometry values are *not* rendered; they are a copy of the CSG tree.  This makes them less expensive to preview, but doesn't expose (or even calculate!) the vertices and edges.  You can pass them to the `render()` function to yield a value that is a great deal like a VNF.