openscad/openscad #5475: Assert causes syntax error at for in function with assert/let/for

PaulWay, 2024-12-05:
**Describe the bug**
Given a function definition with an `assert` and then a `let`, a `for` loop generates a syntax error.  Without the `assert` it works.

**To Reproduce**
Code:
```
function between_pts(pt1, pt2, steps=2) = [
    // assert(steps>1, "There must be at least two steps")
    let(stop=steps-1, step=(pt2-pt1)/stop)
    for(i=[0:stop]) [pt1.x+step.x*i, pt1.y+step.y*i, pt1.z+step.z*i]
];
echo(between_pts([1, 2, 3], [4, 5, 6]);
```
1. Code as above should echo '[[1, 2, 3], [4, 5, 6]]'
2. Uncomment `assert` line.
3. OpenSCAD indicates a syntax error at the bracket of the `for` loop.

**Expected behavior**
The assertion passes.

Code with `between_pts([1, 2, 3], [4, 5, 6] ,1)` should fail on the assertion.

**Code reproducing the issue**

See above

**Environment and Version info (please complete the following information):**
 - OS: Fedora 41
 - System: Intel PC 64-bit
 - OpenSCAD Version 2021.01, from Flatpak

**Additional context**
As far as I can see from the examples in the `assert` documentation this should be allowed.

kintel, 2024-12-05:
Using `assert()` inside a list comprehension is not allowed, hence the syntax error:
https://github.com/openscad/openscad/blob/master/src/core/parser.y#L547

Pulling the assert outside of the list comprehension fixes the error:
```
function between_pts(pt1, pt2, steps=2) =
  assert(steps>1, "There must be at least two steps")
  [
    let(stop=steps-1, step=(pt2-pt1)/stop)
    for(i=[0:stop]) [pt1.x+step.x*i, pt1.y+step.y*i, pt1.z+step.z*i]
  ];
echo(between_pts([1, 2, 3], [4, 5, 6]));
```

However, it's a valid feature request to allow `assert()` in this case - I just think nobody complained about it until now.
Let's keep this open in case others want to chime in.

PaulWay, 2024-12-05:
Interesting!  I hadn't thought of the list comprehension being 'returned' by the `assert`, as described in 'Using assertions in function' in https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Other_Language_Features#assert .

When I think about it as a list comprehension, it makes much more sense for the `assert` to be outside.  Maybe this could be cleared up in the documentation for `assert` or for list comprehensions.  Let me see if I can add that, at least.

jordanbrown0, 2024-12-05:
Just for sanity and understandability, we should have a rule that anything that's allowed in expression context is allowed in list comprehension context.

kintel, 2024-12-06:
@jordanbrown0 that may extend to more functions than `assert()`. Not sure it's worth digging to deep into, but if you happen to have an idea of the scope, feel free to write it up here for posterity, and in case we get to it some day.

jordanbrown0, 2024-12-06:
> that may extend to more functions than assert().

I haven't looked to compare the two but yes, exactly.  Any difference is an opportunity for confusion.