openscad/openscad #5478: 2D union occasionally yields spurious gap

jordanbrown0, 2024-12-06:
**Describe the bug**
A collection of translated squares, with a couple of them touching edge-to-edge, occasionally leaves a spurious line/micro-gap between two of them.

**To Reproduce**
F6 the program below.  Note that removing any square makes the problem go away, and detaching any of the three "attached" squares makes the problem go away.

**Expected behavior**
"Good" union.

**Code reproducing the issue**
```
square([50, 3]);
translate([0, -3]) // Note this one is -3
    square(3);
translate([6, -3]) // Note this one is -3
    square(3);
translate([12, -6])
    square(3);
translate([19, -6])
    square(3);
translate([25, -6])
    square(3);
translate([31,-6])
    square(3);
translate([38,-6])
    square(3);
translate([44, -3])  // Note this one is -3
    square(3);
translate([50, -6])
    square(3);
translate([57,-6])
    square(3);
translate([63,-6])
    square(3);
translate([69,-6])
    square(3);
translate([76, -6])
    square(3);
translate([82,-6])
    square(3);
translate([88,-6])
    square(3);
translate([95,-6])
    square(3);
```

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2021.01, 2024.10.28

kintel, 2024-12-07:
Since we're using floating point numbers under the hood, translations will not be exact. This is expected, but the resulting 2D object, and any subsequent 3D object should still work and be able to participate in further CSG operations.

jordanbrown0, 2024-12-07:
Floating point is exact for reasonably small integers.  (~10 million for 32-bit float, ~10^16 for 64-bit float.)

Floating point inaccuracy might explain the problem at the one problematic join.  But it doesn't explain why deleting a square that doesn't touch the main body affects that problematic join.

kintel, 2024-12-07:
I have a plausible explanation: The 2D union is performed using the Clipper library. Before computing the union, they quantize the space covered by the bounding box of all polygons into some sub-64-bit integer space. This means that if you change the bounds of the union, precision will change too.

jordanbrown0, 2024-12-07:
Removing a middle square doesn't change the bounding box :-(

kintel, 2024-12-07:
Huh, gotta find another explanation then :)

pca006132, 2024-12-07:
Wonder if this can happen with clipper2

kintel, 2024-12-07:
Clipper2 currently uses single precision, but otherwise looks pretty much the same I think. Single precision causes other issues which is why we haven't switched yet, but I want to try again soon

pca006132, 2024-12-07:
I don't think clipper2 uses single precision, we use double precision with clipper2, see https://www.angusj.com/clipper2/Docs/Units/Clipper/Types/PointD.htm

kintel, 2024-12-07:
I remembered wrong: Clipper1 used 128-bit integers in their core algos while Clipper2 uses 64-bit integers. The double interface is just the front-end before they quantize to int.

pca006132, 2024-12-07:
but is 64 bit not enough?

kintel, 2024-12-07:
We had a problem earlier where they used 64-bit int also for intermediate results, and something did go wrong. I believe it was fixed by using 128-bit math in some places.

Anyway, essentially, I think the net effect is that Clipper2 will be more aggressive at merging vertices than Clipper1.

pca006132, 2024-12-07:
That sounds like a bug in clipper rather than a precision issue...

kintel, 2024-12-07:
Yeah, but I think they were unable to address it in other ways than just throwing bits at it..

pca006132, 2024-12-07:
I see,. It would be nice if you can dig out the issue and see if clipper2 is fine with that, and file a report if it does not work.

kintel, 2024-12-07:
Related: https://github.com/AngusJohnson/Clipper2/issues/777

kintel, 2024-12-08:
Turns out we don't yet use the Manifold triangulator, so I think this is actually an issue with our existing triangulator.
Perhaps a good time to attempt a switch again. I believe all issues we opened has been fixed, I just haven't had time to go through all the torture tests..

kintel, 2024-12-08:
Related to #5482

kintel, 2024-12-08:
Btw., I just realized I cannot reproduce this; it might come down to CPU differences..

pca006132, 2024-12-08:
Reproduced on Linux with an Intel and AMD CPU, so yeah, sounds like either due to mac or arm...

pca006132, 2024-12-08:
Wonder what if you compile for x86 and run it with rosetta?

kintel, 2024-12-08:
Humm, not with Rosetta either.

Anyway, I just merged #5483 which fixes the upstream issue which is about making sure extruded polygons like this become manifold in 3D space.

But we can keep this open until we upgrade to Clipper2 and see if that changes the behavior of this.

kintel, 2024-12-20:
@jordanbrown0 Could you give it a spin with the latest master, where Clipper2 support was merged?