openscad/openscad #5480: Faster VBO

pca006132, 2024-12-07:
It seems to me that most of the time we don't really need a vector (SmallVector can avoid allocation in most cases), and we probably don't need the shared pointers.

Consider the following patch that shows the size of the vectors and the reference count numbers:

```diff
diff --git a/src/glview/VertexArray.cc b/src/glview/VertexArray.cc
index 0363e0007..0aa418d8b 100644
--- a/src/glview/VertexArray.cc
+++ b/src/glview/VertexArray.cc
@@ -62,6 +62,9 @@ void VertexArray::createVertex(const std::array<Vector3d, 3>& points,
                     primitive_index, shape_size, outlines);
   }
   
+  printf("use_count: %ld\n", data()->positionData().use_count());
+  printf("use_count: %ld\n", data()->normalData().use_count());
+  printf("use_count: %ld\n", data()->colorData().use_count());
   addAttributeValues(*(data()->positionData()), points[active_point_index][0], points[active_point_index][1], points[active_point_index][2]);
   if (data()->hasNormalData()) {
     addAttributeValues(*(data()->normalData()), normals[active_point_index][0], normals[active_point_index][1], normals[active_point_index][2]);
diff --git a/src/glview/VertexArray.h b/src/glview/VertexArray.h
index b1dad6913..3b1bed18d 100644
--- a/src/glview/VertexArray.h
+++ b/src/glview/VertexArray.h
@@ -68,6 +68,7 @@ void addAttributeValues(IAttributeData&);
 // Template helper function to load multiple attribute values in one call
 template <typename T, typename ... Args>
 void addAttributeValues(IAttributeData& attrib, T value, Args... values) {
+  printf("%ld\n", attrib.size());
   attrib.addData(value);
   addAttributeValues(attrib, values ...);
 }
```

This is what I got:

```
use_count: 2
use_count: 2
use_count: 2
0
1
2
0
1
2
0
1
2
3
0
1
2
3
use_count: 2
use_count: 2
use_count: 2
0
1
2
0
1
2
0
1
2
3
0
1
2
3
...
```

Note that even though `use_count` is 2, because the function copies the `std::shared_ptr`, we actually only have one single reference. And the vectors are mostly small, at least in the example I got.

This may improve preview performance when constructing the gl buffer.

pca006132, 2024-12-07:
Probably not something we want to work on before a new release though.

kintel, 2024-12-07:
I think there are bigger issues with VBOs; VBOs are used both for rendered results and for previews, but building VBOs for previews is a lot more complex, and the current code is still pretty hard to follow. I was thinking about perhaps splitting those VBO builders, to make it easier to reason about and to perform appropriate optimizations.

closed 2026-10-10 by github-actions[bot]