openscad/openscad #5485: Ever increasing memory when generating animation using Manifold backend (leak?)

hzeller, 2024-12-09:
When creating an animation with the Manifold backend (version 3.0), the memory use is ever increasing.

Let's have a simple animation
```openscad
difference() {
  sphere(r=10, $fn=128);
  rotate([$t*3600, $t*1234, $t*3210]) {
    translate([0, 0, -11]) cylinder(r=3, h=22);
  }
}
```

... then create some 3000 frames with it

```
openscad --animate=3000 --imgsize=2048,2048 --backend=Manifold --render=force some-animation.scad
```

The resident size of RAM is ever increasing; towards the end, the process uses about 10GiB of RAM, while when using the CGAL backend, the memory eventually settles on some (lower) steady-state.

This might be a leak somewhere (it could of course also be some sort of increasing cache without ever cleaned up; but in that case CGAL seems to be more efficient in this regard).

pca006132, 2024-12-09:
I recall kintel talked about some issue with cache size and wanted manifold to implement a mesh size estimation or something. Forgot how that went in openscad.

kintel, 2024-12-09:
Oh, right. I think we register 0 bytes cache size. That would be hard on memory..

kintel, 2024-12-09:
Manifold caching is a bit of a tricky topic, perhaps time to revisit it..

pca006132, 2024-12-09:
I'm actually quite curious about how the geometry cache works. Do we have examples in which there are many hits?

kintel, 2024-12-09:
Lots of hits would happen if there are repeated parts, but often one hit is enough to make a difference.
Consider a conceptual examples like this:
```
minkowski() {
  cube(20);
  sphere(2, $fn=64);
}
translate([10,10,-5]) cylinder(h=30, r=5);
```

If you don't change the minkowski object very often, but work on positioning the hole, you can shave off some time.
Especially in preview mode, where you could move the hole around using Alt-arrows in real-time despite the minkowski taking ~5 seconds to calculate.

pca006132, 2024-12-11:
I see. But as mentioned a long time ago, this kind of cache can also inhibit certain optimizations that manifold performs internally. I think we need to think about what should be cached and what should not, probably require some complicated heuristics.

pca006132, 2024-12-11:
Minkowski should definitely be cached, but for other boolean operations I am not so sure.

pca006132, 2026-01-13:
It should not matter. Holding references already destroys optimizations. This is why I think there should be some heuristics in openscad that decides whether something is worth caching.

pca006132, 2026-01-13:
And regarding performance, the csg tree rewriting we do can have ~50x performance improvement if you are unioning several thousands meshes, so it is quite significant.

coryrc, 2026-01-13:
> I think the stopgap measure we can do is to only cache minkowski. It avoids memory leak, improves performance when the cache is not hit, and is simple. Cost measure for deciding whether to cache something else can be left for later.

I thought it sounded worse, but it does a reasonable starting point at this point.

Some other heuristic ideas: maybe once the CSG tree gets enough children? Maybe have a heap where we only keep access_count * children for each CSG tree hash and we only cache things which reach a certain threshold?

pca006132, 2026-01-14:
Ideally you want the metric to take into account of the complexity of the mesh, e.g., estimated face count.

pca006132, 2026-01-15:
> > Ideally you want the metric to take into account of the complexity of the mesh, e.g., estimated face count.
> 
> Do you have suggestions on how to estimate it? If we can't evaluate anything then I'm not sure what we can do besides, besides have each primitive provide an estimate based on its inputs and have the operators just sum their children? (hmm, intersection could be `min` I think)

That should be a good enough estimate.