openscad/openscad #5486: macOS Windows resizing closes app

IOIO72, 2024-12-09:
**Describe the bug**
Double click on windows title closes the app.

**To Reproduce**
Steps to reproduce the behavior:
1. Open OpenSCAD
2. Open a file
3. Double click on windows title
4. App closes

**Expected behavior**
The window should resize to full viewport size.

**Environment and Version info (please complete the following information):**
 - OS: macOS 15.1.1 Sequoia
 - System: MacBook Pro 16" 2023
 - OpenSCAD Version 2024.12.01 and later

**Library & Graphics card information**
GL context creator: CGL (new)
PNG generator: Core Foundation

GLAD version: 2.0.4
OpenGL Version: 2.1 Metal - 89.3
GL Renderer: Apple M2 Pro
GL Vendor: Apple
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes

sillyfrog, 2024-12-10:
I'm also seeing it crash when I attempt to move the screen (ie: even with a blank canvas it crashes when moving).

**Environment and Version info (please complete the following information):**
 - OS: MacOS 15.1.1 (24B91)
 - System: MacBook Pro M1 Max
 - OpenSCAD Version: 2024.12.09 (git d70aaa0d4) (running from disk image, downloaded from https://openscad.org/downloads.html#snapshots

**Library & Graphics card information**
```
./OpenSCAD --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.12.09 (git d70aaa0d4)
System information: macOS 15.1.1 arm64 MacBookPro18,4 10 CPUs 64.00 GB RAM
User Agent: OpenSCAD/2024.12.09 (git d70aaa0d4) (macOS 15.1.1 arm64 MacBookPro18,4)
Compiler: Clang "15.0.0 (clang-1500.3.9.4)"
MinGW build: No
Debug build: No
Boost version: 1_86
Eigen version: 3.4.0
CGAL version, kernels: 6.0, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Manifold version: 3.0.0
Qt version: 6.5.3
QScintilla version: 2.14.1
InputDrivers:
GLib version: 2.83.0
lodepng version: 20230410
libzip version: 1.9.2
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.18.0
lib3mf version: 2.3.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Volumes/OpenSCAD/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /System/Library/Assets/com_apple_MobileAsset_Font3
  /System/Library/Assets/com_apple_MobileAsset_Font4
  /usr/local/share/fonts
  /usr/share/fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /System/Library/Fonts/Supplemental

GL context creator: CGL (new)
PNG generator: Core Foundation

GLAD version: 2.0.4
OpenGL Version: 2.1 Metal - 89.3
GL Renderer: Apple M1 Max
GL Vendor: Apple
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

I've also attached the MacOS crash report in case it's of value:
[report.txt](https://github.com/user-attachments/files/18070116/report.txt)

kintel, 2024-12-10:
Could someone affected by this try to remove `~/Library/Preferences/org.openscad.OpenSCAD.plist` and see if that changes anything?

sillyfrog, 2024-12-10:
Same thing unfortunately. It creates a new .plist when I create a new file, but still crashes as soon as I try to pan the view.

kintel, 2024-12-10:
Does it make a difference if you use only a single screen?

FYI: It works for me when testing on macOS 15.1.1, so it's a bit hard to understand what could go wrong..

sillyfrog, 2024-12-10:
I just tried a single screen (laptop screen, everything else disconnected), and the same thing (screen shot of config below, unlike external monitor, it's using "default" scaling, which I thought maybe the cause).

I tried deleting the .plist file again as well - however I did notice that all my preferences are still there (recent files, enabled modules etc), is there somewhere else that preferences are saved? The location `--info` gave me does not exist (` ~/Library/Application Support/OpenSCAD`), so there must be configuration getting stored somewhere else 🤷‍♂️.

I've also copied over the application to my `/Applications` folder as I know some apps have weird issues when run for else where (still the same).

[image]

kintel, 2024-12-10:
If you deleted the `~/Library/Preferences/org.openscad.OpenSCAD.plist` but the preferences are still there, that sounds very suspicious. Are you sure you closed all instances of OpenSCAD before deleting it? Otherwise, OpenSCAD will just overwrite the file on exit from in-memory values.

sillyfrog, 2024-12-10:
Trying again this morning (my time, few minutes ago), and confirmed there are no other OpenSCAD instances running, and things are working! :) 

I'm still using the same version I downloaded yesterday (`2024.12.09 (git d70aaa0d4)`)

I was about to reboot to _really_ make sure there was nothing else running (I had already checked with `ps aux | grep -i openscad`), but before I did that I ran `rm ~/Library/Preferences/org.openscad.OpenSCAD.plist`, then opened it (still had "recent files" etc listed), and now things are fine. I've changed nothing (which I know is horrible when trying to diagnose an issue).

I've now renamed it so it's my "Default" OpenSCAD (ie: when I press `F5` in VScode, it's opening up that version), and that's wall working well also (the new engine is awesome! :) )

I've even tried restoring the .plist file from the copy I took yesterday, and things are still working. I've also opened the previous version I was using to try and update/replace the .plist file, and things are still working great.

So from my perspective it's "fixed" - but I've no clue as to _why_ unfortunately :(

sillyfrog, 2024-12-10:
Spoke too soon - it's back :( 

It started happening after opening Cura (5.8 and 5.9 - I opened the wrong version first). That's super weird as that shouldn't be a thing... I'll try and see if I can replicate again and update further...

sillyfrog, 2024-12-10:
OK, figured it out! There's caching for .plist files on MacOS, the way to _really_ reset preferences is to run:
```
rm ~/Library/Preferences/org.openscad.OpenSCAD.plist; killall -u $USER cfprefsd
```
That reliably gets it going for me, and restoring the old .plist file, and doing the `killall`, gets the old one back, and causes it to crash again.

kintel, 2024-12-11:
I think plist files should be safe to share; would you mind sharing the file causing the crash?

sillyfrog, 2024-12-11:
I'm probably a bit paranoid here ;-)

I've uploaded to Google Drive here: https://drive.google.com/file/d/1Fq2mJOZ3-TZXFWD_Qav4Olj0uEuoL2f0/view?usp=drive_link

If you can please let me know when you've downloaded it, I'll then remove it so it's not staying public.

You should be able to copy it in place, and then run `killall -u $USER cfprefsd`, open OpenSCAD, start a new document and pan the view to replicate the issue.

Let me know if you need anything else.

Thanks for your help!

kintel, 2024-12-11:
Done

kintel, 2024-12-11:
I can reproduce the crash as well, even on macOS 14 - thanks a lot!

kintel, 2024-12-11:
Try this: https://output.circle-artifacts.com/output/job/caf2217a-eab5-4644-ad91-788752e5f738/artifacts/0/tmp/out/OpenSCAD-2024.12.11.dmg

sillyfrog, 2024-12-11:
👍 That works for me (after restoring my old preferences file etc)! :) 

Thanks!

rshev, 2025-01-16:
Had the same issue today launching preview from VSCode and OpenSCAD 2025.01 was crashing trying to move its window to another monitor. Same remediation steps helped (removing prefs), so it seems the root cause is still not fixed. Can share my prefs plist if needed.

kintel, 2025-01-16:
@rshev If you could share your prefs and write up a quick description how to reproduce, that would be great!

rshev, 2025-01-19:
Sure, please find my prefs plist attached.

Steps to repro:
- install VSCode OpenSCAD extension https://marketplace.visualstudio.com/items?itemName=Leathong.openscad-language-support (it was crashing with other extensions as well)
- open scad model attached (this was crashing every single time, but it was crashing with others as well)
- launch preview from VSCode
- try to move OpenSCAD window to another monitor or resize using macOS resize menu [image]
- crash

[plist.zip](https://github.com/user-attachments/files/18469570/plist.zip)
[pi5_ssd_vertical.zip](https://github.com/user-attachments/files/18469587/pi5_ssd_vertical.zip)

kintel, 2025-01-19:
See #5599 for another example

Johnr24, 2025-01-20:
if it helps for de-bugging, I also had the VScode extension installed,

PR #5489 (2024-12-11): Workaround for Qt crash bug related to resizing/moving a bad geometry
Fixes #5486