openscad/openscad #5488: Non-fatal assertion failure during boolean operation

logiclrd, 2024-12-10:
I have a model I'm trying to create that is the subtraction of a Minkowski product from simple geometry. When I run a Render pass, I get an assertion failure in the log output. However, rendering is not aborted and the resulting product appears correct.

**To Reproduce**

1. Use this external file:

https://drive.google.com/file/d/1B9nlEM_7Ju50JIH0-M9yKaNuXpEoBSC7/view?usp=sharing

2. Use this SCAD program:

```
difference()
{
  translate([-60, -20, -45])
  cube([120, 220, 50]);
  
  minkowski()
  {
    import("Box form decimated from DynaMesh smoothed 2.stl");
    
    cube([1, 1, 100]);
  }
}
```

3. Run a Render pass (F6).

**Expected behavior**
Rendering completes with no errors.

**Actual Result**
Rendering completes, but the log contains an assertion failure indicating some incorrect assumption within a class called `CGALUtils`:

```
ERROR: CGAL error in CGALUtils::applyUnion3D: CGAL ERROR: assertion violation!
Expr: itl != it->second.end()
File: /usr/include/CGAL/Nef_3/SNC_external_structure.h
Line: 1150
```

**Screenshots**
This screenshot shows both the succesful result of the Render and the assertion failure in the log:

[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 24.04
 - System: AMD Ryzen 7 7840HS w/ Radeon 780M Graphics
 - OpenSCAD Version: 2024.12.09.ai21615 (git d70aaa0d4), running from downloadable `.AppImage`

**Library & Graphics card information**
```
OpenSCAD Version: 2024.12.09.ai21615 (git d70aaa0d4)
System information: Linux 6.8.0-49-generic #49-Ubuntu SMP PREEMPT_DYNAMIC Mon Nov 4 02:06:24 UTC 2024 x86_64 Ubuntu 24.04.1 LTS 16 CPUs 30.66 GB RAM
User Agent: OpenSCAD/2024.12.09.ai21615 (git d70aaa0d4) (Linux x86_64; Ubuntu 24.04.1 LTS)
Compiler: GCC "9.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.1, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.0
Manifold version: 3.0.0
Qt version: 5.15.4
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20230410
libzip version: 1.5.1
fontconfig version: 2.13.1 (runtime: 2.15.0)
[... 412 lines cut ...]
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_shader_atomic_int64
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_NV_vdpau_interop
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

logiclrd, 2024-12-10:
I just realized I didn't have OpenSCAD set to use Manifold. I switched it, cleared caches and re-ran the render, and the assertion failure only happens in the CGAL path. Which I guess makes sense given that the assertion is in a class that contains "CGAL" in its name. :-P

kintel, 2024-12-13:
We probably shouldn't print that error. It means that our attempt at running an optimized version of minkowski failed, and we fall back to a slower, but more robust, implementation.

closed 2026-10-11 by github-actions[bot]