openscad/openscad #5490: When the entire model is #ed, it previews as opaque orange instead of translucent pink

jordanbrown0, 2024-12-11:
**Describe the bug**
When the entire model is #ed, it previews as opaque orange rather than translucent pink.

**To Reproduce**
```
#cube(10);
```

[image]

**Expected behavior**

[image]

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel i7
 - OpenSCAD Version: 2021.01, 2024.10.28

**Additional context**
The CSG Products list for the solo "#cube(10);" has the cube in both the normal and highlighted rendering chains.  With a normal shape in the model the highlighted cube is only in the highlighted list.

kintel, 2026-02-03:
>  except that a couple of times OpenSCAD just crashed. I haven't isolated the scenario.

I made some unfortunate changes in MainWindow, working on it. Let me know if it happens consistently.

jordanbrown0, 2026-02-04:
I filed a separate issue #6610 against the crashing.