openscad/openscad #5493: Boolean operation fails with message that mesh is non-manifold, but mesh appears to be manifold

logiclrd, 2024-12-13:
I have a model file produced by ZBrush. After some manipulation, I have it at a point where no tool I've tried can detect anything wrong with it. Meshlab says:

```
V: 18364 E: 55098 F: 36732

Unreferenced Vertices 0

Boundary Edges 0

Mesh is composed by 1 connected component(s) 

Mesh is two-manifold 

Mesh has 0 holes

Genus is 2
```

The FormWare online STL repair tool says:

```
-> Uploading..
-> In queue..
-> Assigned to repair node
-> Transfering to repair node
-> Transfer done.
-> Analyzing file
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
--> 0 Non-Manifold edges  (?)
--> 0 Inverted normals  (?)
--> 0 Duplicate faces  (?)
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 1530
--> Vertex count changed from 18364 to 18364 (0)
--> Triangle count changed from 36732 to 36732 (0)
-> Ready for download.
```

When I try to do a boolean operation with this mesh in OpenSCAD, the Preview works fine:

[image]

But the Render pass fails with this error:

```
Rendering Polygon Mesh using Manifold...
WARNING: PolySet -> Manifold conversion failed: NotManifold
Trying to repair and reconstruct mesh..
ERROR: [manifold] Input mesh is not closed!
ERROR: [manifold] Surface_mesh -> Manifold conversion failed: NotManifold
```

**To Reproduce**
Steps to reproduce the behavior:
1. Download the model file: https://drive.google.com/file/d/1bTryMxa8yYs1Gnxfges2UE08AjvYhcCg/view?usp=sharing
2. Run the following OpenSCAD program in Render mode:

```
model = "Box form decimated DynaMesh repaired.stl";

difference()
{
  translate([-50, -10, -60])
  cube([100, 200, 50]);

  import(model, convexity = 10);
}
```

**Expected behavior**
A box is generated with the negative of the supplied model.

**Actual result**
An error is emitted indicating that OpenSCAD thinks the model is non-manifold, and the boolean operation is skipped, producing a box with nothing subtracted from it.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2024.12.09 Development Snapshot daily build downloaded from openscad.org

**Library & Graphics card information**
Will edit in later.

kintel, 2024-12-13:
The file you posted about isn't recognized by OpenSCAD as a valid binary STL file. It seems the file size is wrong; the file appears to contain more data than the number of triangles indicate. Are you certain this is the file you've tested on?

kintel, 2024-12-30:
@logiclrd Any update on the corrupted STL?

logiclrd, 2024-12-30:
@kintel Sorry, yes, I forgot :-) I did find a work-around, by the way. Someone suggested simply loading the mesh into Metasequoia and re-saving it. There's no indication of what, if anything, was actually wrong with the mesh, but after Metasequoia re-serialized it, it could be processed.

I had to go through the files, and I think this should be the mesh used to create the screenshot:

https://drive.google.com/file/d/1_bXaJ879qdQXvHfUmIiQdwu2geQaAj0g/view?usp=sharing

But I can't get it to fail at this point. I must have fat-fingered something somewhere either before or after.

kintel, 2024-12-31:
The model you just posted works fine with the latest development snapshot. Closing for now.

closed 2024-12-31 by kintel