openscad/openscad #5496: Shader error for preview window

pca006132, 2024-12-14:
**Describe the bug**

When I build openscad with Qt6 and run the preview, it shows the following:
```
/home/pca006132/code/openscad/src/gui/MouseSelector.cc: OpenGL vertex shader Error:
0(2) : error C7614: OpenGL/ES doesn't allow use of reserved word gl_ModelViewProjectionMatrix
0(2) : error C7614: OpenGL/ES doesn't allow use of reserved word gl_Vertex

/home/pca006132/code/openscad/src/gui/MouseSelector.cc: OpenGL fragment shader Error:
0(1) : error C7573: OpenGL/ES requires precision specifier on this float type (there is no default precision)

/home/pca006132/code/openscad/src/gui/MouseSelector.cc: OpenGL Program Linker Error:
Vertex info
-----------
0(2) : error C7614: OpenGL/ES doesn't allow use of reserved word gl_ModelViewProjectionMatrix
0(2) : error C7614: OpenGL/ES doesn't allow use of reserved word gl_Vertex
(0) : error C2003: incompatible options for link

Fragment info
-------------
0(1) : error C7573: OpenGL/ES requires precision specifier on this float type (there is no default precision)
(0) : error C2003: incompatible options for link

/home/pca006132/code/openscad/src/gui/MouseSelector.cc: OpenGL symbol retrieval went wrong, id is -1
```

**Environment and Version info (please complete the following information):**
 - OS: NixOS 24.11
 - System: Intel PC 64-bit (13th gen i5)
 - OpenSCAD Version: 5e8738563c61ec53ff51e995f912f0b57754ecf6

**Library & Graphics card information**

```
OpenSCAD Version: 2024.09.19
System information: Linux 6.6.63 #1-NixOS SMP PREEMPT_DYNAMIC Fri Nov 22 14:38:37 UTC 2024 x86_64 NixOS 24.11 (Vicuna) 20 CPUs 31.21 GB RAM
User Agent: OpenSCAD/2024.09.19 (Linux x86_64; NixOS 24.11 (Vicuna))
Compiler: Clang "18.1.8 " 64bit
MinGW build: No
Debug build: No
Boost version: 1_81
Eigen version: 3.4.0
CGAL version, kernels: 5.6.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Manifold version: 3.0.0
Qt version: 6.8.0
QScintilla version: 2.13.2
InputDrivers: 
GLib version: 2.82.1
lodepng version: 20230410
libzip version: 1.11.1
fontconfig version: 2.15.0
freetype version: 2.13.3
harfbuzz version: 10.0.1
cairo version: 1.18.2
lib3mf version: 2.2.0
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /home/pca006132/code/openscad/build
Documents Path: /home/pca006132/.local/share
User Documents Path: /home/pca006132/Documents
Resource Path: /home/pca006132/code/openscad
User Library Path: /home/pca006132/.local/share/OpenSCAD/libraries
User Config Path: /home/pca006132/.config/OpenSCAD
Backup Path: /home/pca006132/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/pca006132/.local/share/OpenSCAD/libraries
  /home/pca006132/code/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
[... 235 lines cut ...]
  /nix/store/wwfcmjpwr5chs6c6160qw4sp7bmr1lsl-lmodern-2.005/texmf-dist/fonts/enc/dvips/lm
  /nix/store/wwfcmjpwr5chs6c6160qw4sp7bmr1lsl-lmodern-2.005/texmf-dist/fonts/map/dvipdfm/lm
  /nix/store/wwfcmjpwr5chs6c6160qw4sp7bmr1lsl-lmodern-2.005/texmf-dist/fonts/map/dvips/lm
  /nix/store/wwfcmjpwr5chs6c6160qw4sp7bmr1lsl-lmodern-2.005/texmf-dist/fonts/opentype/public/lm
  /nix/store/wwfcmjpwr5chs6c6160qw4sp7bmr1lsl-lmodern-2.005/texmf-dist/fonts/opentype/public/lm-math
  /nix/store/wwfcmjpwr5chs6c6160qw4sp7bmr1lsl-lmodern-2.005/texmf-dist/fonts/tfm/public/lm
  /nix/store/wwfcmjpwr5chs6c6160qw4sp7bmr1lsl-lmodern-2.005/texmf-dist/fonts/type1/public/lm

GL context creator: EGL (new)
EGL version: 1.5
PNG generator: lodepng

GLAD version: 2.0.4
OpenGL Version: 4.6.0 NVIDIA 560.35.03
GL Renderer: NVIDIA GeForce GTX 970/PCIe/SSE2
GL Vendor: NVIDIA Corporation
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

**Additional context**

It seems that this is deprecated more than 10 years ago: https://stackoverflow.com/questions/4665723/how-should-modern-opengl-shaders-be-written-so-as-to-be-compatible-with-each-oth

pca006132, 2024-12-14:
Probably due to using unsupported GLES

closed 2024-12-14 by pca006132

kintel, 2024-12-15:
It's suspicious though, as your log shows that desktop OpenGL is used