openscad/openscad #5499: POV-Ray export for polyhedron is missing faces

UBaer21, 2024-12-17:
```
tips=7;
loop=tips*2-1; // the length or number points of a star3D

function star3D(r=[10,5],tips=4,z=0) = [
let(chg=360/(tips*2))
  for(i=[0:tips*2-1])
    concat([cos(i*chg),sin(i*chg)]* (i%2?r[0]:r[1]) , z )
];

points=[
each star3D(r= [10,5],z=0,tips=tips),// base
each star3D(r= [10,5],z=5,tips=tips), // 2nd level
each star3D(r= [1,3],z=2,tips=tips) // 3rd level
];

faces= [
[for(i=[0:loop])i ], // base face
[for(i=[loop:-1:0])i + tips*4 ], // 3rd level face in reverse order

for(i=[0:loop])[(i+1)%(tips*2),i,i+ tips*2 ,(i +1)%(tips*2) + tips*2],// 1st level sides creates a quad group for each side
for(i=[0:loop])[(i+1)%(tips*2),i,i+ tips*2 ,(i +1)%(tips*2) + tips*2] +[1,1,1,1]*tips*2// 2nd level sides just add the number of tips*2

];

polyhedron(points,faces=faces);
```
export is
[image]

should be like
[image]

kintel, 2024-12-17:
This looks like a duplicate of #5459

UBaer21, 2024-12-17:
Oh yes you are right, using a union and adding a cube() also solved that here.

closed 2024-12-17 by kintel