openscad/openscad #5502: OpenSCAD app does not automatically pick up file changes, nor picks up file changes on explicit re-render

lgarron, 2024-12-18:
**Describe the bug**
`OpenSCAD.app` does not seem to pick up changes from disk, no matter what I try (until I reopen the app).

**To Reproduce**
Steps to reproduce the behavior:
1. Open any file in `OpenSCAD.app` and preview or render.
2. Make any change to the file and preview or render again.

OpenSCAD behaves as if the file was never changed on disk, whereas normally it normally re-renders automatically after file changes (and definitely picks up changes when explicitly asked to re-render).

**Expected behavior**
The file changes are reflected.

**Code reproducing the issue**
Any code. This will do:

```
cube(10);
```

**Screenshots**
Everything looks normal apart from the fact that file changes are not picked up.

**Environment and Version info (please complete the following information):**
 - OS: macOS 15.1.1
 - System: … macOS?
 - OpenSCAD Version: 2024.12.17 installed via `brew install openscad@snapshot`

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.

```
OpenSCAD Version: 2024.12.17 (git dcf58ff55)
System information: macOS 15.1.1 arm64 Mac14,12 10 CPUs 32.00 GB RAM 
User Agent: OpenSCAD/2024.12.17 (git dcf58ff55) (macOS 15.1.1 arm64 Mac14,12)
Compiler: Clang "15.0.0 (clang-1500.3.9.4)"
MinGW build: No
Debug build: No
Boost version: 1_86
Eigen version: 3.4.0
CGAL version, kernels: 6.0, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.6.0
Manifold version: 3.0.0
Qt version: 6.5.3
QScintilla version: 2.14.1
InputDrivers: 
GLib version: 2.83.0
lodepng version: 20230410
libzip version: 1.9.2
fontconfig version: 2.14.1
freetype version: 2.12.1
harfbuzz version: 6.0.0
cairo version: 1.18.0
lib3mf version: 2.3.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /Applications/OpenSCAD.app/Contents/MacOS
Documents Path: /Users/lgarron/Documents
User Documents Path: /Users/lgarron/Documents
Resource Path: /Applications/OpenSCAD.app/Contents/Resources
User Library Path: /Users/lgarron/Documents/OpenSCAD/libraries
User Config Path: /Users/lgarron/Library/Application Support/OpenSCAD
Backup Path: /Users/lgarron/Documents/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /Users/lgarron/Documents/OpenSCAD/libraries
  /Applications/OpenSCAD.app/Contents/Resources/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /System/Library/Assets/com_apple_MobileAsset_Font3
  /System/Library/Assets/com_apple_MobileAsset_Font4
  /Users/lgarron/.data/fonts
  /usr/local/share/fonts
  /usr/share/fonts
  /Users/lgarron/.fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /Users/lgarron/Library/Fonts
  /System/Library/Fonts/Supplemental

GL context creator: CGL (new)
PNG generator: Core Foundation

GLAD version: 2.0.4
OpenGL Version: 2.1 Metal - 89.3
GL Renderer: Apple M2 Pro
GL Vendor: Apple
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

**Additional context**

jordanbrown0, 2024-12-18:
Check that Design / Automatic Reload and Preview is checked.

kintel, 2024-12-18:
Other ways to reload:
* File->Reload (Ctrl-R)
* Design->Reload and Preview (F4)

lgarron, 2024-12-19:
I'm sorry to report none of those are working for me.

> Check that Design / Automatic Reload and Preview is checked.

This is enabled.

> * File->Reload (Ctrl-R)

⌘R prints "Loaded design…" but does not reflect any changes to the file.

> * Design->Reload and Preview (F4)

This seems to do nothing. 😕

Those are indeed very reasonable suggestions, as they work for me perfectly on the computer where I normally do my OpenSCAD work. But it's just broken on this other computer, even if I put a lot of care into setting up the same development environment. (I've also made sure OpenSCAD has been given "Full Disk Access" in macOS settings.)

lgarron, 2024-12-19:
> But it's just broken on this other computer, even if I put a lot of care into setting up the same development environment. (I've also made sure OpenSCAD has been given "Full Disk Access" in macOS settings.)

Actually, this is inaccurate! It's now also failing on both computers, now that they're both on 2024.12.17. I'm going to see if I can bisect a regression.

kintel, 2024-12-19:
I'll run a test on a fresh macOS 15 machine as well

lgarron, 2024-12-19:
Okay, it's quite possible that I mixed up a few different issues while testing this, and unfortunately I had a typo in one of my test files that actually didn't produce an expected visual result. I've gone through and re-setup OpenSCAD from scratch, and it now appears to be working. Apologies for the trouble. 🥺

(On that subject, is there a recommended way to sync OpenSCAD settings using dotfiles? I think at least part of it is that I was unintentionally using inconsistent settings between the devices, but I've lost track of that due to debugging this issue.)

kintel, 2024-12-19:
We store preferences in `~/Library/Preferences/org.openscad.OpenSCAD.plist`

closed 2024-12-19 by kintel

lgarron, 2024-12-19:
> We store preferences in `~/Library/Preferences/org.openscad.OpenSCAD.plist`

Thanks, that's helpful! (And thanks for the debugging help again.)