openscad/openscad #5504: roof(method = "voronoi") is very sensitive to Clipper precision

kintel, 2024-12-19:
We currently have to use a custom Clipper precision (32-bit bounding box range) for the voronoi version of roof. If we use the default (10^8) precision used elsewhere in OpenSCAD, the "Voronoi diagram" text in the following example fails:
https://github.com/openscad/openscad/blob/master/examples/Basics/roof.scad

If we instead use 64-bit precision (to use Clipper's maximum precision), the clipper operation fails.

Source:
https://github.com/openscad/openscad/blob/master/src/geometry/roof_vd.cc

While this works today, this is very suspicious and smells like a bug.

gsohler, 2025-10-05:
I just looked into some roof examples today by chance.
I testest with latest master and I see this . The Text appears correct to me.

also : constexpr int DEFAULT_PRECISION = 8; is in ClipperUtils.h

Can you tell me, what to look for ? Running here in Linux Mint

[image]

gsohler, 2025-10-06:
Ahh ok. 
I believe I know, whats going on here. This scaling to 32 bit precision is comparable to round-to grid.
Issue is that the algorithm calculates cross points  at a certain point in different way. So the result is very similar but not equal
which results in non-manifold solids. Rouding to grid solves that/closes the holes.

I could study the algorithm more detailled and think about possible improvements