openscad/openscad #5510: Composing paired negative and positive object

lgarron, 2024-12-20:
I often find myself needing to keep track of a negative and positive object together, and I have yet to find a sufficiently [DRY](https://en.wikipedia.org/wiki/Don%27t_repeat_yourself) pattern for this. Suppose I have a `cube(…)` and I'd like to model placing some bolts, where each bolt needs to have some clearance around it. I'd like to write:

```scad
$fn = 64;
CLEARANCE = 0.5;

module hole_and_bolt(r, depth) {
    // Hole ("negative" object)
    translate([ 0, 0, -depth ]) cylinder(h = depth, r = r + CLEARANCE);

    // Bolt ("positive" object)
    color("gray") translate([ 0, 0, CLEARANCE ]) union() {
        scale([ 1, 1, 0.5 ]) cylinder(h = 2, r = r + 2, $fn = 6); // head
        translate([ 0, 0, -depth ]) cylinder(h = depth, r = r);   // stem
    }
}

compose() {
    #color("white") translate([ 0, 0, -10 ]) cube([ 100, 100, 20 ], center = true);

    translate([ 0, 10, 0 ]) hole_and_bolt(r = 4, depth = 10);
    translate([ 40, 40 ]) hole_and_bolt(r = 2, depth = 10);
    translate([ -40, 40 ]) hole_and_bolt(r = 2, depth = 10);
    translate([ 30, -35 ]) hole_and_bolt(r = 3, depth = 10);
    translate([ -30, -35 ]) hole_and_bolt(r = 3, depth = 10);
}
```

A simple implementation of `compose()` could be as follows:

```scad
module extract_negative() { nested_children(0); } // not possible?
module extract_positive() { nested_children(1); } // not possible?

module compose() {
    union() {
        difference() {
            children(0);
            for (i = [0:$children - 2])
                extract_negative() children(i);
        }
        for (i = [0:$children - 2])
            extract_positive() children(i);
    }
}
```
(Ideally the negatives and positives would be sequential rather than parallel, but even the parallel version here would help make a lot more code DRY.)

However, this is not currently possible because `extract_negative()` and `extract_positive()` see the hole and bolt as a combined child instead of two children. The children of any `module hole_and_bolt(…)` get flattened into a single object, and I don't know of any way to construct and refer to both of them together (or of a way to index a "nested" child).

In researching how to implement this, I've found various proposed workarounds that would each let me implement what I need:

- A "negative" transformation: https://github.com/openscad/openscad/issues/3818
- Syntax for arbitrary sequential union and difference without nesting: https://github.com/openscad/openscad/issues/4722
- Unwrap operator: https://github.com/openscad/openscad/wiki/OEP4%3A-Unwrap-Operator#examples
- Object literals: https://github.com/doug-moen/openscad2/blob/master/rfc/Objects.md#object-literals
- Dictionaries / associative arrays: https://github.com/openscad/openscad/issues/4203
- Function literals: https://github.com/openscad/openscad/issues/3088
- Module literals: https://github.com/openscad/openscad/pull/4478

However, these all appear to be stalled. So I wanted to ask:

1. Is there a way to implement this already? I'm okay with a complicated implementation, as long as it has the correct result while being easy to use.
2. Would there be interest in in specifically implementing this concept of a negative + positive object pair?

lgarron, 2024-12-21:
To illustrate the problem further, here's the "best" implementation I know of, for the sample code:

```scad
$fn = 64;
CLEARANCE = 0.5;

module hole(r, depth)
{
    // Hole
    translate([ 0, 0, -depth ]) cylinder(h = depth + CLEARANCE, r = r + CLEARANCE);
}

module bolt(r, depth)
{
    // Bolt
    color("gray") translate([ 0, 0, CLEARANCE ]) union()
    {
        scale([ 1, 1, 0.5 ]) cylinder(h = 2, r = r + 2, $fn = 6); // head
        translate([ 0, 0, -depth ]) cylinder(h = depth, r = r);   // stem
    }
}

union()
{
    difference()
    {
#color("white") translate([ 0, 0, -10 ]) cube([ 100, 100, 20 ], center = true);

        translate([ 0, 10, 0 ]) hole(r = 4, depth = 10);

        translate([ 40, 40 ]) hole(r = 2, depth = 10);
        translate([ -40, 40 ]) hole(r = 2, depth = 10);
        translate([ 30, -35 ]) hole(r = 3, depth = 10);
        translate([ -30, -35 ]) hole(r = 3, depth = 10);
    }

    translate([ 0, 10, 0 ]) bolt(r = 4, depth = 10);

    translate([ 40, 40 ]) bolt(r = 2, depth = 10);
    translate([ -40, 40 ]) bolt(r = 2, depth = 10);
    translate([ 30, -35 ]) bolt(r = 3, depth = 10);
    translate([ -30, -35 ]) bolt(r = 3, depth = 10);
}
```

As you can see, this results in significant amount of duplicated code that can get out of sync. With any non-trivial project, the repeated code blocks are unlikely to even fit on the same screen.
It would be possible to implement a version of `compose(…)` that alternatingly includes children in a difference and a union:

```scad
$fn = 64;
CLEARANCE = 0.5;

module hole(r, depth)
{
    // Hole
    translate([ 0, 0, -depth ]) cylinder(h = depth + CLEARANCE, r = r + CLEARANCE);
}

module bolt(r, depth)
{
    // Bolt
    color("gray") translate([ 0, 0, CLEARANCE ]) union()
    {
        scale([ 1, 1, 0.5 ]) cylinder(h = 2, r = r + 2, $fn = 6); // head
        translate([ 0, 0, -depth ]) cylinder(h = depth, r = r);   // stem
    }
}

module compose_alternating()
{
    union()
    {
        difference()
        {
            children(0);
            for (i = [1:2:$children - 1])
            {
                echo(jj = i);
                children(i);
            }
        }
        for (i = [2:2:$children - 1])
        {
            echo(ii = i);
            children(i);
        }
    }
}

compose_alternating()
{
#color("white") translate([ 0, 0, -10 ]) cube([ 100, 100, 20 ], center = true);

    translate([ 0, 10, 0 ]) hole(r = 4, depth = 10);
    translate([ 0, 10, 0 ]) bolt(r = 4, depth = 10);

    translate([ 40, 40 ]) hole(r = 2, depth = 10);
    translate([ 40, 40 ]) bolt(r = 2, depth = 10);
    translate([ -40, 40 ]) hole(r = 2, depth = 10);
    translate([ -40, 40 ]) bolt(r = 2, depth = 10);
    translate([ 30, -35 ]) hole(r = 3, depth = 10);
    translate([ 30, -35 ]) bolt(r = 3, depth = 10);
    translate([ -30, -35 ]) hole(r = 3, depth = 10);
    translate([ -30, -35 ]) bolt(r = 3, depth = 10);
}
```

I don't consider this a particularly great improvement, as debugging could get even hairier if you accidentally omit (e.g comment out) or include a single child instead of a pair.

(Also, I don't know a way to generalize either of these to a version that applies each difference and union for one child at a time in sequence, instead of in parallel.)

jordanbrown0, 2024-12-21:
BOSL2 does this with its `diff()` and `tag()` operations.
https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#module-diff

Here's a simple implementation of similar concepts.  Note that you need *three* distinct kinds of volumes:  you need "positive" volumes that you're going to subtract from, "negative" volumes that you're going to subtract, and "keep" volumes that you're *not* going to subtract from.  You don't want the holes subtracted away from the bold shafts.

```
CLEARANCE=0.1;

module compose() {
    difference() {
        children($mode="positive");
        children($mode="negative");
    }
    children($mode="keep");
}

module positive() {
    if ($mode == "positive") children();
}
module negative() {
    if ($mode == "negative") children();
}
module keep() {
    if ($mode == "keep") children();
}

module hole_and_bolt(r, depth) {
    // Hole ("negative" object)
    negative() translate([ 0, 0, -depth ]) cylinder(h = depth + CLEARANCE, r = r + CLEARANCE);

    // Bolt ("positive" object)
    keep() color("gray") translate([ 0, 0, CLEARANCE ]) union() {
        scale([ 1, 1, 0.5 ]) cylinder(h = 2, r = r + 2, $fn = 6); // head
        translate([ 0, 0, -depth ]) cylinder(h = depth, r = r);   // stem
    }
}

compose() {
    positive() color("white") translate([ 0, 0, -10 ]) cube([ 100, 100, 20 ], center = true);

    translate([ 0, 10, 0 ]) hole_and_bolt(r = 4, depth = 10);
    translate([ 40, 40 ]) hole_and_bolt(r = 2, depth = 10);
    translate([ -40, 40 ]) hole_and_bolt(r = 2, depth = 10);
    translate([ 30, -35 ]) hole_and_bolt(r = 3, depth = 10);
    translate([ -30, -35 ]) hole_and_bolt(r = 3, depth = 10);
}
```

A simpler scheme has `hold_and_bolt()` accepting a boolean argument that tells it whether to generate the hole or the bolt.

jordanbrown0, 2024-12-21:
For those following e-mail notifications:  I made small comment changes to the example I just posted.

lgarron, 2024-12-21:
> BOSL2 does this with its `diff()` and `tag()` operations.
> https://github.com/BelfrySCAD/BOSL2/wiki/attachments.scad#module-diff

Woah, that is cool, I will definitely give that a go. Thanks!

> A simpler scheme has `hold_and_bolt()` accepting a boolean argument that tells it whether to generate the hole or the bolt.

Yeah, this is actually my "preferred" approach, but it requires additional wrappers / duplicated code if you want to apply any transformations on top of the exact output of a module.

lgarron, 2024-12-21:
> Here's a simple implementation of similar concepts. Note that you need _three_ distinct kinds of volumes: you need "positive" volumes that you're going to subtract from, "negative" volumes that [...]

Alright, I was able to try this out, and it perfectly solves my problem. Thank you so much, @jordanbrown0! 🤩

I think it would be great to have a sufficiently powerful and ergonomic implementation of this in OpenSCAD, but I'm no longer blocked on it so I'll close this issue.

closed 2024-12-21 by lgarron

jordanbrown0, 2024-12-21:
Yes, dynamic scoping, combined with operators that look at the passed-down special variable to determine whether or not to include their children.

Note:  the children of compose() are executed three times with different settings of $mode.  You must ensure that they always check for $mode (via one of the three modules provided); if you fail to check then the object will be first added in, then subtracted away, and finally added in again.  (Which I guess makes it equivalent to keep(), but doing a lot more work.)

Note:  the implementation I gave above works in the development snapshot, but not in 2021.01.  There have been some changes to exactly how special-variable scoping works.  To make it work in 2021.01, replace the `children($mode==...)` invocations with statements like so:
```
        union() { $mode = "positive"; children(); }
```