openscad/openscad #5511: Draft new RELEASE_NOTES.md

kintel, 2024-12-20:
Update [RELEASE_NOTES.md](https://github.com/openscad/openscad/blob/master/RELEASE_NOTES.md) with all noteworthy changes since the 2021.01 release.
No need to populate bugfixes (for now).

This will serve as a good reminder of what we've accomplished since the last release, and a starting point for updating various docs, tests etc. to reflect these changes, in the hopes that some of this work could be farmed out a bit.

amatulic, 2025-01-07:
I'm trying (and failing) to find what snapshot textmetrics() was added, so I can do a version_num check on some code I'm writing that depends on it. Being able to look through a change log somewhere would be useful.

kintel, 2025-01-08:
We don't track exactly when features are introduced during development.
textmetrics() is also still experimental, but we aim for it to be added in the release described by these release notes.

amatulic, 2025-01-08:
That's fine, I got the approximate date of 19 August 2021 from when textmetrics() was first documented on the Wiki. Good enough for my needs. It's important to me to do version checks based on reasonably accurate version numbers, because it seems that the snapshots are more widely used nowadays than 2021.01, which is now four years old.

jordanbrown0, 2025-01-08:
textmetrics was merged in PR#3684 on 16 August 2021.

amatulic, 2025-01-08:
Ah, thanks. I couldn't find that PR. I changed my version check to 20210816.

jordanbrown0, 2025-01-08:
Remember that it is an experimental feature and so is only available if enabled.

PR #5514 (2025-03-12): Populated release notes for next release
Fixes #5511
closed 2025-03-12 by kintel