openscad/openscad #5513: Feature Request: Export Paths Into SVG

doakey3, 2024-12-22:
[image]

I want to produce an SVG like the one pictured above, including dashed lines in the attached image. Having the ability to use unclosed-paths is important for lasercutting. Something like this:

    layer(name="cut") {
        color("red")
        square([10, 10], attributes={"fill": "none", "stroke": "black", "stroke-width": "0.5"});
    }
    layer(name="engrave") {
        path("m 62,40 h 4", attributes={"stroke": "red", "stroke-width": "0.5"})
    }

Others have requested color and layer-separation export capability (see [issue 1364](https://github.com/openscad/openscad/issues/1364)) which would also be needed for this project.

What are the barriers to getting these capabilities in OpenSCAD? Is this something I could potentially work on and submit as a pull-request? Maybe I could pay one of the developers to implement this for me?

OpenSCAD is so widely used for 3D printing. I wish I could use it as my go-to for lasercut designs as well.

jordanbrown0, 2024-12-22:
See [OEP7: Mixed Dimension Geometry Support](https://github.com/openscad/openscad/wiki/OEP7%3A-Mixed-Dimension-Geometry-Support).

There are two major obstacles:
* Developer time and interest.  OpenSCAD developers all have lives and jobs, and so work on projects that strike their fancy.  I can't say that none are motivated by money, but I'm not aware of any who are.
* It's very very slow to make changes to the language, because projects developed for OpenSCAD have long lifespans and it's very unpopular when a change breaks something.  That means that we want to be very sure that a particular change is a good idea and we're not going to regret it in a few years.

doakey3, 2024-12-23:
Thank you for pointing me to OEP7! It’s an interesting and ambitious proposal, and it seems closely aligned with the feature I had in mind.

I’m curious: has OEP7 gained much traction or engagement from the developer community since it was proposed? If it’s been well-received and there’s still momentum, I’d be interested in seeing how I might contribute—whether it’s refining the ideas, testing, or even contributing code. Do you have any insights into the current status or next steps for moving it forward?

Thanks again for your time and guidance!

t-paul, 2024-12-23:
My interpretation is that it has support from other developers, it certainly has mine. Some details likely still need review, but that probably can only be done on the way testing while being implemented.

It should be possible to chip away some of the features that can stand on their own, like extending support for some mixed mode calculations or just looking at extending the 2D part.

I don't think it does solve the feature request for adding single lines. That is a difficult topic as it does conflict with the existing definition of OpenSCAD as a *solid* modeler.

jordanbrown0, 2024-12-24:
I read 1D-in-2D and 1D-in-3D as supporting lines and paths in 2- and 3-space.  (It's still a solid modeler... this just adds solid 1D models to the existing solid 2D and 3D models :-)  They wouldn't mean much for (existing) 3D export, just as 2D figures don't mean anything for 3D export.  One might imagine a 3D format that includes support for 1D and 2D objects; I don't know whether one exists.  Lines and paths could be straightforwardly exported into SVG.

Perhaps @thehans can comment on how much has been implemented.

closed 2026-05-08 by jordanbrown0