openscad/openscad #5520: linear_extrude: v= and segments=release ready

kintel, 2024-12-23:
Make the new v= and segments= arguments to `linear_extrude()` release ready:

* [ ] Final API validation
* [ ] Validate test coverage
* [ ] Manual entry
* [ ] Cheat sheet entry

eternal-sorrow, 2025-02-15:
Is it not available in 2021.01 release?

kintel, 2025-02-15:
No, these were added after the 2021.01 release.

jordanbrown0, 2026-10-14:
Either it's ready (good), or it's not (need to fix or back out).  Stalled/Idle-Closed isn't the right answer.

API looks sensible and seems to work.
There are several tests for v= but I don't find any for segments=.
There are manual entries for both. though it could use help.
Cheat sheet covers both.

kalj, 2026-11-07:
I've added tests for segments= (PR #7190): parameter validation, plus OBJ-export tests for straight extrusion, segments=0 with twist, segments= vs the $fs/$fa default and vs $fn, outlines that already have enough vertices, and holes.

One API note while testing: segments=-1 is silently clamped to 0, which disables subdivision entirely. That's consistent with slices= (both use the "silently clamp" helper), but a negative typo turning subdivision off without a warning might deserve a warning.

kintel, 2026-11-13:
Did anyone look at #7081 ?

kalj, 2026-11-13:
I've tested #7081 (comment there): it fixes `segments=` too, which currently gives fewer vertices than asked for whenever N doesn't fit the outline's equal edges (e.g. a square with `segments=6` gets 4).