openscad/openscad #5521: SVG import improvements: Release ready

kintel, 2024-12-23:
Make SVG import improvements release ready:
  * Support $fn, $fs, $fa
  * Added id= and layer= parameters to SVG import()

* [ ] Final API validation
* [ ] Validate test coverage
* [ ] Manual entry
* [ ] Cheat sheet entry

jordanbrown0, 2024-12-23:
We had a [mailing list comment (29 Oct 2024, 1500UTC(?))](https://lists.openscad.org/empathy/thread/QZWN3XAGNCKR5WGDIFD3ND2QBJNOCRCR?page=2) saying, perhaps with regard to this work:
>They're huge changes as far as I'm concerned - and for a different reason than the one in this thread.  OpenSCAD is currently importing curved edges in SVG images with hundreds of segments, which can't successfully be rendered.

su-sd, 2025-02-09:
The functionality works for me when exporting in the GUI, but when I use
`openscad -o file.stl file.scad` $fn, $fs and $fa do not work.

kintel, 2025-02-10:
@su-sd Can you provide a trivial example?

su-sd, 2025-02-10:
Sorry, the issue is in the GUI as well as the command-line.
If you remove the `$fn` in the import, it works as expected.
The preview is correct, the rendered part not.

Get file:
```
wget 'https://upload.wikimedia.org/wikipedia/en/0/01/Golden_State_Warriors_logo.svg'
```

scad file:
```
$fn = 256;

difference() {
        translate([0, 0, - 0.4])
                linear_extrude(0.8)
                        translate([79.8, 101.5, 0])
                                circle(d=220);
        linear_extrude(0.4) {
                import("Golden_State_Warriors_logo.svg", $fn=512);
        }
}
```

kintel, 2025-02-11:
Could you explain which problem you're encountering? If I do `import("Golden_State_Warriors_logo.svg", $fn=512);` I get far more faces than `import("Golden_State_Warriors_logo.svg", $fn=10);`

su-sd, 2025-02-11:
The `difference` does not work. The difference works, when I remove the `, $fn=512`.

kintel, 2025-02-11:
Your second linear_extrude height needs to be increase a bit to avoid coincident surfaces. Otherwise, everything works fine here.

su-sd, 2025-02-11:
The circle is from -0.4 until 0.4, the import is from 0 to 0.4.
It works fine, if I remove the `$fn=10` in the import (only in the rendered view, the preview is fine).
If there is a bug in my code, the result should be consistent whether the `$fn` is there or not.

I can make screenshots.

su-sd, 2025-02-11:
[image]
[image]

su-sd, 2025-02-12:
Ah, thank you, I did not know OpenSCAD has UB.

kalj, 2026-11-13:
Some input for "Validate test coverage" and "Final API validation":

- `id=` / `layer=`: covered by `svg/id-selection-test.scad` and `svg/id-layer-selection-test.scad` (incl. unknown id/layer).
- `$fn`/`$fs`/`$fa`: no test passes them to an SVG `import()`.

I checked how they behave, with vertices per outline for a test SVG on current master:

| shape | default | `$fn=6` | `$fn=100` | `$fs=0.5, $fa=2` |
|---|---|---|---|---|
| `<circle>` / `<ellipse>` | 40 | 40 | 100 | 126 |
| arc (`A`) | 16 | 15 | 51 | 64 |
| cubic Bézier (`C`) | 21 | 21 | 101 | 21 |

Two API points:
1. Circles/ellipses have a floor of 40 segments (`shape.cc`) and Bézier curves a floor of 20 (`path.cc`), so these can only increase the resolution, e.g. `$fn=6` doesn't give a hexagon. I guess that's intentional to keep old output unchanged, but it may be worth documenting.
2. Bézier curves ignore `$fs`/`$fa` (`getPathSegmentCount()` only uses `$fn`), while circles, ellipses and arcs use all three.

Happy to add tests once it's clear which behaviour is intended.

kintel, 2026-11-14:
Since we didn't have a full release since these features were introduced, we don't really need to manage backwards compatibility, so I say let's explore the two API points

kalj, 2026-11-15:
I made a prototype of this: https://github.com/kalj/openscad/tree/svg-tessellation ([diff](https://github.com/openscad/openscad/compare/master...kalj:openscad:svg-tessellation))

One more finding first. `$fs` (and the radius used for `$fn`/`$fa`/`$fs`) was applied in SVG user units, before the viewBox/dpi scale and the transforms. So the result depended on the file's internal units. In `spec-shapes-circle01.svg` one unit is 0.1 mm, so "`$fs=8` (79 segs)" in the new test is really 2π·100/8.

The prototype:
- Computes the segment count from the curve's size in mm, including the page scale and all `transform`s of the shape and its parents.
- Drops the 40/20 minimums, so imported curves match `circle()`, and `$fn=6` gives a hexagon.
- Makes Béziers honour `$fa`/`$fs`. A Bézier is treated like an arc with the same length and turning angle (both estimated from the control polygon), so a quarter-circle Bézier gets the same count as the matching arc. `$fn` then means segments per full turn, as for arcs, instead of per curve.

Vertex counts for the same 10 mm shapes, written in mm units / 0.1 mm units / mm units inside `scale(10)`:

| | `circle(r=10)` | SVG circle, master | SVG circle, branch | quarter Bézier, master | quarter Bézier, branch | quarter `circle(r=10)` |
|---|---|---|---|---|---|---|
| default | 30 | 40 / 40 / 40 | 30 | 22 | 10 | 10 |
| `$fn=6` | 6 | 40 / 40 / 40 | 6 | 22 | 4 | 4 |
| `$fa=1,$fs=0.5` | 126 | 126 / 360 / 40 | 126 | 22 | 34 | 34 |
| `$fa=0.1,$fs=1` | 63 | 63 / 629 / 40 | 63 | 22 | 18 | 18 |

This changes 47 expected outputs, including those from #7220. The bounding boxes of all test SVGs are unchanged; only the segment counts differ. Not handled yet: scaling on a `<use>` element (the referenced shapes are tessellated where they are defined), and non-uniform scale uses the largest factor.

If this is the direction you want, I can open it as a PR.

kintel, 2026-11-15:
Actually, I just realized one complication: In OpenSCAD-2021.01, we could import an SVG _without_ specifying $f* modifiers. If we respect the defaults going forward, we'll break existing designs. Ideally we'd change behavior only for cases where $f* was actually specified.
This could be tricky, and needs some thinking or special-case handling inside OpenSCAD.

kalj, 2026-11-15:
Good point. I've pushed a second commit to the [branch](https://github.com/openscad/openscad/compare/master...kalj:openscad:svg-tessellation): when `$fn`, `$fa` and `$fs` all have their default values, SVG import uses exactly the 2021.01 segment counts (40 per circle/ellipse, 20 per Bézier, |angle|/18+4 per arc). As soon as any of them is set, the new mm-based counts are used.

A few things to note:
- OpenSCAD can't tell `$fa=12` written explicitly from the default, so "specified" means "differs from the default". That seems harmless, since such designs just keep the 2021.01 result.
- Designs that set `$fa`/`$fs`/`$fn` globally (e.g. `$fa=1; $fs=0.4;` at the top of the file) will get different tessellation of imported SVGs than in 2021.01. Usually finer, but different. I don't see a way around that without a separate parameter on `import()`.
- The current master/RC0 arc code already differed from 2021.01 at defaults (it uses the larger of `$fn/$fa/$fs` and the old formula). This restores the 2021.01 count.

With this, only 10 expected outputs change: the new tessellation test and `spec-paths-arcs01/02`.

kintel, 2026-11-15:
> * OpenSCAD can't tell `$fa=12` written explicitly from the default, so "specified" means "differs from the default". That seems harmless, since such designs just keep the 2021.01 result.

Yeah, that's the direction I'm thinking too modulo a discontinuity in values.

> * Designs that set `$fa`/`$fs`/`$fn` globally (e.g. `$fa=1; $fs=0.4;` at the top of the file) will get different tessellation of imported SVGs than in 2021.01. Usually finer, but different. I don't see a way around that without a separate parameter on `import()`.

Caveat: When exporting to .csg, we write current values explicitly. This would make importing an exported design render differently, which is quastionable. Not sure how to manage this though..

> * The current master/RC0 arc code already differed from 2021.01 at defaults (it uses the larger of `$fn/$fa/$fs` and the old formula). This restores the 2021.01 count.
> 
Yeah, but that's the name of the game. Let's focus on preserving compatibility with 2021.01 designs, and disregard compatibility of designs made using a development snapshot if necessary.

kalj, 2026-11-15:
On `.csg`: "specified" is decided by value, not by whether `$f*` is written, so an exported `$fn = 0, $fa = 12, $fs = 2` still gives the 2021.01 result. I checked that `.scad` and exported `.csg` render the same.

On the discontinuity: I pushed a fix. When `$fn` isn't set, the count never goes below the 2021.01 count, so `$fa`/`$fs` can only refine. `$fn` is still exact.

jordanbrown0, 2026-11-15:
I would say that the exact behavior with SVG has always been murky, and so we should not obsess over compatibility at the expense of current clarity.

Ideally, if you import a SVG circle, the $fX would apply exactly as they do to OpenSCAD circles of the same size (as imported).  For an SVG Bézier curve, they should do something reasonable but we should explicitly disclaim any particular behavior.

See also #6059 for my opinions on how we should evolve the meaning of $fa, $fs, and $fe.  ($fn must continue to generate n-gons.)

kintel, 2026-11-15:
I generally agree. I just have an aversion towards breaking people's designs, so trying to find a middle ground.

One ideal, clean solution could be to offer a setting to "interpret like OpenSCAD-YYYY.MM" and move the current version forward without too much legacy baggage, but that's a larger change.

We haven't really discussed whether $fe can make it into the next release. That could be something to work on after RC1.

jordanbrown0, 2026-11-15:
I would say that, especially here where behavior has been murky, yielding more precise results in a future release is not breakage. Perhaps we should guarantee "at least as good as the old release" results. (Though for the special case of SVG circles, I'd say that any mismatch with OpenSCAD circles has always been a bug.)

jordanbrown0, 2026-11-15:
All uses of $fX are based on value, not on whether they were specified.  It must be that way, because they are always specified - their values are set before the program runs.

kintel, 2026-11-15:
Challenge with SVG is that resolution was already way too high in some cases, especially for bezier curves, and I assume the $f* support wes introduced partially to be able to curb that a bit.

jordanbrown0, 2026-11-15:
"Stupidly high resolution" is indeed a problem.

The math is beyond me, but I intuitively suspect that Bézier curves will always have cases where the results are stupidly high (and maybe cases where they are stupidly low), because handling the resolution settings anything like "correctly" requires stepping along the Bézier path rather than stepping the inputs to the Bézier function, and that's a lot harder.  You can have a single Bézier curve that is both very tight (and so needs lots of segments) and nearly straight (and so needs very few segments.)

```
include <BOSL2/std.scad>
bez = [[0,0], [-30,10], [60,0], [80,0]];
debug_bezier(bez, N=len(bez)-1);
```

[image]

jordanbrown0, 2026-11-15:
But as I think about it more, I suspect that stepping along the input actually produces a desirable result, if you could figure out the right number of steps in the first place.  You *want* lots of points at the left side of the curve above, and you *want* few points at the right side, and I think the equation does that automatically... but how to relate that behavior to $fa and $fs is not at all clear.

Anyhow, net, I think that the best that you can do for Bézier curves is to have $fa and $fs yield results that are at least sort of similar to their results on circles.

$fn could be well-defined:  the number of points in the curve.  But I suspect that that would not be a useful definition for real SVG files.  Real SVG files may have very curvy individual curves that need lots of points, and small or straightish individual curves that need very few points.  Trying to apply a single value of $fn to a non-trivial drawing will yield problems.

kintel, 2026-11-16:
In an attempt to (overly?) structure this, here is a test grid we could use to point out behaviors we like/don't like:

https://gist.github.com/kintel/4c80fd8b69e5bb1dde400e219a6e9722

jordanbrown0, 2026-11-16:
It is unfortunate that SVG circles (as imported) do not just follow OpenSCAD circle behavior.  (Ellipses are trickier, but I would treat them as circles with the circle radius equal to the ellipse major radius.)

It wouldn't be silly to fix them so that that's true, but in theory at least that could mean models could have lower-resolution circles.