openscad/openscad #5524: cdc33d4f81 breaks export buttons

smartavionics, 2024-12-23:
**Describe the bug**
When commit cdc33d4f81 is included, file export menu actions and buttons no longer do anything.

It's working OK up until the previous commit (1ad142a441).

**Expected behavior**
Pressing the STL button should pop up a file save dialog.

**Code reproducing the issue**
If applicable, add trimmed down, self-contained code demonstrating the issue.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment and Version info (please complete the following information):**
```
System:
  Host: toucan Kernel: 5.15.0-126-generic x86_64 bits: 64 
  Desktop: Xfce 4.16.1 Distro: Ubuntu 20.04.6 LTS (Focal Fossa) 
Machine:
  Type: Laptop System: GIGABYTE product: GB-BSi3-6100 v: 1.x 
  serial: <superuser/root required> 
  Mobo: GIGABYTE model: MKLP3AP-00 v: 1.x serial: <superuser/root required> 
  UEFI [Legacy]: American Megatrends v: F5 date: 01/29/2016 
CPU:
  Topology: Dual Core model: Intel Core i3-6100U bits: 64 type: MT MCP 
  L2 cache: 3072 KiB 
  Speed: 2167 MHz min/max: 400/2300 MHz Core speeds (MHz): 1: 2167 2: 2000 
  3: 2045 4: 1735 
Graphics:
  Device-1: Intel Skylake GT2 [HD Graphics 520] driver: i915 v: kernel 
  Display: x11 server: X.Org 1.20.13 driver: modesetting unloaded: fbdev,vesa 
  resolution: 1920x1080~60Hz 
  OpenGL: renderer: Mesa DRI Intel HD Graphics 520 (SKL GT2) 
  v: 4.6 Mesa 21.2.6 
Audio:
  Device-1: Intel Sunrise Point-LP HD Audio driver: snd_hda_intel 
  Device-2: Logitech HD Pro Webcam C920 type: USB 
  driver: snd-usb-audio,uvcvideo 
  Sound Server: ALSA v: k5.15.0-126-generic 
Network:
  Device-1: Intel Ethernet I219-V driver: e1000e 
  IF: enp0s31f6 state: up speed: 1000 Mbps duplex: full 
  mac: 40:8d:5c:f4:c2:44 
  Device-2: Intel Wireless 3165 driver: iwlwifi 
  IF: wlp1s0 state: down mac: ac:2b:6e:15:50:54 
Drives:
  Local Storage: total: 1.82 TiB used: 1.02 TiB (56.2%) 
  ID-1: /dev/sda vendor: Samsung model: SSD 870 EVO 1TB size: 931.51 GiB 
  ID-2: /dev/sdb type: USB vendor: Western Digital model: WD Elements 2621 
  size: 931.48 GiB 
Partition:
  ID-1: / size: 915.32 GiB used: 291.14 GiB (31.8%) fs: ext4 dev: /dev/sda5 
Sensors:
  System Temperatures: cpu: 42.0 C mobo: 29.8 C 
  Fan Speeds (RPM): N/A 
Info:
  Processes: 271 Uptime: 14d 5h 34m Memory: 15.49 GiB used: 7.18 GiB (46.4%) 
  Shell: bash inxi: 3.0.38 
```

Here's CMakeCache.txt, maybe that's useful?

[CMakeCache.txt](https://github.com/user-attachments/files/18230561/CMakeCache.txt)

kintel, 2024-12-24:
Works fine here. Does this happen only when you build from source, or also when using our prebuilt development snapshots?

smartavionics, 2024-12-25:
I haven't tried the snapshots, I'm building from source.

smartavionics, 2024-12-25:
I just tried completely deleting everything in my build directory and then running cmake and make again to ensure the build was not using any stale state but it made no difference, the export buttons and menu items are still non-functional.

LocoDelAssembly, 2024-12-26:
I've tried snap 0+git.28c540500 and also found out I cannot export to STL, nothing happens when pressing F7 nor clicking the STL button. Works fine with "stable" (0+git.dbd7d4f3d)

I'm using Ubuntu 20.04 and installed from "Ubuntu Software" openscad-nightly which is: https://snapcraft.io/openscad-nightly

smartavionics, 2024-12-26:
I'm wondering if it's caused by the version of Qt or some other library that is supplied with Ubuntu 20.04.

Here's my library info...

```
OpenSCAD Version: 2024.12.25
System information: Linux 5.15.0-126-generic #136~20.04.1-Ubuntu SMP Thu Nov 14 16:38:05 UTC 2024 x86_64 Ubuntu 20.04.6 LTS 4 CPUs 15.49 GB RAM
User Agent: OpenSCAD/2024.12.25 (Linux x86_64; Ubuntu 20.04.6 LTS)
Compiler: GCC "9.4.0" 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.0.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.4.2
Manifold version: 3.0.0
Qt version: 5.12.8
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20230410
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.6.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof*, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /usr/local/bin
Documents Path: /home/markb/.local/share
User Documents Path: /home/markb/Documents
Resource Path: /usr/local/share/openscad
User Library Path: /home/markb/.local/share/OpenSCAD/libraries
User Config Path: /home/markb/.config/OpenSCAD
Backup Path: /home/markb/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/markb/.local/share/OpenSCAD/libraries
/usr/local/share/openscad/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/markb/.local/share/fonts
/home/markb/.fonts
/usr/share/texmf/fonts/opentype/public/lm
/usr/share/texmf/fonts/opentype/public/lm-math
/usr/share/texmf/fonts/opentype/public/tex-gyre
/usr/share/texmf/fonts/opentype/public/tex-gyre-math
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
/home/markb/Library/Fonts
/usr/share/fonts/X11
/usr/share/fonts/cMap
/usr/share/fonts/cmap
/usr/share/fonts/opentype
/usr/share/fonts/truetype
/usr/share/fonts/type1
/home/markb/.fonts/Library
/home/markb/.fonts/noto
/home/markb/.fonts/r
/usr/share/fonts/X11/Type1
/usr/share/fonts/X11/encodings
/usr/share/fonts/X11/misc
/usr/share/fonts/X11/util
/usr/share/fonts/cmap/adobe-cns1
/usr/share/fonts/cmap/adobe-gb1
/usr/share/fonts/cmap/adobe-japan1
[... 334 lines cut ...]
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_compute_shader_derivatives
GL_NV_conditional_render
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_fragment_shader_interlock
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
```

kintel, 2024-12-26:
Oh, yeah. I’ve only tested with Qt 6, and there were some Qt run-time issues related to this earlier. I’ll take another look at that code.

LocoDelAssembly, 2024-12-27:
I've just tried Ubuntu 22.04 and still same issue with F7/STL button.

Sorry for off-topic, but how do you enable Manifold in recent OpenSCAD versions without using `--backend=manifold` in command line? In earlier versions it was available via UI in Preferences > Features, but now it doesn't appear and renders are done with CGAL. Is this behaviour unique to the snap version?

[image]

Here my `--info` and `lsb_release`:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
$ openscad-nightly --info
QObject::startTimer: Timers can only be used with threads started with QThread
OpenSCAD Version: 2024.12.26
System information: Linux 6.8.0-49-generic #49~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov  6 17:42:15 UTC 2 x86_64 Ubuntu Core 20 16 CPUs 62.64 GB RAM
User Agent: OpenSCAD/2024.12.26 (Linux x86_64; Ubuntu Core 20)
Compiler: Clang "10.0.0 " 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Manifold version: 3.0.0
Qt version: 5.12.8
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20230410
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.6.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof, input-driver-dbus, lazy-union, vertex-object-renderers-indexing, textmetrics, import-function, predictible-output
Application Path: /snap/openscad-nightly/4827/usr/bin
Documents Path: /home/hernan/snap/openscad-nightly/4827/.local/share
User Documents Path: /home/hernan/snap/openscad-nightly/4827
Resource Path: /snap/openscad-nightly/4827/usr/share/openscad-nightly
User Library Path: /home/hernan/snap/openscad-nightly/4827/.local/share/OpenSCAD/libraries
User Config Path: /home/hernan/snap/openscad-nightly/4827/.config/OpenSCAD
Backup Path: /home/hernan/snap/openscad-nightly/4827/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
  /home/hernan/snap/openscad-nightly/4827/.local/share/OpenSCAD/libraries
  /snap/openscad-nightly/4827/usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
  /usr/share/fonts
  /usr/local/share/fonts
  /home/hernan/snap/openscad-nightly/4827/.local/share/fonts
  /home/hernan/snap/openscad-nightly/4827/.fonts
  /snap/openscad-nightly/4827/usr/share/fonts
  /usr/X11/lib/X11/fonts
  /System/Library/Fonts
  /Library/Fonts
  /home/hernan/snap/openscad-nightly/4827/Library/Fonts
  /usr/share/fonts/X11
  /usr/share/fonts/cMap
  /usr/share/fonts/cmap
  /usr/share/fonts/opentype
  /usr/share/fonts/truetype
  /usr/share/fonts/type1
  /snap/openscad-nightly/4827/usr/share/fonts/truetype
  /usr/share/fonts/X11/Type1
  /usr/share/fonts/X11/encodings
  /usr/share/fonts/X11/misc
  /usr/share/fonts/X11/util
  /usr/share/fonts/cmap/adobe-cns1
  /usr/share/fonts/cmap/adobe-gb1
  /usr/share/fonts/cmap/adobe-japan1
  /usr/share/fonts/cmap/adobe-japan2
  /usr/share/fonts/cmap/adobe-korea1
  /usr/share/fonts/opentype/malayalam
  /usr/share/fonts/opentype/noto
  /usr/share/fonts/opentype/urw-base35
  /usr/share/fonts/truetype/Gargi
  /usr/share/fonts/truetype/Gubbi
  /usr/share/fonts/truetype/Nakula
  /usr/share/fonts/truetype/Navilu
  /usr/share/fonts/truetype/Sahadeva
  /usr/share/fonts/truetype/Sarai
  /usr/share/fonts/truetype/abyssinica
  /usr/share/fonts/truetype/arphic
  /usr/share/fonts/truetype/dejavu
  /usr/share/fonts/truetype/droid
  /usr/share/fonts/truetype/fonts-beng-extra
  /usr/share/fonts/truetype/fonts-deva-extra
  /usr/share/fonts/truetype/fonts-gujr-extra
  /usr/share/fonts/truetype/fonts-guru-extra
  /usr/share/fonts/truetype/fonts-kalapi
  /usr/share/fonts/truetype/fonts-orya-extra
  /usr/share/fonts/truetype/fonts-telu-extra
  /usr/share/fonts/truetype/fonts-yrsa-rasa
  /usr/share/fonts/truetype/freefont
  /usr/share/fonts/truetype/kacst
  /usr/share/fonts/truetype/kacst-one
  /usr/share/fonts/truetype/lao
  /usr/share/fonts/truetype/liberation
  /usr/share/fonts/truetype/liberation2
  /usr/share/fonts/truetype/libreoffice
  /usr/share/fonts/truetype/lohit-assamese
  /usr/share/fonts/truetype/lohit-bengali
  /usr/share/fonts/truetype/lohit-devanagari
  /usr/share/fonts/truetype/lohit-gujarati
  /usr/share/fonts/truetype/lohit-kannada
  /usr/share/fonts/truetype/lohit-malayalam
  /usr/share/fonts/truetype/lohit-oriya
  /usr/share/fonts/truetype/lohit-punjabi
  /usr/share/fonts/truetype/lohit-tamil
  /usr/share/fonts/truetype/lohit-tamil-classical
  /usr/share/fonts/truetype/lohit-telugu
  /usr/share/fonts/truetype/lyx
  /usr/share/fonts/truetype/malayalam
  /usr/share/fonts/truetype/noto
  /usr/share/fonts/truetype/padauk
  /usr/share/fonts/truetype/pagul
  /usr/share/fonts/truetype/samyak
  /usr/share/fonts/truetype/samyak-fonts
  /usr/share/fonts/truetype/sinhala
  /usr/share/fonts/truetype/teluguvijayam
  /usr/share/fonts/truetype/tibetan-machine
  /usr/share/fonts/truetype/tlwg
  /usr/share/fonts/truetype/ttf-khmeros-core
  /usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/type1/gsfonts
  /usr/share/fonts/type1/urw-base35
  /snap/openscad-nightly/4827/usr/share/fonts/truetype/dejavu
  /snap/openscad-nightly/4827/usr/share/fonts/truetype/ubuntu
  /usr/share/fonts/X11/encodings/large

GL context creator: GLX (new)
GLX version: 1.4
PNG generator: lodepng

GLEW version: 2.1.0
OpenGL Version: 4.6 (Compatibility Profile) Mesa 21.2.6
GL Renderer: Mesa Intel(R) UHD Graphics 630 (CFL GT2)
GL Vendor: Intel
RGBA(8888), depth(24), stencil(8)
GL_ARB_framebuffer_object: yes
GL_EXT_framebuffer_object: yes
GL_EXT_packed_depth_stencil: yes
```

nophead, 2024-12-27:
Manifold is now the Preferences Advanced Tab
[image]

kintel, 2024-12-29:
Update: Also works on Qt5 on macOS

kintel, 2024-12-29:
@LocoDelAssembly Does export from the File menu also fail for you, or just the STL/F7 button?

closed 2024-12-29 by kintel

LocoDelAssembly, 2024-12-29:
Doesn't work either:

[Screencast from 29-12-24 19:14:37.webm](https://github.com/user-attachments/assets/790aac7a-e79b-41ff-99b0-d5ef31e245b7)

LocoDelAssembly, 2024-12-29:
Sorry, forgot to add this. It appears right after the app is opened. No message is output when I click the button, press `F7` or use the menu to export on any format (also nothing happens on GUI-side in all 3 methods of generating the export file).

Version used: `2024.12.29` 

```
$ openscad-nightly 
Qt: Session management error: Could not open network socket
propsReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.1300\" (uid=1000 pid=89005 comm=\"/snap/openscad-nightly/4843/usr/bin/openscad-night\" label=\"snap.openscad-nightly.openscad-nightly (enforce)\") interface=\"org.freedesktop.DBus.Properties\" member=\"GetAll\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1110 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
nmReply "An AppArmor policy prevents this sender from sending this message to this recipient; type=\"method_call\", sender=\":1.1300\" (uid=1000 pid=89005 comm=\"/snap/openscad-nightly/4843/usr/bin/openscad-night\" label=\"snap.openscad-nightly.openscad-nightly (enforce)\") interface=\"org.freedesktop.NetworkManager\" member=\"GetDevices\" error name=\"(unset)\" requested_reply=\"0\" destination=\"org.freedesktop.NetworkManager\" (uid=0 pid=1110 comm=\"/usr/sbin/NetworkManager --no-daemon \" label=\"unconfined\")"
"Object path cannot be empty"
QMetaObject::connectSlotsByName: No matching signal for on_syntaxHighlight_textActivated(QString)
QObject::connect: No such signal QSignalMapper::mappedInt(int)
QObject::connect:  (receiver name: 'MainWindow')
```

kintel, 2024-12-29:
@LocoDelAssembly The fix probably won't make it until tomorrow's nightly build.

t-paul, 2024-12-30:
Fix confirmed via Snap package.

LocoDelAssembly, 2024-12-30:
Also works for me now (`0+git.9454b4315`). Thanks!

PS: 
> @LocoDelAssembly The fix probably won't make it until tomorrow's nightly build.

When replied I failed to notice the issue was closed with fix. Sorry, I was intending to answer to your question about file menu actually  :sweat_smile:

smartavionics, 2024-12-30:
Just built from master branch. All good again, thanks.

PR #5545 (2024-12-29): Don't use the mappedInt signal before Qt-5.15
Fixes #5524