openscad/openscad #5525: `ERROR: SSL handshake failed` along with other ssl-related errors

elliotf, 2024-12-23:
First, thank you so much for OpenSCAD!

**Describe the bug**
Upon starting OpenSCAD, I get a number of messages relate to SSL:

```
$ ./OpenSCAD-2024.12.22.ai21806-x86_64.AppImage
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
ERROR: SSL handshake failed
```

This may be innocuous, but OpenSCAD hasn't had this behavior in the past. The prior two versions I was using were `OpenSCAD-2023.01.15.ai13503-x86_64.AppImage` and `OpenSCAD-2023.03.24.ai14242-x86_64.AppImage` which are (admittedly) very old but do not exhibit this error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://openscad.org/downloads.html
2. Click on 'Linux - AppImage' under 'Development Snapshots'
3. Make .AppImage file executable
4. Run .AppImage file
5. See unexpected SSL-related output

**Expected behavior**
Expect to see OpenSCAD open and run without SSL-related errors

**Code reproducing the issue**
n/a, I think?

**Screenshots**
n/a, text output is above.

**Environment and Version info (please complete the following information):**
* Ubuntu 22.04.5 LTS
* Intel 9700K with Radeon 6950xt
* Nightly OpenSCAD releases `OpenSCAD-2024.12.22.ai21806-x86_64.AppImage` and `OpenSCAD-2024.12.17.ai21648-x86_64.AppImage` downloaded from https://openscad.org/downloads.html

**Library & Graphics card information**
n/a, I think.

**Additional context**
Add any other context about the problem here.

jordanbrown0, 2024-12-28:
Those look like an OpenSSL library version problem. **Possibly** the Qt is built for OpenSSL 1.1.x and the system has OpenSSL 3.x.

For those who might be wondering (as I was) why there's a network connection:  OpenSCAD phones home to retrieve a list of print services and their associated metadata.

kintel, 2024-12-28:
@jordanbrown0 In terms of network connectivity: On macOS, there is also an software update check, but that's not relevant in this case.

felsgaertner, 2025-01-08:
> For those who might be wondering (as I was) why there's a network connection: OpenSCAD phones home to retrieve a list of print services and their associated metadata.

Is there a way to disable this network communication? At least I don't need it, as I use it only locally for rendering to STL.

t-paul, 2025-01-08:
@felsgaertner Unrelated to that issue, but I agree it should not a) do it on startup and b) have an option to disable (I think there's none right now). Please create a separate issue for this. I'll mark this off-topic later.

gsohler, 2025-02-05:
The Callstack to the TLS issue is:
MainWindow::MainWindow()
PrintInitDialog::PrintInitDialog()
PrintService::getPrintServices()
PrintService::createPrintServices()
Qt::NetworkRequest ( https://app.openscad.org/print-service.json )

could not see the option to turn off the network request. is there one ?

jordanbrown0, 2025-02-05:
Pull request #5600 .

jordanbrown0, 2026-05-08:
Not duplicated on Ubuntu 22.04.5 LTS with OpenSCAD 2026.04.26.

Retest, please.