openscad/openscad #5532: Manifold conversion failed on simple part: notManifold

Gus-the-looper, 2024-12-25:
I am using the latest nightly. I am getting a manifold error on a simple object. Expected behaviour: nor error ;)

I have attached a file to reproduce the error. Enable manifold in the options and render it.
You will receive

WARNING: PolySet -> Manifold conversion failed: NotManifold
Trying to repair and reconstruct mesh..

Running on Ubuntu 24.04.1 LTS
OpenSCAD 2024.12.24.ai21853 (git 481186b9f)
NVIDIA GeForce RTX 3060/PCIe/SSE2
[bug_report.txt](https://github.com/user-attachments/files/18245553/bug_report.txt)

nophead, 2024-12-25:
I get the same error on a very large project. Hard to know where it is with no line number info.

kintel, 2024-12-25:
Duplicate of #5135

closed 2024-12-25 by kintel