openscad/openscad #5536: text(..., font=":charset=junk") crashes OpenSCAD

jordanbrown0, 2024-12-27:
**Describe the bug**
Setting "charset" in a font specification crashes OpenSCAD.

**To Reproduce**
```
text("hello", font=":charset=foo");
```
F5, OpenSCAD exits.
Note that the colon is a critical part of the specification.

**Expected behavior**
Doesn't exit :-)

I believe that fontconfig's usual behavior would be to ignore an unknown specifications.

I'm not sure from the fontconfig docs exactly what this property does, but exiting isn't it.

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: 13th Gen Intel(R) Core(TM) i7-13700   2.10 GHz
 - OpenSCAD Version: 2024.12.26, development snapshot (and in 2021.01)

**Library & Graphics card information**
fontconfig version: 2.14.2

**Additional context**
Also fails for `textmetrics()` and `fontmetrics()`.

kintel, 2024-12-30:
Can reproduce on macOS too

kintel, 2024-12-30:
Attempted fix in #5552 - let me know if you think we should fail in a different fashion, or if you have more torture tests lying around :)

kintel, 2024-12-30:
I also added a test case showing reasonable use of charset=: https://github.com/openscad/openscad/blob/8a4ab31ad240192d8e84619b67258700ff2a2aeb/tests/data/scad/2D/features/text-font-charset.scad

kintel, 2024-12-30:
Btw., an interesting use of `textmetrics` could be to look up a font supporting particular characters you want to use, then use that font to render text.

t-paul, 2024-12-30:
If you don't care which font, then just `text("\U01F600", font=":charset=1f600");` will do that.

jordanbrown0, 2024-12-30:
Reading the fontconfig docs I couldn't figure out what the intended argument was.  Thanks.

You would use fontmetrics(), not textmetrics(), but yes, interesting. Actually, you don't need to use either metrics() function; just give the pattern directly to text(). (And of course that is exactly the intent of the fontconfig matching scheme.)

jordanbrown0, 2024-12-30:
But hmm... using fontmetrics() instead of passing directly to text() lets you pick one font that will handle all of the code points you use, even when they are split across multiple text() calls.

t-paul, 2024-12-30:
True, but that really would be better handled in a layout library like pango.

PR #5552 (2025-01-03): Catch failure to parse font
Also adds charset= test

Fixes #5536
closed 2025-01-03 by kintel