openscad/openscad #5547: Convert old Qt signal connection scheme to the typesafe new one

kintel, 2024-12-29:
Qt5 introduced a more modern way of connecting signals to functions: https://wiki.qt.io/New_Signal_Slot_Syntax

The new way offers the benefit of being able to perform checks at compile-time for whether the signal and slot exists and that their parameters are correct.

TODO: Convert all manual `connect()` invocations to use the new style.

akashmahesh2805, 2025-01-14:
Hi, this seems like a good first issue for me to start contributing! Could you please assign it to me?

scat25, 2025-03-03:
Hi, I would like to fix this issue. The existing pr is stalled, is it ok if I create a new pr for this?

t-paul, 2025-03-03:
@scat25 Sure, you can create a new PR. We can clean-up open PRs once something is merged.

PR #5726 (2025-03-06): Upgraded old string-based connects to new functor-based syntax
I replaced all the old string-based connects with the new functor-based syntax.
I made some small changes in some Interfaces to achieve this, see the changed header files. Overloaded functions were resolved with QOverload where applicable.

The code compiles and I can run OpenSCAD.

This fixes #5547
closed 2025-03-06 by kintel