openscad/openscad #5551: Pasting a name with umlaut like ü into save dialog name field crashes immediately

fah, 2024-12-30:
**Describe the bug**
I copied a German umlaut from folder name to name a new file. Pressed Apple+s for save then Apple+v to paste the name. -> crash

**To Reproduce**
Steps to reproduce the behavior:
See above
OR
Add an "ü" into the clipboard (Apple+c). Go to "save as ... " dialog paste the clipboard into the name field.

**Expected behavior**
No crash

**Code reproducing the issue**
n/a

**Screenshots**
n/a

**Environment and Version info (please complete the following information):**
 - OS: Mac 13.6.4, German environment
 - System: Intel
 - OpenSCAD Version 2021.01 via official dmg

kintel, 2024-12-30:
Is this fixed in the latest development snapshot?

kintel, 2025-04-13:
Assuming fixed

closed 2025-04-13 by kintel