openscad/openscad #5554: Issues with offset() in the nightly build

50an6xy06r6n, 2025-01-02:
**Describe the bug**
After updating my nightly build, I started noticing some strange geometry bugs around 2-D offset operations. In particular, some large offsets create strange pointy artifacts either inwards or outwards. I'm able to reproduce the outward pointiness, but the inward pointiness went away randomly as I was playing around with it. I cleared the caches and that didn't seem to affect it at all. I was previously running the nightly from about a week ago, and it didn't have this issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone this branch of my repo: https://github.com/50an6xy06r6n/hotswap_pcb_generator/tree/openscad_bug_repro
2. Open `scad/plate.scad`
3. Compare the results from the nightly build to the results from the 2021.01 release

**Expected behavior**
[image]

**Code reproducing the issue**
Couldn't get an isolated reproduction, and the behavior was somewhat inconsistent

**Screenshots**
[image]

**Environment and Version info (please complete the following information):**
 - OS: Ubuntu 2024.04.3
 - System: Virtualbox VM on a 64-bit Windows host
 - OpenSCAD Version: 2024.12.30, installed via snap

**Library & Graphics card information**
If applicable, add the content of the Help->Library Info window, or the output of running the `openscad --info` command.
OpenSCAD Version: 2024.12.30
System information: Linux 6.5.0-18-generic #18~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Feb 7 11:40:03 UTC 2 x86_64 Ubuntu Core 20 8 CPUs 31.34 GB RAM
User Agent: OpenSCAD/2024.12.30 (Linux x86_64; Ubuntu Core 20)
Compiler: Clang "10.0.0 " 64bit
MinGW build: No
Debug build: No
Boost version: 1_71
Eigen version: 3.3.7
CGAL version, kernels: 5.5.2, Cartesian<Gmpq>, Extended_cartesian<Gmpq>, Epeck
OpenCSG version: OpenCSG 1.5.1
Manifold version: 3.0.1
Qt version: 5.12.8
QScintilla version: 2.11.2
InputDrivers: 
GLib version: 2.64.6
lodepng version: 20230410
libzip version: 1.5.1
fontconfig version: 2.13.1
freetype version: 2.10.1
harfbuzz version: 2.6.4
cairo version: 1.16.0
lib3mf version: 1.8.1
Features: roof*, input-driver-dbus, lazy-union, vertex-object-renderers-indexing*, textmetrics*, import-function*, predictible-output
Application Path: /snap/openscad-nightly/4859/usr/bin
Documents Path: /home/gareth/snap/openscad-nightly/4859/.local/share
User Documents Path: /home/gareth/snap/openscad-nightly/4859
Resource Path: /snap/openscad-nightly/4859/usr/share/openscad-nightly
User Library Path: /home/gareth/snap/openscad-nightly/4859/.local/share/OpenSCAD/libraries
User Config Path: /home/gareth/snap/openscad-nightly/4859/.config/OpenSCAD
Backup Path: /home/gareth/snap/openscad-nightly/4859/.local/share/OpenSCAD/backups
OPENSCADPATH: <not set>
OpenSCAD library path:
/home/gareth/snap/openscad-nightly/4859/.local/share/OpenSCAD/libraries
/snap/openscad-nightly/4859/usr/share/openscad-nightly/libraries

OPENSCAD_FONT_PATH: <not set>
OpenSCAD font path:
/usr/share/fonts
/usr/local/share/fonts
/home/gareth/snap/openscad-nightly/4859/.local/share/fonts
/home/gareth/snap/openscad-nightly/4859/.fonts
/snap/openscad-nightly/4859/usr/share/fonts
/usr/X11/lib/X11/fonts
/System/Library/Fonts
/Library/Fonts
[... 291 lines cut ...]
GL_MESA_texture_signed_rgba
GL_MESA_window_pos
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_copy_image
GL_NV_depth_clamp
GL_NV_fog_distance
GL_NV_half_float
GL_NV_light_max_exponent
GL_NV_packed_depth_stencil
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_NV_texture_env_combine4
GL_NV_texture_rectangle
GL_OES_EGL_image
GL_OES_read_format
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_border_clamp
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays

**Additional context**
Happens with both Manifold and CGAL, which makes sense since it seems like it's probably a Clipper2 issue

kintel, 2025-01-02:
Thanks, we need more Clipper2 testing so don't hesitate to report more such issues!

kintel, 2025-01-03:
When testing the example, I get a bunch of `WARNING: Can't open include file 'layout.scad'. in file param_processing.scad, line 4` and no geometry rendered. Did you forget to commit a file?

If you have a way of reproducing this with a single-file minimal example, that would be awesome!

50an6xy06r6n, 2025-01-03:
Ah I totally forgot that `layout.scad` is in the `.gitignore` 🤦🏻‍♂️ If you copy `example_layouts/eggcat/eggcat.scad` to `layout.scad` it should reproduce. I'll push a commit to add that when I'm at my computer. I'll also look into a more minimal reproduction.

kintel, 2025-01-04:
I haven't been able to reproduce; I get the expected result when testing plate.scad with the most recent instructions.

50an6xy06r6n, 2025-01-04:
I have a one-file repro:
```
/**
 * Business Logic
 */

// Create inner fillets
// offset(-plate_inner_fillet,$fn=360)
offset(delta=plate_inner_fillet)

// Create outer fillets
offset(plate_outer_fillet,$fn=360)
offset(delta=-plate_outer_fillet)

// Offset plate margin and do scale-based fudging to merge points (deprecated)
offset(delta=plate_margin)

// Trim outer edge of the plate to clean up the profile
difference() {
    // Hull each group and then union separately, which allows for concavity
    union() for (group = plate_layout) {
        hull() {
            layout_pattern(group) {
                let(component_type = $extra_data[0]) {
                    if (component_type == "switch")
                        switch_footprint($borders);
                    else if (component_type == "mcu"){
                        mcu_footprint($borders);}
                    else if (component_type == "trrs")
                        trrs_footprint($borders);
                    else if (component_type == "stab")
                        stabilizer_footprint($borders, $extra_data[1]);
                }
            }
        }
    }

    // Generate a negative external profile of the plate (useful for cleaning up convex intersections)
    union() for (group = plate_layout) {
        layout_pattern(group) {
            let(component_type = $extra_data[0]) {
                if (component_type == "switch")
                    switch_footprint_trim($borders, $trim);
                else if (component_type == "mcu"){
                    mcu_footprint_trim($borders, $trim);}
                else if (component_type == "trrs")
                    trrs_footprint_trim($borders, $trim);
                else if (component_type == "stab")
                    stabilizer_footprint_trim($borders, $extra_data[1], $trim);
            }
        }
    }
}

/**
 * Parameters
 */
socket_size = 14;
// Switch spacing distance
unit = 19.05;
// Horizontal unit size (18mm for choc keycaps)
h_unit = unit;
// Vertical unit size (17mm for choc keycaps)
v_unit = unit;

mcu_socket_width = 22;
mcu_socket_length = 37;

// Align mcu to a unit
mcu_unit_resolution = .5;  // Grid size to snap to (as fractional unit)
[... 300 lines cut ...]
                    square([2000, 1000]);
                }
            }
        }
    }
}

function direction_to_vector(direction) = (
    [
        [ 0,  1], // top
        [ 0, -1], // bottom
        [-1,  0], // left
        [ 1,  0], // right
    ][direction]
);

function direction_to_rotation(direction) = (
    [
        0, // top
        180, // bottom
        90, // left
        270, // right
    ][direction]
);
```

kintel, 2025-01-05:
Can't reproduce here. Is this still behaving flakily or does it fail every time for you?

kintel, 2025-01-05:
We've been using ClipperV1 for many years, and upgraded to ClipperV2 in December.

50an6xy06r6n, 2025-01-05:
Hmmm interesting. It reproduces pretty consistently for me (though I can't reproduce the concave defect). It persists through restarts and cache clearing as well.

kintel, 2025-01-05:
Ok, might be CPU-dependent - I'll try again on Intel

rcolyer, 2025-01-05:
For the initial plate.scad with eggcat.scad as layout:
OpenSCAD-2024.12.17.ai21648-x86_64.AppImage -- Has the expected behavior.
OpenSCAD-2024.12.20.ai21770-x86_64.AppImage -- Exhibits the reported bug.

kintel, 2025-01-05:
@rcolyer Yeah, the Clipper2 update was merged Dec 19th

kintel, 2025-01-05:
I managed to slim down the original example to the following. It exhibits a different spike, but seems to fail everywhere, also on Apple Silicon:
```
offset(delta=+1)
offset(+27)
offset(delta=-1)

polygon([
  [111.775,6.286],
  [92,5],
  [73,3],
  [21,3],
  [2.5,5],
  [2.5,76],
  [78.2932,76.776],
  [110.629,95.4452],
  [124.94,103.708],
]);
```

kintel, 2025-01-06:
Further analysis shows that the result after applying the first two offsets above is:
```
polygon([
[75.774, -22.8516], [94.2054, -20.9115], [112.646, -19.7123], [118.209, -18.7592], [123.453, -16.6703],
[128.148, -13.5367], [132.089, -9.49555], [135.103, -4.72334], [137.059, 0.571316], [137.65, 3.61505],
[150.433, 98.2079], [150.6, 103.85], [149.59, 109.403], [147.448, 114.626], [144.267, 119.288],
[140.186, 123.188], [135.383, 126.154], [130.069, 128.056], [124.475, 128.812], [118.847, 128.388],
[113.429, 126.804], [110.176, 125.206], [97.6286, 117.962], [97.6289, 117.962], [71.2031, 102.705],
[3.22358, 102.009], [-2.3837, 101.361], [-7.73384, 99.5622], [-12.593, 96.69], [-16.7488, 92.8703],
[-20.0197, 88.2701], [-22.2626, 83.0903], [-23.3796, 77.5574], [-23.5, 75.0102], [-23.5, 5.89772],
[-22.91, 0.284103], [-21.1657, -5.08417], [-18.3435, -9.97248], [-14.5665, -14.1672], [-10, -17.485],
[-4.84346, -19.7808], [0.59799, -20.9459], [18.1519, -22.8436], [21.0539, -23], [72.9475, -23], ]);
```

Upon inspecting that polygon carefully, the two vertices `[97.6286, 117.962], [97.6289, 117.962]` represent a self-intersection.

The clipper documentation says (https://www.angusj.com/clipper2/Docs/Units/Clipper.Offset/Classes/ClipperOffset/_Body.htm):
_"Offset solutions will often contain tiny segments that don't contribute to path shape. These tiny segments can cause problems when offset solutions are themselves offset. The shape of these new solutions will degrade, and offsetting performance will slow. So, after offsetting, tiny segments are best removed (see the [SimplifyPaths](https://www.angusj.com/clipper2/Docs/Units/Clipper/Functions/SimplifyPaths.htm) function)."_

OpenSCAD currently does not perform such a simplification operation after performing an offset. We do simplify when processing user-provided polygons though, which is why this issue isn't reproducible using a single offset operation.

TODO: Simplify offset results and try again.

kintel, 2025-01-06:
See also https://github.com/AngusJohnson/Clipper2/issues/927

kintel, 2025-01-06:
Update: Managed to reproduce also with a single offset, so we may need to perform some more input cleaning as well:
```
offset(delta=+110) polygon([
[75.774, -22.8516], [94.2054, -20.9115], [112.646, -19.7123], [118.209, -18.7592], [123.453, -16.6703],
[128.148, -13.5367], [132.089, -9.49555], [135.103, -4.72334], [137.059, 0.571316], [137.65, 3.61505],
[150.433, 98.2079], [150.6, 103.85], [149.59, 109.403], [147.448, 114.626], [144.267, 119.288],
[140.186, 123.188], [135.383, 126.154], [130.069, 128.056], [124.475, 128.812], [118.847, 128.388],
[113.429, 126.804], [110.176, 125.206], [97.6286, 117.962], [97.6289, 117.962], [71.2031, 102.705],
[3.22358, 102.009], [-2.3837, 101.361], [-7.73384, 99.5622], [-12.593, 96.69], [-16.7488, 92.8703],
[-20.0197, 88.2701], [-22.2626, 83.0903], [-23.3796, 77.5574], [-23.5, 75.0102], [-23.5, 5.89772],
[-22.91, 0.284103], [-21.1657, -5.08417], [-18.3435, -9.97248], [-14.5665, -14.1672], [-10, -17.485],
[-4.84346, -19.7808], [0.59799, -20.9459], [18.1519, -22.8436], [21.0539, -23], [72.9475, -23]]);
```

kintel, 2025-01-11:
Minimal repro of single-offset case:
```
offset(delta=10) polygon([
  [0, 0],  [100, 0],  [100, 110],
  [50,100],  [55,100], // These two vertices create a very sharp spike
  [0, 90],
]);
```

Note: This actually works as intended; delta offset involving acute convex angles will cause spikes in the result. The way to avoid this in OpenSCAD is to specify `chamfer=true` as an offset parameter, although that's not always what is desired.

Relevant Clipper documentation: https://www.angusj.com/clipper2/Docs/Units/Clipper/Types/JoinType.htm
Specifying chamfer=true in OpenSCAD is equivalent to specifying `JoinType.Square` in Clipper.

Potential item for the wishlist: Allowing unchamfered offsets, but with a miter limit to limit large spikes (`JoinType.Miter` with a configurable miter limit).

There is still something fishy going on with multiple offsets, will look into that next.

kintel, 2025-01-11:
In terms of the multi-offset spike issue: There are two ways of solving this issue:
1. Update Clipper to include the fix for https://github.com/AngusJohnson/Clipper2/issues/916
2. Perform polygon simplification between each offset invocation

Since it will take time to get all clipper dependencies updated to 1., especially considering that this isn't yet included in an official Clipper release, let's go with 2. (which is also recommended by Clipper in genera).

kintel, 2025-01-12:
@50an6xy06r6n Please give the latest master a try. It seems to fix all the issues I could reproduce.

PR #5581 (2025-01-12): Simplify polygons after offset operations
Clipper offset operations can introduce self-intersecting polygons. Their docs recommend simplifying after performing an offset, especially if performing a subsequent offset: https://www.angusj.com/clipper2/Docs/Units/Clipper.Offset/Classes/ClipperOffset/_Body.htm

Fixes #5554
closed 2025-01-12 by kintel