openscad/openscad #5555: rotate_extrude() can generate non-manifold geometry in some cases

kintel, 2025-01-03:
(Separated out from #5135 as separate issue.)

`rotate_extrude()` can, in some cases, generate a non-manifold geometry.
This appears to happen when the extrusion angle is very close to 180 degrees.

Initial analysis indicates that the formula for calculating number of segments from `$fn/$fa/$fs` doesn't match the formula for increasing the angle for each segment, causing the last segment to become close to the precision used by `PolySetBuilder` in rotate_extrude.

Minimal example:
```
eps = 0.0094;
$fn = 3;

module triangle() {
    polygon([[1,0], [2,0], [1,1]]);
}

render() {
  rotate_extrude(angle = 180 + eps) triangle();

  translate([10,0,0]) cube(1);
}
```

For the original discussiont, see comment thread starting with https://github.com/openscad/openscad/issues/5135#issuecomment-2567768225

jordanbrown0, 2025-01-03:
I don't think it has to do with $fa/$fs/$fn.  Rather, I think it is that the calculations used to generate the points for the "end" faces are different from the calculations used to generate the "ring" faces.  I believe they are mathematically equivalent, but when you do two calculations that are mathematically equivalent in different ways, you can end up with differing errors in the very lowest bits of the results.

kintel, 2025-01-03:
Yeah, it's very possible that you're right; I struggle reading that section of the code :)
Ideally, we should fix this by not relying on `PolySetBuilder` and magic merging of vertices using quantization: #5046

kintel, 2025-03-14:
Look like this changed after merging #5553.

kintel, 2025-03-14:
New minimal example:
```
$fn = 3;

module triangle() {
    polygon([[1,0], [2,0], [1,1]]);
}

render() {
  rotate_extrude(start=183, angle = 180) triangle();

  translate([10,0,0]) cube(1);
}
```

kintel, 2025-04-18:
Fixed by #5764

closed 2025-04-18 by kintel