openscad/openscad #5565: Delta offset creates artifacts

UBaer21, 2025-01-05:
version 2024.12.30

also see #5554 

when source polygon has near 0 distance it start generating spikes. Effect stronger on higher resolutions.

```
offset(delta=0.2)
difference(){
  circle(3,$fs=.1,$fa=1);
  translate([0,1])circle(2,$fs=.1,$fa=1);
}
```
[image]

test 2
source 
[image]

```
offset(delta=.6)
difference(){
  union(){
    square(10);
    circle(3);
  }
  translate([2,0])offset(-1.5)square(10);
  translate([0,1])circle(2);
}

```
result
[image]

verbose test code:
```
$fs=.1;$fa=1; // has near zero influence

for(x=[0:10],y=[0:10])translate([x*20,y*20])offset(x/10) offset(-x/10) offset(-y/10) offset(delta=y/10)
difference(){
  union(){
    square(10);
    circle(3);
  }
  translate([2,0])offset(-1.5)square(10);
  translate([0,1.0])circle(2);
}

translate([50,155])color("red")
difference(){
  square(120,true);
  square(115,true);
}
```

result

[image]

UBaer21, 2025-01-05:
These two should not have that big difference

```
offset(delta=0.5)polygon([[+0.00003,0],[0,0],[0,15]]);
translate([1,0])offset(delta=0.5)polygon([[+0.00003,0],[0,0],[0,16]]);
```

[image]

kintel, 2025-01-05:
In the first few examples, the spike is expected. Depending on how sharp edges occur in the source polygon, spikes can be very large. The general way of avoiding spikes is to add `chamfer=true` to any `offset(delta=)`.

In the last case, I believe what happens is that the vertex at the tip of the spike overflows Clipper's coordinate system, in which case all bets are off.

UBaer21, 2025-01-06:
Yes but shouldn't a counter offset remove the spike completely? 
```
offset(-0.2)offset(delta=0.2)
difference(){
  circle(3,$fs=.1,$fa=1);
  translate([0,1])circle(2,$fs=.1,$fa=1);
}
```
At least this works in most cases `offset(-.5)offset(delta=0.5)circle(10,$fn=3);`

UBaer21, 2025-01-06:
There are also some weird steps
[image]

when we are in the 10⁻⁶ region

```
x=+0.000001;
for(i=[0.00:500])translate([i*(x+1),0])
offset(-1)offset(delta=1)polygon([[x/2,0],[-x/2,0],[0,i]]);
```

kintel, 2025-01-12:
Simple version of the initial example:
```
offset(delta=100) polygon([[0,0.0001], [10,0], [-10,0]]);
```

Challenges:
* This is "works-as-intended" by Clipper. Clipper, however, have a mechanism called _miter limit_ which can be used to cap very long spikes (as a multiplier of the offset amount). OpenSCAD currently doesn't surface this feature, but perhaps we should. We do, however support a `chamfer=true` argument, which will cap offset amount of acute convex angles to be <= offset distance from the original vertex, effectively removing any spikes, see `JoinType.Square` here: https://www.angusj.com/clipper2/Docs/Units/Clipper/Types/JoinType.htm
* At extremely long spikes or for extremely small details, we run into issues due to Clipper working in integer coordinates (essentially fixed-point 64-bit integers). The precision currently used in OpenSCAD is to use 27 fractional bits. Once you (or intermediate calculations) get close to the limits, things like `offset(10) offset(-10)` may not cancel each other out without introducing some error.

@UBaer21 given the above, and the discussion in #5554, is there anything you're like to see changed in OpenSCAD (within the realm of realistic possibilities) ?

UBaer21, 2025-01-12:
As we also had these in version 2019 this isn't a new issue. And a result of a bad design in the first place. Not #5554 though. But that seems to be a very specific rare issue, which sure is annoying if it happens - and shouldn't. It sure is easy to workaround but as these are errors that are hard to predict they are a problem in generative design.  
In general some clamping to 10⁻¹⁵ may eliminate these errors - and they are hard to find as echo will not tell these small deviations. So we don't have points that are very close nearby.
But else i am fine.

UBaer21, 2025-01-15:
I was hoping that using chamfer removes the problem but it doesn't. For certain fn values it now generates these edges
[image]

kintel, 2025-01-15:
Hm, could you explain what's wrong with your screenshot?

UBaer21, 2025-01-15:
should look like 
[image]

kintel, 2025-01-16:
@UBaer21 Could you share the exact code which triggers the artifact?

UBaer21, 2025-01-16:
here a smaller example
```
use<ub.scad>;
Rund(+0.4,0.5,fa=1)Cycloid(option=2,fn=75,modul=3,f=3);
Tz(.2)color("blue")Cycloid(option=2,fn=75,modul=3,f=3);
```
[image]

```
group() {
  offset(r = -0.5, $fn = 0, $fa = 1, $fs = 0.2) {
    offset(delta = 0.5, chamfer = true, $fn = 0, $fa = 1, $fs = 0.2) {
      offset(r = 0.4, $fn = 0, $fa = 1, $fs = 0.2) {
        offset(delta = -0.4, chamfer = true, $fn = 0, $fa = 1, $fs = 0.2) {
          group() {
            group() {
              group();
            }
            multmatrix([[0.951057, 0.309017, 0, 0], [-0.309017, 0.951057, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
              polygon(points = [[5.68932, -1.90802], [5.72426, -1.80047], [5.76276, -1.69425], [5.80468, -1.58965], [5.84988, -1.48697], [5.89819, -1.38648], [5.94944, -1.28845], [6.0034, -1.19314], [6.05987, -1.10077], [6.11859, -1.01157], [6.17931, -0.925737], [6.24177, -0.843447], [6.30567, -0.764859], [6.37073, -0.690112], [6.43665, -0.619319], [6.5031, -0.552572], [6.56979, -0.489939], [6.63638, -0.431465], [6.70257, -0.377169], [6.76801, -0.327047], [6.8324, -0.281072], [6.89542, -0.23919], [6.95676, -0.201325], [7.01611, -0.167379], [7.07317, -0.13723], [7.12767, -0.110733], [7.17931, -0.0877253], [7.22786, -0.0680205], [7.27305, -0.0514151], [7.31467, -0.0376874], [7.3525, -0.026599], [7.38634, -0.0178964], [7.41603, -0.0113123], [7.44141, -0.00656725], [7.46236, -0.00337115], [7.47876, -0.00142507], [7.49054, -0.000422852], [7.49763, -5.29022e-05], [7.5, 0], [7.50289, 9.69737e-05], [7.51154, 0.000774794], [7.52587, 0.00260934], [7.54575, 0.00616658], [7.57098, 0.0119978], [7.60133, 0.0206347], [7.63653, 0.032585], [7.67623, 0.0483283], [7.72006, 0.0683113], [7.76762, 0.0929448], [7.81844, 0.122599], [7.87206, 0.157603], [7.92795, 0.198238], [7.98559, 0.244737], [8.04441, 0.297284], [8.10385, 0.35601], [8.16333, 0.420994], [8.22227, 0.492261], [8.28007, 0.569781], [8.33616, 0.65347], [8.38996, 0.743194], [8.44093, 0.838762], [8.48853, 0.939937], [8.53223, 1.04643], [8.57157, 1.15791], [8.60607, 1.274], [8.63534, 1.39429], [8.65899, 1.51831], [8.67669, 1.64558], [8.68814, 1.7756], [8.69312, 1.90781], [8.69143, 2.04166], [8.68294, 2.17658], [8.66756, 2.31198], [8.64526, 2.44727], [8.61606, 2.58186], [8.58004, 2.71516], [8.53733, 2.84661], [8.48811, 2.97562], [8.43263, 3.10167], [8.37115, 3.22423], [8.30401, 3.34281], [8.23158, 3.45696], [8.15427, 3.56624], [8.07253, 3.67027], [7.98684, 3.76872], [7.89771, 3.86129], [7.80568, 3.94772], [7.7113, 4.02783], [7.61515, 4.10147], [7.5178, 4.16854], [7.41984, 4.22901], [7.32187, 4.28288], [7.22446, 4.33024], [7.12819, 4.3712], [7.03362, 4.40594], [6.9413, 4.43468], [6.85173, 4.45769], [6.76541, 4.47531], [6.6828, 4.48788], [6.60433, 4.49581], [6.53037, 4.49956], [6.46127, 4.49958], [6.39732, 4.49638], [6.33877, 4.4905], [6.28581, 4.48247], [6.2386, 4.47287], [6.19723, 4.46228], [6.16174, 4.45126], [6.1321, 4.4404], [6.10826, 4.43029], [6.09009, 4.42149], [6.07742, 4.41455], [6.07002, 4.41001], [6.06763, 4.40839], [6.06568, 4.40704], [6.05973, 4.40317], [6.04961, 4.39706], [6.03519, 4.38899], [6.01637, 4.37926], [5.99304, 4.36818], [5.96515, 4.35606], [5.93266, 4.34321], [5.89554, 4.32994], [5.8538, 4.31659], [5.80748, 4.30346], [5.75662, 4.29087], [5.70131, 4.27912], [5.64165, 4.26853], [5.57777, 4.25938], [5.5098, 4.25196], [5.43792, 4.24654], [5.36232, 4.24338], [5.2832, 4.24273], [5.2008, 4.24481], [5.11534, 4.24983], [5.02709, 4.258], [4.93633, 4.26947], [4.84333, 4.28441], [4.74839, 4.30294], [4.65182, 4.32517], [4.55393, 4.35118], [4.45504, 4.38105], [4.35546, 4.4148], [4.25552, 4.45244], [4.15555, 4.49398], [4.05586, 4.53937], [3.95679, 4.58856], [3.85863, 4.64145], [3.76171, 4.69796], [3.66631, 4.75794], [3.57273, 4.82125], [3.48125, 4.88772], [0, 0]], paths = [[0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29, 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, 42, 43, 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, 56, 57, 58, 59, 60, 61, 62, 63, 64, 65, 66, 67, 68, 69, 70, 71, 72, 73, 74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152]], convexity = 1);
            }
            group();
          }
        }
      }
    }
  }
  group() {
    group() {
      group();
    }
  }
  group();
}
group() {
  multmatrix([[1, 0, 0, 0], [0, 1, 0, 0], [0, 0, 1, 0.2], [0, 0, 0, 1]]) {
    color([0, 0, 1, 1]) {
      group() {
        group() {
          group();
        }
        multmatrix([[0.951057, 0.309017, 0, 0], [-0.309017, 0.951057, 0, 0], [0, 0, 1, 0], [0, 0, 0, 1]]) {
          polygon(points = [[5.68932, -1.90802], [5.72426, -1.80047], [5.76276, -1.69425], [5.80468, -1.58965], [5.84988, -1.48697], [5.89819, -1.38648], [5.94944, -1.28845], [6.0034, -1.19314], [6.05987, -1.10077], [6.11859, -1.01157], [6.17931, -0.925737], [6.24177, -0.843447], [6.30567, -0.764859], [6.37073, -0.690112], [6.43665, -0.619319], [6.5031, -0.552572], [6.56979, -0.489939], [6.63638, -0.431465], [6.70257, -0.377169], [6.76801, -0.327047], [6.8324, -0.281072], [6.89542, -0.23919], [6.95676, -0.201325], [7.01611, -0.167379], [7.07317, -0.13723], [7.12767, -0.110733], [7.17931, -0.0877253], [ [... cut ...]  74, 75, 76, 77, 78, 79, 80, 81, 82, 83, 84, 85, 86, 87, 88, 89, 90, 91, 92, 93, 94, 95, 96, 97, 98, 99, 100, 101, 102, 103, 104, 105, 106, 107, 108, 109, 110, 111, 112, 113, 114, 115, 116, 117, 118, 119, 120, 121, 122, 123, 124, 125, 126, 127, 128, 129, 130, 131, 132, 133, 134, 135, 136, 137, 138, 139, 140, 141, 142, 143, 144, 145, 146, 147, 148, 149, 150, 151, 152]], convexity = 1);
        }
        group();
      }
    }
  }
  group() {
    group() {
      group();
    }
  }
  group();
}
```

kintel, 2025-01-17:
Something is odd indeed, here's a minimal example showing that a tiny spike was generated:
```
offset(r = 0.4, $fn = 3)
  polygon([
    [6.43207, 3.70126], 
    [4.79025, 3.88787],
    [4.77452, 3.89029], 
    [4.77205, 3.89077], 
    [2.34519, 1.63433]
  ]);
```

kintel, 2025-01-17:
The main branch of Clipper2 behaves nicer, but there's still a smaller spike showing up

UBaer21, 2025-01-17:
using your example with r=0.07  shows the spike is not even on the edge, while 0.066 doesn't has a visible spike

[image]

kintel, 2025-01-17:
Yeah, but that was fixed in Clipper's main branch recently. I'll try to understand what's happening before updating, in case there is something we can do on our side to avoid this.

kintel, 2025-01-18:
Opened https://github.com/AngusJohnson/Clipper2/issues/934 - let's see if they can spot an issue in Clipper.

kintel, 2025-01-19:
I've attempted to address this in #5598 - if you're building OpenSCAD yourself you could play with that branch, otherwise wait until it's merged.

kintel, 2025-01-19:
Also note that Clipper1 also introduces small spiky artifacts, so the behavior after #5597 and before #5598 should be very similar to the behavior before porting to Clipper2.

closed 2026-10-16 by github-actions[bot]