openscad/openscad #5570: add fn(), a function that tells you how many sides a circle would have

jordanbrown0, 2025-01-06:
**Is your feature request related to a problem? Please describe.**
Some applications require that you match up the number of sides of various round objects, perhaps of different sizes.  (Imagine building a low-poly tower of different-sized cylinders.)  You could do this by explicitly setting $fn, but that comes with the usual problems of using $fn.  The formula that OpenSCAD uses is documented, so you could just transcribe it, but it's complicated.

**Describe the solution you'd like**
Offer a function `fn(r|d)` that will tell you the calculated number of sides, for the specified radius or diameter and the current settings of $fn, $fs, and $fa.  Note that you could specify $fn, $fs, and/or $fa as arguments to the function.

You could use this `fn()` function to calculate an appropriate number of sides for one of your objects, and then use that value for $fn to create your other objects.

jordanbrown0, 2025-01-06:
Note that if we ever decide to tweak the formula used for the number of sides - perhaps to fix #5101, or perhaps to always generate a multiple of four sides - then this function would be kept consistent with the circle implementation.

nophead, 2025-01-06:
I have both of those in NopSCADlib.
```
function r2sides(r) = $fn ? $fn : ceil(max(min(360/ $fa, r * 2 * PI / $fs), 5));    //! Replicates the OpenSCAD logic to calculate the number of sides from the radius
function r2sides4n(r) = floor((r2sides(r) + 3) / 4) * 4;                            //! Round up the number of sides to a multiple of 4 to ensure points land on all axes
```

jordanbrown0, 2026-05-08:
Note that the introduction of $fe means that all existing userspace implementations of this function are no longer complete.  That kind of change is precisely why it should be a builtin.

MichaelAtOz, 2026-11-01:
The cheatsheet has `$fn  - number of segments`.

The wiki section`Circle resolution: $fa, $fs, and $fn`,  `segment` is used 17 times.

That section also has `facet` - 'faceted polygon' (used once). So `facets()` is an option.
**f**acets starts with `f` - so sort-of goes with $**f**n/$**f**a/$**f**s, but you'd want to edit the wiki to use it instead of segments.
But looking at the dictionary, facets is commonly related to surface rather than lines. 

The wiktionary entry for Segment is further below. But I think this is the important part:

> 3. (mathematics) A portion.
    A straight path between two points that is the shortest distance between them; a line segment.
    (geometry) The part of a circle between its circumference and a chord (usually other than the diameter).

So the word to describe what we are talking about is `Segment`. 
**Why** do we need something other than `segments()`?

But as I said before, I have nothing against `fn()`, it is the best don't-use-a-meaningful-word option.

 ## Wiktionary
 ### Etymology
From Latin segmentum (“a piece cut off, a strip, segment of the earth, a strip of tinsel”), from secāre (“to cut”).

### Noun

segment (plural segments)

    1. A length of some object.

        a segment of rope

    2. One of the parts into which any body naturally separates or is divided; a part divided or cut off; a section; a portion. 

    Synonyms: cleft, clove
    a segment of an orange; a segment of a compound or divided leaf

    3. (mathematics) A portion.

    A straight path between two points that is the shortest distance between them; a line segment.
    (geometry) The part of a circle between its circumference and a chord (usually other than the diameter).
    (geometry) The part of a sphere cut off by a plane.
    (topology) Any of the pieces that constitute an order tree.

    4. (sciences) A portion.

    (phonology) A discrete unit of speech: a consonant or a vowel.
    (botany) A portion of an organ whose cells are derived from a single cell within the primordium from which the organ developed. 

    (zoology) One of several parts of an organism, with similar structure, arranged in a chain; such as a vertebra, or a third of an insect's thorax.

    5. (broadcasting) A part of a broadcast program, devoted to a topic.

        The news showed a segment on global warming.

    6. (computing) An Ethernet bus.
    7. (programming) A region of memory or a fragment of an executable file designated to contain a particular part of a program.
    8. (travel) A portion of an itinerary: it may be a flight or train between two cities, or a car or hotel booked in a particular city.

### Synonyms

    (part or section of a whole): lith
    (straight path): line segment
    (area of a circle): circular segment

closed 2026-11-11 by jordanbrown0