openscad/openscad #5571: Add a distort() transformation

ErezAmir63, 2025-01-06:
##A very generic distort transformation
**syntax**:
`distort(function([x,y,z]) [newX, newY, newZ]) 

**sample usage **
do nothing:
`distort(function([x,y,z]) [x,y,z]) cube([1,1,1]); `
Translate X:
`distort(function([x,y,z]) [x+5,y,z]) cube([1,1,1]); `
Scale all X 2
`distort(function([x,y,z]) [x*2,y*2,z*2]) cube([1,1,1]); `
Mirror x values
`distort(function([x,y,z]) [-x,y,z]) cube([1,1,1]); `
Trapezoid
`distort(function([x,y,z]) [x*(1-z),y*(1-z),z]) cube([1,1,1]); ` // x,y values getting smaller as Z gets larger
Twist
`distort(function([x,y,z]) let a=z*180 [x*cos(a)+y*sin(a),y*cos(a)-z*sin(a),z]) cube([1,1,1]); `

pca006132, 2025-01-06:
The issue is that this can easily create invalid meshes. Ideally, you can do this with #4478 or something similar that allows conversion of geometry to list of points that openscad user code can work with.

jordanbrown0, 2025-01-06:
Other notes:

- I think all of the examples that you give can be done using multmatrix.  But there are others (e.g. bends) that can't.
- The distort() function would need to render its children, which is a performance hit.
- There's a subtlety in any non-trivial distortion, that you will often want to break line segments up so that what was a straight line is no longer.  For instance, if you want a *real* twist, you need to break the cube's vertical lines up into many small segments.  This means that a point-by-point approach really isn't powerful enough.

gsohler, 2025-01-06:
Distortion of a solid only makes sense, 
when its applied equaly   on all surface.
right now,  distortion can only be applied to the vertice, which can change coordinate.
to get a read good distortion effect,  all surfaces need to be split into sub-rectnagles firs which will yield yet more  vertices, which can be disorted to achieve a well balanced distortion effect.