openscad/openscad #5573: Customizer "lock parameter set" feature

MethylBromide, 2025-01-07:
**Is your feature request related to a problem? Please describe.**
I often write scripts with a lot of parameters to satisfy a bunch of different use cases. I then might end up putting a fair amount of effort into finding out the correct set of parameters for specific applications. I give the set of parameters a name to keep track of it in case I need to reproduce that part again later, or to use it as the starting point to create a variant.

The problem is that it's **way** too easy to accidentally change one or more parameters without realizing it. For instance, numbers can change because I'm trying to scroll in the Customizer view and focus is in a number field, which increments or decrements by the scroll wheel. And then I may save my changes and unknowingly overwrite the saved configuration that I put good effort into getting exactly right.

**Describe the solution you'd like**
I want something in the UI to lock a parameter set so it can't be accidentally modified. A little padlock icon at the top of the settings, for instance. Once I've got a configuration exactly right, I would lock it. I can unlock it if I need to, or If I make a copy using the "+" button, the copy is unlocked. The "-" control should be disabled for locked parameter sets.

**Describe alternatives you've considered**
This seems the natural way to solve the problem, but I'm open to alternatives. Another thing that would be helpful would be to allow the "edit > undo" menu selection or Ctrl+Z operate in the Customizer window. That would not solve the problem, however.

**Additional context**

jordanbrown0, 2025-01-08:
Another option would be to require an explicit save of a set of parameters, probably independent of saving the primary model.

MethylBromide, 2025-01-09:
> Another option would be to require an explicit save of a set of parameters, probably independent of saving the primary model.

That would improve matters, but it sounds more complicated to implement and not really as good a solution. It would have to include something to not delete a parameter set you accidentally hit "-" on, while saving other changes. You could easily end up producing an export file with the unintentionally changed parameters before being prompted, realizing you probably screwed up, and having to back up and create new export files from a reverted version. It just seems like more steps and more bother to go back and fix the problem rather than preventing it as I suggested,