openscad/openscad #5574: crash in projection function

MethylBromide, 2025-01-08:
**Describe the bug**
An example STL import file causes a client crash when **projection** is called on the imported object.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the sample STL file from https://www.printables.com/model/493676-zardoz?lang=en
2. Execute the code below in the client (adjust the value of importfile to the actual filepath of the sample STL file).

**Expected behavior**
Expected to see a 2D object which is the "shadow" of the imported STL file.

**Code reproducing the issue**
```
importfile = "zardoz_PD.stl";
projection()
import(importfile);
```

**Screenshots**
n/a

**Environment and Version info (please complete the following information):**
 - OS: Windows 11
 - System: Intel PC 64-bit
 - OpenSCAD Version: 2025.01.06 (git 92a13b4f0) from openscad website.

**Library & Graphics card information**
n/a

**Additional context**

pca006132, 2025-01-08:
Are you using the manifold backend or CGAL backend?

MethylBromide, 2025-01-08:
Manifold.

UBaer21, 2025-01-08:
It shouldn't crash but the STL has some errors 
[image]

pca006132, 2025-01-08:
I found the issue, it is due to a nullptr in openscad. I fixed it and is setting up regression test for it.

PR #5575 (2025-01-09): Fix manifold project nullptr
Fixes #5574.
closed 2025-01-09 by kintel