openscad/openscad #5576: faulty update on SVG rectangles

petaflot, 2025-01-08:
**Describe the bug**
modifying the nodes of an SVG rectangle is not registered when reloading (F5)

**To Reproduce**
in (ie.) inkscape, create a `rectangle.svg` file that contains a single rectangle

in openscad, create a `rectangle.scad` file that contains:
```
import("rectangle.svg");
```

the rectangle is shown.

* go back to inkscape, hit `R` or `F2` and move a corner node of the rectangle. save the file.
* go back to openscad, hit `F5`. rectangle is unchanged
* in inkscape, move said rectangle, save file
* go back to openscad, hit `F5`. rectangle is changed (and moved)
* go back to inkscape, move rectangle back to original position (rather, undo last move), save file
* go back to openscad, hit `F5`. rectangle is changed (and where it should be)

**Expected behavior**
no need to "change node position, move, save, reload, undo move, save, reload" ; instead only "change node position, save, reload"

**Code reproducing the issue**
see above. no problems with paths ; not tested with ellipses and spirals and star/polygon etc.

**Environment and Version info (please complete the following information):**
 - OS: Gentoo Linux
 - System: AMD64
 - OpenSCAD Version: 2024-12-30

**Additional context**
hihi

kintel, 2025-01-12:
Just tested here, and it works as expected for me; moving a corner node shows up on F5.
Is this still an issue with the latest snapshot. If so, I suspect there is something about the changes on the second save that we're not picking up. Could you provide some more info, and perhaps share the files in question?

t-paul, 2025-01-12:
I'll have to verify again, but while looking at 3MF I've noticed the same non-update  issue, so maybe it's a Linux issue?

closed 2026-10-17 by github-actions[bot]

jordanbrown0, 2026-10-17:
Not duplicated on 2026.10.05 on Windows 11, with Inkscape 1.3.2.