openscad/openscad #5580: OpenSCAD appimage icon Missing.

juanpc2018, 2025-01-11:
some .appimage have issues with icons,

seems libfuse2 & 3 installed at same time is the reason.

OpenSCAD .appimage does Not display icon

t-paul, 2025-01-11:
Which "some"? Icon is missing where exactly (screenshot!)? Which exact distribution and desktop environment?

Why did you ignore the bug reporting form?

juanpc2018, 2025-01-13:
Happens in Kubuntu 22.04 / .5 .4
Does Not happen in 24.04.0
But 24.04.0 has other problems...

I had to reinnstall several times, didnt had time to take a screenshot. Later will do.

kintel, 2025-04-13:
Assuming fixed, awaiting confirmation

juanpc2018, 2025-04-13:
0% AppImage Icons showing on 20.04 LTS
50% showing in 22.04 LTS 50% dont.
100% in 24.04 LTS 
its an Nvidia propietary driver issue, 
drivers are "incomplete",

Nouveau NV137 driver all AppImage icons show in 20.04.4 LTS

right Now i have Nouveau driver installed = cant confirm,

but the only way fixing that problem "becoming the 50% that shows on 22.04"
is using a different AppImageKit version
or a different package method like https://github.com/VHSgunzo/sharun

https://github.com/geoffreybennett/alsa-scarlett-gui/issues/157#issuecomment-2789418076