openscad/openscad #5583: Add echo_export() module

CrashCash, 2025-01-12:
I use OpenSCAD in my 3D printing pipeline. It's one of the best pieces of software I've ever dealt with.

I export to OBJ since it's a better format (IMHO) than STL. One of its advantages is allowing comments.

For example, this is a perfectly valid beginning of file:
```
# OpenSCAD obj exporter
# wire_gap=2.25
# wire_width=6
v 0 0 0
v 0 0 5
v 37.4167 14.1507 0
```

Would it be possible to have an "echo_export()" function that works like echo() but embeds comments in the export file, if the file format supports it? I wouldn't mind if it was just OBJ files. I assume during rendering you could generate a list of strings for use by the export code, but that's only a suggestion.

This would be extremely useful with the customizer. You could do "echo_export(wire_width=wire_width);" and it would put a comment line in the header as above. This would allow you to simply inspect the file to see what customizer values you used to create it. It would be far better than trying to jam them all into the export filename, especially for large numbers of variables.

Thanks for considering this.

CrashCash, 2025-01-12:
It might also be sufficient to simply write all the customizer variables to the exported file. 

I looked at the source, and I don't know which would be quicker to develop. I also discovered I'm 100% not a C++ developer.

CrashCash, 2025-01-12:
Don't know why github closed this issue.

gsohler, 2025-01-12:
Never mind, i like your idea, too

gsohler, 2025-01-12:
This could be a reference implementation.
I think i am flexible to adapt the internal mechanisms, once openscad introduces @ Annotations

[image]