openscad/openscad #5589: Combo boxes in Preferences are all blank

monroewilliams, 2025-01-15:
**Describe the bug**
All combo boxes in the preferences window are blank.

**To Reproduce**
Open the Preferences window in the app. Try to change any of the settings that are set by a combo box. Note that all the combo box lines are blank.

**Expected behavior**
Combo box popups should have text.

**Code reproducing the issue**
N/A

**Screenshots**
[image]
[image]

**Environment and Version info (please complete the following information):**
 - OS: MacOS 15.1.1 
 - System: Mac 
 - OpenSCAD Version: current head of master (first noticed in daily build, then confirmed by building from source)

**Library & Graphics card information**
N/A

**Additional context**
It appears that commit a71071c73 broke this. It added a "name" field to the SettingsEntryEnum::Item structure, but did not take into account that many of these structures are initialized using direct structure assignment, which depends on the order of data members in the struct and is inherently unsafe. (See Settings.cc, line 149 and onwards for the initializations in question).

For example, in this line:

```
SettingsEntryEnum<std::string> Settings::tabKeyFunction("editor", "tabKeyFunction", {{"Indent", _("Indent")}, {"InsertTab", _("Insert Tab")}}, "Indent");
```

`{"InsertTab", _("Insert Tab")}` is the initializer for one SettingsEntryEnum::Item structure. Before the change, it would have assigned "InsertTab" to .value and "Insert Tab" to .description, after the change it assigns "Insert Tab" to .name and leaves .description set to the empty string.

This means that when `InitConfigurator::initComboBox()` later does this:
```
    comboBox->addItem(QString::fromStdString(item.description), QString::fromStdString(item.name));
```
it adds the item with an empty string as its contents.

monroewilliams, 2025-01-15:
@t-paul , you might want to take a look at this.

kintel, 2025-01-15:
Thanks for detailed report and analysis! Fixed in #5590 - but not sure if I have time to merge it before I go to bed..

t-paul, 2025-01-15:
I'll have a look. It's supposed to be set via the serialized value that's also used in the config file (and later command line). I guess additional initialization is fine, but ideally it should not be needed.

commit 8f49712f (2025-01-15): Merge pull request #5592 from openscad/fix-settings-comboboxes

Prevent incomplete initialization; add names to enum-settings (fixes #5589).
commit a98e01dc (2025-01-15): Prevent incomplete initialization; add names to enum-settings (fixes #5589).
closed 2025-01-15 by t-paul