openscad/openscad #5591: lower $fs to at least 1μm

UBaer21, 2025-01-15:
`WARNING: $fs too small - clamping to 0.010000 in file`
0.01 is still a lot and we should allow 0.001 as fragment size.

jordanbrown0, 2026-08-30:
It's a bit hard to see the use case for $fs values that small - typical 3D printer resolution is around 0.1mm.

But the behavior would be well-defined, so why not?

UBaer21, 2026-08-30:
The 3mf precision is 10⁻⁶  so using mm we getting down to nm ( 0.000 001mm) , and μm (0.001mm) are the min for any CNC.  The G-Code is usually down to μm resolution - to keep this precision also for converting moves into acr (G2/G3 move) the polygon need to have higher resolution. I would consider a 10μm precision in a digital file as quite bad (today). DLP printer are currently (2026) at 10-20μm pixel precision. So while physical output is around 0.01mm accuracy (repeatability) and with a 400μm nozzle diameter this may seem unneeded but we should aim for the future allowing high precision but not as default. The better would be to use  splines. However the 1 µm chordal deviation is the goal which is already met with a $fs of 0.1mm  -  so finer $fs might not be needed (and will slow down).  Just don't like software telling me limits that came from 20yr ago.